BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #1: Will this work with the newest Raspberry Pi Zero (w/ WiFi and Bluetooth)

BaReinhard, 2017-02-28:
Will this repo/install worth the newest Raspberry Pi Zero, which comes with onboard WiFi and Bluetooth?

BaReinhard, 2017-02-28:
I currently, have a new Raspberry Pi Zero, with WiFi and Bluetooth, on order and should receive it within the week. I have also ordered a HiFiBerry DAC+ Zero to test with the setup as well.

 I would assume that given a proper power supply there should be no issues with this installation. 

Given,

> It uses the same Cypress CYW43438 wireless chip as Raspberry Pi 3 Model B to provide 802.11n wireless LAN and Bluetooth 4.0 connectivity.

Taken from the Raspberry Pi Article on the new product, I don't foresee any issues with hostapd drivers, which means there won't be a need to compile hostapd yourself and can use `apt-get install hostapd` to install it. Further meaning, this installation package should work alongside the latest Raspbian lite image

RogueM, 2017-03-03:
I can confirm it works perfectly using your `ap_config` script, yes. I'm planning to use it at the Raspberry Pi Big Birthday Party so I'll let you know if things don't pan out but so far working fine.

BaReinhard, 2017-03-03:
What are you planning to use it for and do you mean option 3 or the car version with the ap included?

closed 2017-03-04 by BaReinhard

RogueM, 2017-03-05:
I am using my phone (or visitors' phone) to play music to a zero W fitted with a pHAT BEAT.