BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #4: Jittery audio with wifi enabled on RPI3 Model B

nilayp, 2017-03-02:
I have a RPI3 Model B with a HifiBerry Digi+. I just ran through the installation and it worked great... but I have a problem with stuttering audio. I found this article: https://github.com/raspberrypi/linux/issues/1402 ... which states that this is a known issue when wifi is enabled. Sure enough, when I disabled wifi, the stuttering stopped.

I didn't see anything in this project that referenced this problem. Were you able to resolve it? If not - I guess I'll buy a Wifi dongle to workaround the issue.

Thanks!

-- Nilay

BaReinhard, 2017-03-02:
Hey Nilay, thanks for your concern with the project. I haven't had any issue with jittery audio. I have been using the project with a amp+. Might you please let me know what installation you chose, 1,2,3,4.  Also how are you powering the Pi, I know issues arise when the power supply being used isn't suited for the needs of the board.

BaReinhard, 2017-03-02:
I also took a look at the issue you referenced it's interesting that many others have the same issue. Like I said I tested it on a hifiberry amp+, which uses a 12v power source. I have a DAC+ pro I just received today. I'll test it on that to see if I can reproduce your issue. 

Could you also include what OS you are running and if it's Jessie whether or not it's the lite or pixel version. Thanks!

BaReinhard, 2017-03-03:
One last thing I am curious about is, what is your Pi doing when this occurs? Are you surfing websites? Downloading some files?, etc...?

nilayp, 2017-03-03:
Brett - thanks for the quick response! I chose installation option #2 and am powering the Pi with a 5V, 2.5A adapter. I tried it with a 12V power adapter... it continues to stutter.

Lastly - my RPI is headless and dedicated to being an audio receiver. At the time I was experiencing the stuttering, the only operation active was the bluetooth audio. I was connected to the RPI via ssh from my computer, but wasn't running any commands. I'm running the latest Jessie lite. I just set up Jessie a few tools and ran your installation program (which is magic!) yesterday.

BaReinhard, 2017-03-04:
Thank you for the kind words. What other tools did you setup prior to using the installation program?

It's just a bit weird to me that this all started happening, as I haven't had an issue with it. I want to test the issue you are having but haven't been able to reproduce it. Possibly, its one of the extra tools you downloaded that is clashing.

nilayp, 2017-03-04:
I tried to install it all over again, with the same result. This time, I kept careful notes and disregarded installing any random software.

(1) I got the latest version of Jessie from here https://www.raspberrypi.org/downloads/raspbian and installed it with SSH enabled. 
(2) I recreated the SSH keys
```
$ sudo rm /etc/ssh/ssh_host_*
$ sudo dpkg-reconfigure openssh-server
$ sudo service ssh restart
```
(3) After install, I updated, upgrade the OS and installed git:
```
$ sudo apt-get update
$ sudo apt-get upgrade
$ sudo apt-get git
```
(4) I used raspi-config to change the hostname
(5) I enabled Wifi by modifing  /etc/wpa_supplicant/wpa_supplicant.conf with my ssid and psk.
(6) I rebooted and unplugged the ethernet.
(7) After reboot, I once again SSH'ed into the rpi, cloned your git repo and ran the installation script. This time, I ran the custom installer and only installed airplay and bluetooth, selecting the Digi+ as my sound card.
(8) Like last time, airplay works just fine, but bluetooth jitters. If I disable the wifi, bluetooth works perfectly.

You are trying to reproduce this with RPI 3 model B?

If you don't have any other thoughts, I'll try buying a USB Wifi dongle and try that way. I feel like I'm really close to making this work. :)

-- Nilay

BaReinhard, 2017-03-04:
Thanks Nilay for the detailed description. I am at a loss as to why you are experiencing the jitter and I am not. I just tested the project last night with the latest Raspbian Jessie Image 03/02/17 or 02/03/17 depending how you write the dates.

The only thing I did not do was recreated the SSH keys (which I don't think should make any difference). The repo will automatically update and upgrade the OS for you (for future reference), but that shouldn't make any difference.

I had the install working fine with the use of a USB audio card (Bluetooth with no jitter) and the wifi connected to my phone as a hotspot. I will try again later tonight with my home router and the DAC+ PRO.

I am using a RPI 3 model B.

I guess for the time being the fix for this will be using a USB wifi adapater. 

I will keep this issue open as it still needs fixing and will be a good reference for those looking with a similar issue.

nilayp, 2017-03-04:
It was Jessie Lite.

I just disabled the Wifi and plugged the RPI into an ethernet switch. It works flawlessly. I'll try with a USB wifi adapter and report back.

smadgal, 2017-03-05:
I can confirm that I am having the exact same issue as @nilayp using a RPI3 Model B running Raspbian Jessie. Disabling the wifi does resolve the issue

BaReinhard, 2017-03-06:
I was able to reproduce the issue today, it looks as though this may be a driver issue with the bcm chipset. I have tried changing priority and niceness of pulseaudio but to no avail. I am continuing to look for solutions, but given that is is an issue on the RPIFoundations own github repo I doubt there is a solution to this at the moment. Especially, since they just released the Pi Zero W I would also assume that they are working to fix this issue.

aycgit, 2017-03-09:
same problem here

BaReinhard, 2017-03-09:
@aycgit , so with the latest fix you were able to install on Jessie pixel without the boot loop?

aycgit, 2017-03-09:
it auto-rebooted 3 times when installing but then the pi started. but the audio is still funky

edit: and it ruins pulseaudio perms but ill make a issue for that in a minute

nilayp, 2017-03-12:
I purchased the CanaKit WiFi adapter (https://www.amazon.com/gp/product/B00GFAN498/) and it solved the jittery problem with bluetooth on my RPI3. I first disabled the onboard wifi with the following command: ```$sudo iwconfig wlan0 txpower off```
And then plugged in the USB wifi. It showed up as wlan1 and connected to my wifi network. Airplay and Wifi both work.
The disabled onboard wifi seems to stay disabled after *MOST* reboots... but not all of them. I have no idea why that is. Anyone have any suggestions for that?

BaReinhard, 2017-03-12:
I would suggest disabling the drivers, given that this doesn't disable wlan1 as well. What you could alternatively do is set `iwconfig wlan0 txpower` off in `/etc/rc.local` or a similar command to disable it on each boot. You could alternatively use `crontab` `@reboot`. However, `@reboot` has been known to be problematic on several linux distro's, I believe Raspbian is also plagued with this, but it may be worth a shot.

BaReinhard, 2017-05-19:
Were you able to get something similar implemented? @nilayp

nilayp, 2017-05-19:
I couldn't find a reliable way of scripting this. When you have both the onboard Wifi and Wifi via USB, on reboot, either NIC can be wlan0... it's super annoying. If the system reboots and the audio is jittery (the onboard Wifi becomes primary,) I have to pull out the USB Wifi, reboot the system. Connect a computer via the ethernet port and disable the onboard wifi... and then plug in the USB wifi.

giobauermeister, 2017-05-20:
Disabling the WiFi also worked for me.
`sudo ifdown wlan0`

kstawiski, 2017-05-30:
I have faced the same problem. It's strange but if you do not need airplay or other sophisticated services this tool can offer you can just flash raspian and **install only Bluetooth reciver (choose option nr 4 at the beginning)**. It worked in my case.

BaReinhard, 2017-05-31:
@giobauermeister I am glad that that line of code worked for you! It's what I used on my system for work. I am still looking into solving this issue without needing another dongle, there seems to be some promise in one aspect of this that I will be trying out (hopefully) relatively soon.

@kstawiski Absolutely, if you only need bluetooth I suggest you do just that. As far as consistency and universal playback, this is what I would suggest. You can also disable the wifi completely and only use a ethernet for updates to ensure no wonky playback happens. I have been experiencing some choppiness in my project that even uses an external dongle. It's likely some other issue, but I may just choose to solve it by what you have suggested here.

maximus5684, 2017-07-31:
Just some FYI on what is going on here. The Bluetooth transceiver and the WiFi transceiver are both on the same chip which is only connected to a single USB bus. When WiFi is enabled, it causes interrupts to trigger on that bus, interrupting the flow of the Bluetooth data from the chip while WiFi data are transmitted. This is why an external WiFi adapter is required if you are going to use Bluetooth for audio streaming.

BaReinhard, 2017-12-01:
I wonder if we might want to implement a fix for this in the form of shutting down wlan0 while bluetooth is connected and power it back up when bluetooth is disconnected. We could possibly create a config file it checks before doing so, in the case of wanting to debug an issue with ssh and bluetooth streaming at the same time.

BaReinhard, 2017-12-21:
the above mentioned, shutdown of wifi when bt device is connected with a simple script, it is not used by default, but can be enabled.

closed 2017-12-21 by BaReinhard