BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #5: Volume level is low with digital output; need to crank up receiver

nilayp, 2017-03-04:
I using a RPI with a Hifiberry Digi+. I've connected the RPI to my audio receiver using a SPDIF cable. 
I also have an AppleTV, Tivo and Playstation connected with digital inputs to the same receiver. The AppleTV, Tivo and Playstation all produce about the same sound level when the receiver is set to 55 dB. 

To get the RPI to the same sound level, I have to crank up the receiver to 70-75 dB. And that's when the streaming device is turned up all the way. In my case, this happens with my Amazon Tap connected via Bluetooth or my iPhone connected via Airplay.

Since the RPI is only outputting a digital stream, I don't think there is any "volume" control on it. That said, it appears from readlng online that software volume control is something that can be enabled. Is it enabled in this configuration? If so, how do you turn it up and down?

BaReinhard, 2017-03-04:
Hey nilayp thanks for the reminder, I've been meaning to alter the volume levels, most of my testing is done when my daughter is asleep so hence the quieter volumes. Have you tried setting the volumes higher via amixer? Bluetooth should change volume when the streaming device changes volume. However, I don't believe it first mimics the devices initial volume and only does so after volume change from the device. You should be able to set the volume for AirPlay by editing the  shairportstart.sh script given its enabled in the shairport-sync.conf settings. Give me a couple minutes to get in front of a computer and I'll outline it better.

BaReinhard, 2017-03-04:
Uncomment this line:
https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/airplay_config.sh#L69

and the following line to use /home/pi/shScripts/shairportstart.sh & shairportend.sh

Following that you can adjust this:
https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/shScripts/shairportstart.sh#L2 
to set the volume shairport starts at.

If that doesn't work you can use a different setting with amixer:

`amixer -c 1 set Master 80%`
Depending on how the hifiberry deems the pcm you may have to change `Master` to `Speaker` or whatever is shown when you run `amixer`

For bluetooth you should be able to set the volume in a similar manner `amixer set Master 80%` or use the CLI GUI `alsamixer`

BaReinhard, 2017-03-06:
Lines 24 and 27 have been adjusted to a higher volume to allow for more range 
https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/usr/local/bin/bluez-udev#L24

closed 2017-03-07 by BaReinhard

nilayp, 2017-03-12:
I can confirm this problem has been fixed with the latest changes.