BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #6: Change default sink in PulseAudio (to combine with snapcast)?

totti2, 2017-03-05:
Hi,

I want to add bluetooth-sources to [Snapcast's](https://github.com/badaix/snapcast) Multi-Room-Audio-Setup.
I'm stuck while defining a new pipe-sink in PulseAudio as suggested [here](https://github.com/badaix/snapcast/blob/master/doc/player_setup.md#pulseaudio).
I added the following to `/etc/pulse/system.pa` :
```
load-module module-pipe-sink file=/tmp/snap_blue sink_name=snapcast
set-default-sink snapcast
```
I now see the new sink, but it's not in use:
```
# pactl list sinks
Sink #0
	State: RUNNING
	Name: alsa_output.0.analog-stereo
	Description: bcm2835 ALSA Analog Stereo
	Driver: module-alsa-card.c
	Sample Specification: s16le 2ch 44100Hz
	Channel Map: front-left,front-right
	Owner Module: 1
	Mute: no
	Volume: front-left: 65537 / 100% / 0.00 dB,   front-right: 65537 / 100% / 0.00 dB
	        balance 0.00
	Base Volume: 56210 /  86% / -4.00 dB
	Monitor Source: alsa_output.0.analog-stereo.monitor
(...)

Sink #1
	State: SUSPENDED
	Name: snapcast
	Description: Unix FIFO sink /tmp/snap_blue
	Driver: module-pipe-sink.c
	Sample Specification: s16le 2ch 44100Hz
	Channel Map: front-left,front-right
	Owner Module: 13
	Mute: no
	Volume: front-left: 32768 /  50% / -18.06 dB,   front-right: 32768 /  50% / -18.06 dB
	        balance 0.00
	Base Volume: 65536 / 100% / 0.00 dB
	Monitor Source: snapcast.monitor
(...)
```
I then tried reordering sink-inputs and sinks according to [this thread](https://askubuntu.com/questions/71863/how-to-change-pulseaudio-sink-with-pacmd-set-default-sink-during-playback)

```
# pactl list sink-inputs
Sink Input #2
	Driver: module-loopback.c
	Owner Module: 15
	Client: n/a
	Sink: 0
	Sample Specification: s16le 2ch 44063Hz
	Channel Map: front-left,front-right
	Format: pcm, format.sample_format = "\"s16le\""  format.rate = "44100"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
	Corked: no
	Mute: no
	Volume: front-left: 65537 / 100% / 0.00 dB,   front-right: 65537 / 100% / 0.00 dB
	        balance 0.00
	Buffer Latency: 250051 usec
	Sink Latency: 98543 usec
	Resample method: speex-fixed-1
	Properties:
		media.role = "abstract"
		module-stream-restore.id = "sink-input-by-media-role:abstract"
		media.name = "Loopback from GT-I9100"
		media.icon_name = "audio-card-bluetooth"
```
But any attempt to change the configuration with `pacmd` fails:
```
# pacmd move-sink-input 2 1
No PulseAudio daemon running, or not running as session daemon.
```
Any ideas? It feels like I'm missing something general here.

off:
The headless pairing of unknown bluetooth devices works like a charm. I have been trying many other approaches with no succes. Thank you very much!

BaReinhard, 2017-03-06:
Try changing the line `; daemonize = no` to yes in `/etc/pulse/daemon.conf` reboot and try using pacmd again to set default sink. I am unsure if this will cause any issues elsewhere but I don't think it will.

BaReinhard, 2017-03-06:
Never mind about the previous comment the daemonize doesn't work the way I thought it did. You might try disabling pulseaudio through systemctl and run it in your rc.local. 

To be honest, I will need to research more into this topic as pulseaudio isn't my strong suit. That being said you might be better off asking a similar question on stackoverflow in the Linux and UNIX branch.

I will continue doing research on this as id like to better my skills in pulseaudio as well.

BaReinhard, 2017-03-06:
Naturally, as soon as I post a reply my next research bout turns up with an answer, try the following to see if it works

```
sudo pkill pulseaudio
pulseaudio -D
```
Then try your previous command with `pacmd`

totti2, 2017-03-06:
```
sudo pkill pulseaudio
pulseaudio -D
```
did the trick - thanks a lot!
If I get that right, the pulseaudio-daemon needs to be started by any user but pulse for `pacmd` to work.

Which config-file is the pulseaudio-daemon (started by the user `pulse`) using?

BaReinhard, 2017-03-06:
I believe it should be using `/etc/pulse/system.pa`

BaReinhard, 2017-03-06:
Ok I believe I have solved this issue, you only need to make on change to the following line in `/usr/local/bin/bluez-udev`:

change `audio_sink=0` to `audio_sink=(number yours new sink is)`

Let me know if this doesn't work for you, as it has worked for me.

closed 2017-03-07 by BaReinhard

BaReinhard, 2017-03-13:
So I have been playing around with pulseaudio a lot more, you can also use `pactl  move-sink-input 2 1`

`pactl` controls a running PulseAudio sound server
`pacmd` reconfigures a PulseAudio sound server during runtime

Since `pulseaudio` isn't running as a daemon you will use `pactl` to move sinks. That being said it is still easier to control via `/usr/local/bin/bluez-udev`.

However, if you are trying to add some other scripts that manage pulse I would suggest using `pactl`

totti2, 2017-12-18:
Changing
>  the following line in `/usr/local/bin/bluez-udev`:
> change `audio_sink=0` to `audio_sink=(number yours new sink is)`
works like a charm. Thank you very much!