BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #7: pulseaudio sink

mich4el-git, 2017-03-06:
could you please add the possibility to use the pi also as an pulseaudio-sink over network?

BaReinhard, 2017-03-06:
could you give me an example of what you are trying to do? To be quite honest, my experience with pulseaudio is lacking and is as the top of my list of learning more about it.

mich4el-git, 2017-03-06:
https://wiki.archlinux.org/index.php/PulseAudio/Examples#PulseAudio_over_network

BaReinhard, 2017-03-06:
For the time being I simply don't have time to implement this in a way that I would see fit.

However, you can see this line:

https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/usr/local/bin/bluez-udev#L32

This line adds the proper control that aligns the source with the correct sink. I am unsure how you might add this, but you may look into adding another sink that is set to output to the network, then adjust the sink number at the top of the file here:

https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/usr/local/bin/bluez-udev#L2

BaReinhard, 2017-03-06:
So do you want the Raspberry Pi setup as a pulseaudio server so you can play music from another linux box to the raspberry pi?

mich4el-git, 2017-03-07:
Yes precisly. But I'm not sure if Pulseaudio also is useable at the same time with bluetooth/airplay

BaReinhard, 2017-03-07:
From what I understand shairport-sync has backend audio support but I don't know pulseaudio is among the supported options. That being said Bluetooth works nicely with pulseaudio. You may choose to open an issue on shairport-sync (the AirPlay program). It's owned by @mikebrady

mikebrady, 2017-03-07:
Thanks for the comment. The support Shairport Sync offers for pulseaudio is very rudimentary – basically it is carried over from the original Shairport programme, and it is surprising that it works at all. It uses the "simple" interface. It would take quite a lot of work to integrate pulseaudio into Shairport Sync, and it's not a priority,. Sorry I can't be more helpful.

BaReinhard, 2017-03-09:
What I have found the best way to create a sink over the network, was to setup a main raspberry Pi running Ubuntu mate, install pulseaudio-dlna, then setup remote pi's running raspbian with https://github.com/hzeller/gmrender-resurrect to set them up as UPnP renderers. Pulseaudio-dlna creates a sink for each UPnP renderer on the network. Then for multiroom audio, you can create a combined sink of whichever sinks you want to use and stream to that combined sink. 

I am currently working on getting this implemented into the install script, but it will be a while as Im pretty sure it won't work right on the first try.

BaReinhard, 2017-03-09:
@mikebrady I know the support for pulseaudio is pretty basic, but can you explain how it works?

In essence, does it simply connect to the default sink and start playing over it?

I am also assuming that the run_before scripts would execute before shairport outputs to the sink?

BaReinhard, 2017-03-12:
@mikebrady I actually got things working. I am running shairport-sync on Ubuntu MATE on a RPI 3, with pulseaudio-dlna. I have shairport-sync ran via `shairport-sync -o alsa -d pulse` and then change the sink-input to the chromecast sink I have on the network. It plays great, no chopping or anything.

Unfortunately, this type of setup requires Ubuntu as pulseaudio-dlna isnt available on Raspbian and I simply dont know enough about how it works to find the proper libraries.

mikebrady, 2017-03-12:
Wow, that's really interesting, thanks.

BaReinhard, 2017-03-13:
@mikebrady well it worked once, now I can't see to get it to show up as a 'pactl sink-input` but when it did work it worked great, then I tried to do the same thing with a a2dp Bluetooth connections and the latency went from a few seconds to about half a minute then the next reboot it wouldn't show up at all.

BaReinhard, 2017-03-29:
I believe this issue has been resolved if not please reopen

closed 2017-03-29 by BaReinhard