BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #9: pulseaudio permissions break after install

aycgit, 2017-03-09:
i installed this and i loved the idea but i want to use this, along with a [mopidy](https://www.mopidy.com/) music server it gives me the error: `pulse pulsesink.c:615:gst_pulseringbuffer_open_device:<autoaudiosink0-actual-sink-pulse> error: Failed to connect: Access denied
`
i can give more logs if needed but does this program change perms for pulseaudio?

aycgit, 2017-03-09:
do you want me to send this to the mopidy team or can you do something about it? it seems that it is just this program removing perms, mopidy worked before installng supersimpleraspberrypiaudioreceiver

BaReinhard, 2017-03-09:
if you could open it with mopidy team as well that would be a good idea. I am unfamiliar with the permissions they set and how it accesses pulseaudio.

Is mopidy set as a system daemon and starts at boot or a program thats run by the user?

aycgit, 2017-03-10:
it is set as started by the user, i got it working with the fix in the issue you referenced but when i tried to play to mopidy after playing from bluetooth it still broke.

BaReinhard, 2017-03-10:
What do you mean by it still broke?

aycgit, 2017-03-10:
it didn't play music but it just looked like i had to rerun the pulseaudio server, weird but it works. You can close this issue now. thanks for helping

BaReinhard, 2017-03-11:
Glad it's working. Does it require you to restart pulseaudio every time after boot?

closed 2017-03-11 by BaReinhard

aycgit, 2017-03-11:
yes it does, but i dont reboot often so it's no problem