BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #10: Beeps on (re)boot, then nothing

ahogen, 2017-03-10:
I have no idea where to look for documentation here, so maybe there is supposed to be an implied answer to this, but I didn't see anything in your README and there's no Wiki, so here's what I'm seeing..

After install, it reboots, outputs a bunch of squares to my UART-based terminal (how I'm interfacing with my Pi right now), I hear a couple beeps from the speaker, it finally gets fully booted to the login prompt and I don't see any bluetooth broadcasting on my phone. I log in, still **no bluetooth broadcast from the Pi**.

I'm using a Raspberry Pi 3 running Rasbian Jesse Light. I have my speaker plugging directly into the on-board audio output jack.

During setup/install I chose the following options...

- (4) Install a Custom Raspberry Pi Audio Receiver
- Airplay = no
- Bluetooth = yes
- Access Point = no
- Kodi = yes
- Infrared = no
- Use a sound card = yes
- UPnP = no
- Which soundcard = 0

What's going on here? Without digging through this whole repository, I don't know what tools and libraries are being used and what part in that chain might be broken. Thoughts?

BaReinhard, 2017-03-10:
Hey ahogen, I am only a one man team at the moment and much of my time is spent at work or with my toddler. That being said in my off time I try to push the project further and a "read the docs" or "wiki" is on the list of things to do.

With all that said there are a few places to look. I would first look at the `install.log` in the `super-simple` directory and make sure that nothing failed during the installation process. Then since no Bluetooth Broadcasting is happening I would first look at the `/etc/bluetooth/main.conf` ensure that under `[General]` there is a line `DiscoverableTimeout = 0` is uncommented.

If you want to simply look at the bluetooth install, take a look at the `bt_pa_install.sh` and `bt_pa_config.sh` files. Anything related to bluetooth and pulse audio is located in there.

I would also ensure the bluetoothd is running.

BaReinhard, 2017-03-11:
I had a similar experience. I tried tracking things down, plugged in an older install then back to the new install, after I plugged back in the new install everything seemed to work.

I am unsure as to why the installation does this. I am still tracking down the issue, but it seems as though if you can reboot another time or two the issue should stop and everything should work as it should.

BaReinhard, 2017-03-11:
If you're still having issues please reopen the issue.

nilayp, 2017-03-11:
I'm having a similar problem and am trying to troubleshoot. When you say "plugged in a newer install" ... I presume you mean you pulled the latest from the repo and reran the installer, right? Can you describe how to do this? If I re-run the installer, I get errors.

BaReinhard, 2017-03-11:
There seems to be an issue where after the install is done and it reboots itself the processes of `hostapd , bluetoothd, and shairport-sync` don't seem to appear. The problem persisted for another reboot or two, I wasn't really paying attention to how many reboots I did.

This was with 2 different sd cards, one was running an older installation prior to the latest pull request. Then newer installation was done after the pull request.

What I was outlining was that with the latest installation run, I wasn't able to see the Bluetooth, AP, or AirPlay name. I shutdown the pi and plugged in another sd card in which I knew was working properly. I looked through all the files in the properly working sd to see if any files were missing or weren't properly patched or moved. After I saw there was one file missing in hostapd I fixed it in the github repo. (I didn't change anything on my first sd card (that was having the issues). I shutdown the Pi and plugged the first sd card back in (the one that didn't have bluetooth, etc showing). After it boot up and just before I went to fix any files, I checked for the Bluetooth and it was there.

This leads me to believe that it needed an additional reboot, why? I don't actually know as it should have worked after the first reboot. This will require some more looking into.

One thing I did find was that the latest installation of `hostapd` from `apt-get` sources no longer uses the file `/etc/default/hostapd` I haven't checked as to why yet, but I have still included it in the repo just incase it is needed.

BaReinhard, 2017-03-29:
I believe this issue has been resolved if not please reopen

closed 2017-03-29 by BaReinhard