BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #11: firstrun.sh requires 'pi' user?

ahogen, 2017-03-10:
This seems kind of obnoxious. I never leave the default 'pi' user around. That is the first thing I remove on a fresh install so maybe this is why other things are breaking for me. 

Is this a common thing throughout scripts other than just the firstrun.sh script I looked at?

Why not use the tilde (~) which will work for any user instead of a direct reference like /home/pi/?

BaReinhard, 2017-03-10:
This is something I am in the process of converting over to. I am trying to make the installation process multi platformed, at the moment looking for ubuntu mate. 

There are a few things similar to this where I initally set things up in a way that made since to me at the time. That being said I still pretty new to debian and still learning a lot.

What I plan on doing is moving `firstrun` and similar scripts in `pyScripts` and `shScripts` to `/usr/local/bin`. This will include files such as `Lircmap.xml` and other files that were initially placed in obnoxious places. 

I apologize for inconvenience

ahogen, 2017-03-10:
Okay that makes sense. 

I forked the repo so I can play around with it and see if I can at least get it installed with my non "pi" user account. ☺

BaReinhard, 2017-03-11:
I am spending tonight fixing this, I may get through all of it, but I may not. Either way, it should be accomplished no later than monday afternoon. 

Thank you for bringing this up, with all the testing and troubleshooting I do, I just always use the pi user and it never occurred to me that others didn't do the same thing.

BaReinhard, 2017-03-11:
the latest pull request should have fixed this issue:

https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/pull/12

closed 2017-03-11 by BaReinhard