BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #13: On RPi Zero W

Let0n, 2017-03-13:
After installation, Pi reboot and error showing

end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179, 2)

Installed using option 4, all yes except for audio card.
[image]

Thanks.

BaReinhard, 2017-03-13:
What OS are you running and was it installed from NOOBS?

Let0n, 2017-03-13:
jessi LITE from NOOBS

that line was
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,2)

BaReinhard, 2017-03-13:
Right now the install isn't working properly on NOOBS installations, from what I'm guessing due to where the root fs is located in NOOBS vs Jessie Lite. You can try using the following text inside the `/boot/cmdline.txt` in place of what is already in there:

```
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty3 loglevel=0 root=/dev/mmcblk0p6 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait logo.nologo quiet splash vt.global_cursor_default=0 consoleblank=0
```

This may fix it, but more than likely you will need to install a new OS, I have a branch that doesn't change the `/boot/cmdline.txt`  and will not cause this error, which can be installed via:

```
git clone -b nocmdline https://github.com/bareinhard/super-simple-raspberry-pi-audio-receiver-install
cd super-simple-raspberry-pi-audio-receiver-install
sudo ./install.sh
```

Let0n, 2017-03-13:
That branch works. Installed without issue.

closed 2017-03-13 by Let0n

BaReinhard, 2017-03-13:
Glad it worked for you. Would you mind posting the contents of your `/boot/cmdline.txt`

I will be trying to implement proper functionality on the master branch.

Let0n, 2017-03-13:
`dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait`

I'm using now as a living room audio receiver (played by TV via HDMI), and the last time I changed installation option to 2 (... Home Installation). All works fine except that the bluetooth connection is always  intermittent.

Also if Kodi is installed and Airplay enabled in the Kodi system, there would be an additional Airplay device named ''Kodi(Raspberrypi)'', which for me only works with iTunes in OS X. I was confused somehow, and then realized the one which I named in installation works fine with OS X and iOS.

Let0n, 2017-03-13:
Oh, and in custom installation (4), a question is misleading:
Do you want to use a Sound Card? (y/n)
if I want sound output, I should enter 'y' and in the sound card list enter '0'
"0. No Sound Card"
otherwise if I enter 'n', there would be no sound output.

BaReinhard, 2017-03-13:
Thanks for the cmdline text, I'll try to implement this tonight. The built in kodi AirPlay seems to be remnants of shairport and not shairport-sync, although I cannot confirm this. It has never worked with any of my devices. 

When you say bt is intermittent what exactly do you mean? There is an issue that if you are using wifi and Bluetooth streaming audio is choppy if it plays at all, this is a bug in raspbian and I haven't seem to have found a fix. 

You are right on with the soundcard implementation,I should change that tonight along with the NOOBS install. I believe if you say no it should default to 0 though. Inlay have to take another look at the code. Thanks for your input!

BaReinhard, 2017-03-14:
So I wasn't able to get the multi os update, but I was able to fix the Sound Card issue.

In the master branch, when you say no to the sound card installation, it will default to choice 0 and use the built-in bcm sound