BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #14: Kernel panic

jbonlinea, 2017-03-14:
Hi there, 

On a freshly installed and upgraded raspbian distribution, I ran the command you mentionned and bumbed twice in the same issue beyond my understanding.

During setup/install I chose the following options...
    (4) Install a Custom Raspberry Pi Audio Receiver
    Airplay = yes
    Bluetooth = yes
    Access Point = no
    Kodi = yes
    Infrared = yes
    Use a sound card = yes
    UPnP = yes
    Which soundcard = 2

I also choose differents names.

and afer a while the Rpi reboots and return : 
[1.052827] --- [end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block (179,2)

and that's it !

Any direction would be welcomed 

Regards

ps: I'm on a Rpi3B

BaReinhard, 2017-03-14:
take a look at issue #13 . If you are running Raspbian Jessie from NOOBs or even possibly the Version with pixel you might have a similar issue:

you can try using this installation:

```
git clone -b nocmdline https://github.com/bareinhard/super-simple-raspberry-pi-audio-receiver-install
cd super-simple-raspberry-pi-audio-receiver-install
sudo ./install.sh
```

When this works, would you mind sending me the contents of your `/boot/cmdline.txt`

jbonlinea, 2017-03-16:
Hi, 
Thank's for your reply, I'm indeed using the last raspbian release with NOOBS

here in the content of `/boot/cmdline.txt`

> dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

The instalation went through, but I can identify weird/buggy behaviours, however I haven't done extensive testing. Here is the first experience I had : 

- after reboot, the Rpi display kodi through HDMI
- Kodi retun an error message on boot (see screeshot)
- I can instal add-on, and streem video from them (tested with Arte+7, a french/german tv chanel)
- at the same time, is I acces to the Rpi via VNC, the Rpi display the main raspibian UI in the VNC window, while still displaying Kodi through HDMI

- from the HDMI dsiplay, when trying to exit Kodi, the first time, the popup window display "exit". If pressed, the main raspbian UI is displayed, it is then identical to the one display though VNC
- when starting kodi again from the main raspbian Ui, kodi starts, and 1) is displayed though HDMI, but not though VNC that still display the raspbian UI ; 2)  when trying to exit, the popup windows display "exit", "shutdown", "custom shutdown", and "reboot" and a) at first Rpi boot, and on every reboot, the "exit" button apears in french, as choosen by myself, however when Kodi is (re)launched from the raspbian UI, the above mention serries of button are displayed in english ; b) when pressing exit, kodi exits, but the HDMI screen is black, while VNC works fine and still display the raspbian main UI, shh still works fine as well. I couldn't find a way for the HDMI to display the raspbian UI again ; c) obviously typing `startx` in ssh retunr errors ; d) I can still restart kodi from the raspbian UI display in VNC, everything is fine but we revert to the same 2) situation ; e) when rebooting the Rpi, the hdmi display properly the reboot process (from shut down to reboot)

- after install, kodi wasn't up to date 15.2 was installer while 16.1 is available.
-  I ran sudo `apt-get update` then sudo `apt-get upgrade ` ; The package r`aspberry-pi-ui-mods` wasn't up to date.
- after update, kodi wasn't updated, and the situation remains as 2), even after reboot.

- I tried this tutorial to update kodi https://mtantawy.com/quick-tip-how-to-update-to-latest-kodi-16-jarvis-on-raspberry-pi/
 this packaged were nstalled libbluray1 libcec3 libdvdnav4 libdvdread4 libpulseeight-platform1 librtmp1 libshairplay0 
but kodi and kodi-bin were "conserved" 
after reboot, Kodi screen appreared through HDMI for half a second, then the raspbian ui was displayed.
Trying to start Kodi fails, black screen on HDMI, raspbian ui on VNC, `ps -e` in ssh retunr a bunch of precess but not Kodi

- audio was working through hdmi, but after update, not anymore, neither from the jack nor from the hifiberry DAC+ ; I mean when played formthe raspbian UI 

- Airplay gets connected, but no audio out, on neither jack nor Hifiberry...

- the hifiberry DAC+ standard do not apears in the list of available sound card in kodi

that's about how far I went... 

Hope something is usefull for you 

cheers

BaReinhard, 2017-03-16:
First, thank you for the contents of your cmdline.txt. Another thing is what OS are you running from NOOBS, is is Pixel?

Well theres a lot to respond on here:

First, Kodi will not show over VNC as its not a X-Windows application,  Kodi has to use a separate GPU framebuffer for its screen.

Additionally, Kodi is run at boot I believe as a daemon and is installed via `apt-get`. The apt-get sources only provide Kodi 15.2 even though 16.1 (Jarvis) is already out. You do have the option of removing kodi via `apt-get remove kodi` and try to compile the new version yourself.

If you are dead set on running the new Kodi, I would suggest you start fresh again and use custom installation and just exclude Kodi, then you can try compiling the new version.

My assumption as far as audio goes I would assume that you have access to a GUI sound control. I would suggest looking at that and check what the output is, its entirely possible that it was muted by default. If that is not the case, in an event that Pixel/Gui of raspbian is making sound it is being output to the pcm sound card, if that is the case it is being used by the GUI and shairport-sync is unable to access it, you can look through `/var/log/syslog` to see if there is any verbosity about a sound card being denied access.

As far as being able to see the sound cards in Kodi (an issue I spent combating for days before I found the fix initially) is that it needs access to pulse and lp groups `sudo usermod -a -G pulse kodi && sudo usermod -a -G lp kodi` if that still doesnt work after a reboot I would also try adding kodi to the audio group if it isnt already `sudo usermod -a -G audio kodi`.

If you added a sound card and want to use it for shairport-sync you will need to edit `/etc/shairport-sync.conf` there is a line under `alsa` called `output_device = "hw:1,0";` (This uses card 1, if you want to use card 0 then use `output_device = "hw:0,0";` 

I believe that should fix most issues, I don't have any experience with VNC myself so it's entirely possible that you can make it work over VNC, but I would suspect that it requires a one off of the project that sets it up as an X application.

jbonlinea, 2017-03-23:
Hi BaReinhardt, 

Thank's for your extensive reply ! 

yes at the moment I'm using raspbian jessy with pixel installed from noobs.

I did some few test/observation/thinking and I would like to ask you few (noobs) question and start again from a fresh install. Some of the above are about your script and it's functionning and may facilitate post-install customization or find walkaround when facing simple issues, some others are more questions and ideas for further development.

1) **Your script** : 
1-1) Have you updated the **master branch** for it to work with _NOOBS raspbian_ yet ?
(In case you did I could give it a try)
1-2) Could I use _your script_ as many times as I want, for instance _doing several custom install_, the first time to set _Airplay_, then to set _bluetooth A2DP_, etc. ?
I would argue Yes, but I have issues understanding the purpose of your `sound_card_install.sh` script (see below), and whether I should re-run it every time (I would say no).

2) Regarding **Airplay**, your script is installing `Shairport-Sync`, ok.
2-1) From what I've quickly seen, _your script_ do not do exactly as  describe in the _Shairport-Sync install guide_ [here](https://github.com/mikebrady/shairport-sync), for instance, `xmltoman`. Why ?
2-2) Actually what matters to me is wether your install works exactly as a _manual install of Shaireport-Sync_ for further configuration and uses (multiroom, changing the audio card, etc.) ?

3) Do **Kodi** install actually also install an _airplay receiver_ ? Or do it use the same but create a kind of _second virtual receiver_ (in an other issue discussed in your repo, I've read that some people end up with two airplay receivers...)
3-1) If yes, is it `Shaireport` ?
Even if  `Shaireport` and `Shairport-Sync` are not conflicting, which I don't know about,
3-3) Doing this _double install_ is mandatory ? I assume _Airplay support_ in Kodi comes from a _plugin_ ?! Can't we tune Kodi install not to also install a _(second) Airplay receiver_ ?
3-4) Can't Kodi simply use the _Airplay receiver_ already available ? I mean can't we use our Rpi as an _Airplay receiver_ when Kodi is running without having this confusing/useless/problem-prone _double install_?
3-5) Couldn't we manually uninstall the _Airplay receiver_ installed by Kodi ? 
3-6) _Kodi black screen on exit_ is a well know issus for raspbian users (I did some few search).
And apparently there are fixes... The fact is that the issue is so common that there are to many post and fixes about this, you never really know which one works or not. It would be excellent if you could include a fixe within your Kodi install script, especially since the fixes seems to be linked to a version of Kodi.
In the meantime, the easiest solution I've found is to type  `ctrl+alt+f8` followed by `ctrl+alt+f7` after exiting kodi to return to the raspbian UI

4) About **soundcard**. 
Well my _hifiberry dac+_ works oob in raspbian. Ok, it's not set as the `default sound card`, but I can use the _audio device setting_ from raspbian UI to change this. And indeed, in case of a _manual install of Shairport-Sync_, this won't be taken into account and I have to edit the `config file` of `Shairport-Sync` to select the sound card I want it to use [see](https://github.com/mikebrady/shairport-sync/issues/493). 
4-1) Why does _your script_ ask wether we want to use a soundcard ? 
As far as I experienced only Shaireport-sync `conf file` have to be edited in order for it to use the `output_device 1` instead of `0`. 
4-2) So, what's the point of updating `/etc/modprobe.d/alsa-base.conf`, `/etc/asound.conf`, `/etc/modprobe.d/alsa-base.conf` and `/boot/config.txt` while the sound card is already working oob? Regarding `Shairport-sync` at least ? More on `blutooth A2DP` below.
4-3) I don't belive it act simply like changing the `default audio device` from the `audio device setting` of the gui ?
4-4) _Your script_ suggest `hifi-berry dac (lite/standard/pro)`, but would it work with the standard `hifi-berry dac+ (lite/standard/pro)` ? 
(if you don't have the hardware to test, I may help you especially if you send me disk image for Rpi-3 that I could simply flash on a spare SDcart to test, I have the Dac+ RCA)
4-5) What about if we have **two extra sound card**, one Rpi hat and one usb ? Will _your script_ change `hw 0` and `card 0` to `hw 1` and `card 1` or `hw 2` and `card 2` depending of our Hat or USB card hardware number ? 

5) **Bluetooth**
5-1) I belive _your script_ is using `pulse audio` to _tunel bluetooth audio_ toward the _soud card_. Right ?
5-2) I thus actually wonder if 4-2) is only usefull for _bluetooth tunelling_ through `pulse audio` ? (if yes I need to read your script with more care to understand why and how).
5-3) I have also seen that the scrips at the origin of yours ([see](http://www.instructables.com/id/Turn-your-Raspberry-Pi-into-a-Portable-Bluetooth-A/?ALLSTEPS) and [see](http://www.instructables.com/id/Raspberry-Pi-BluetoothAirplay-Audio-Receiver-combo/?ALLSTEPS)) change the `device bluetooth class`, however I'm not sure yours do ? In both case, what are the consequences for any uses of the Rpi but _bluetooth receiver_ ? Can I still use a bluetooth mouse or game controler for instance ?
5-3) I've seen that your script set up the Rpi in a manner that _activate/inactivate bluetooth discoverability on connection/disconection_. Well I understand that it might be usefull/practical if the Rpi has only one use, being a _bluetooth receiver_, b [... cut ...]
If there is a way to edit Kodi setting in order to advance audio by 1s as default, then magic happens !
Coult it really be that easy ? I actually belive it could.
7-2) Now what ? VLC has a _function to adjust video audio syncing_, so one problem less [see](http://lifehacker.com/367558/fix-desynchronized-video-and-audio-with-vlc)
7-3) Ok what's left is to adjust audio and video syncink in other apps such as web broswer... I assume this might be very dificult to achieve, but hey, between VLC, Kodi and its bunch of plugins, most usage will already be covered.

Am I that close to live this dream ?

BaReinhard, 2017-03-23:
1. The master branch has not yet been updated ( I have been slammed lately). I assume this fix could be easily done, using a `sed` statement to the `/boot/cmdline.txt`.
The sound card install is rather a "dumb" install, is essentially assumes that the person is using a) no sound card aka card 0 or b) a sound card aka card 1 then enables the dtoverlay in the `/boot/config.txt` . I am guessing that in NOOBs it automatically sniffs what is connected and enables the dtoverlay by default (this is not the case for jessie lite).

2. The shairport-sync install is taken exactly from mike brady's repo, as I have used it for numerous projects in the past and haven't had a single issue  with its playback the way it was outlined. The shairport-sync install is literally a script that takes his install process and automates it, you could do the same things by simplay manually installing his project.

3. It seems that Kodi has a deprecated version of shairport (which works well with ios8 devices for streaming video and audio, shairport-sync may or may not have this functionality, but shairport-sync works well with the newer ios versions and several other itunes programs. You could simply go to kodi settings and disable the airplay functionality to remove the second instance of the AirPlay device. Kodi black screen on exit isn't something I have experienced first hand, but if you're inclined to, you can have at a fix by forking the repo, if it works with the changes you have made I have no problem pulling in a new PR. As far as that issue goes, I assume it is due to using a jessie with pixel install as all I ever use is with jessie lite and unfortunately, my limited time makes me choose other issues that I deem more worthwhile to explore, that being said I am all for having others help out with the code.

4. Sound card support may be something different in Noobs than it is in lite, as I have always had to add a line the `/boot/config.txt` to get the DAC's working. If the card is already working oob there is no need fussing with `/boot/config.txt' or any other file for that matter, aside from may `/etc/asound.conf` for setting a default sound card. Also, using `/etc/modprobe.d/alsa-base.conf` is simply unnecessary and as I see it something more aesthetically pleasing, but others choose to do so. the asound file simply uses alsa plugins i.e. pulse and .!default, not all programs recognize default as a supported plugin so some programs may need another pcm and ctl to work properly, or simply have that program use the pulse device instead.
From what  I recall about hifi-berry dac (lite/standard/pro), dac+is the standard, I don't remember if dac pro has a + with it, if it does then the install just doesn't reflect that and its purely superficial.
I have the dac pro and lite that I have tested and both work with the existing setup. So if you could maybe explain more of what you are talking about I could elaborate a bit more. Again, I mentioned earlier thatn the sound card install is essentially a "dumb" install so it only assumed 1 or 0 sound cards. You would have to manually add new pcm and ctl devices to the `/etc/asound.conf`

5. Bluetooth does use pulseaudio to output audio. 4-2 is more for jessie lite users who don't have the GUI and can't simply click a new output device. As far as how bluetooth works, it uses `/usr/local/bin/bluez-udev` which essentially finds the input stream with a name that looks like a bluetooth name, then grabs the sink input and directs it to the default sink ouput defined in the `/usr/local/bin/bluez-udev` file (defaults to 0). The reason, bluetooth is set to be discoverable and not during connection is that when the device automatically connects to is set to stream output to one sink output, having 2 devices doing this would bind the card , meaning that only one device would have access to it, if you wanted to have multiple devices you would need to completely rehaul that bluez-udev script with a completely different task in mind. I also see that peripherals such as mice are better suited with a usb dongle, but again if you wanted bluetooth mouse you would need to change the manner of how the bluetooth connection works (something I don't plan on setting in this repo, sorry again due to time constraints and the headaches I forsee).

6. Absolutely, bluetooth does have bandwidth issues and as such I have found a much more economical way of having multiroom audio than using RPI's and pulseaudio, even snap cast would be a more expensive route than the current setup I am using. If you are interested in multiroom-audio I suggest you look at my other repo multiroom-audio-ios, if you are purely an android user I would just suggest buying a few chromecast-audio devices ($30 ea) and come with a simple app you can use to set them up. However, if you are an iOS user than the project is a godsend, as you can stream audio via airplay to as many chromecasts as you want and can use the integrated Web UI to control the sinks and volume.

7. as far as video goes, especially when using pulseaudio, is that pulseaudio doesn't have any way to reliably sync audio and set delays. This is why shairport-sync uses an alsa backend and doesn't rely on pulseaudio at all. You can set kodi's audio, but again it falls back to output audio through pulse and gives the aforementioned issue. I suppose it is possible to  find a way for kodi to directly use alsa and your problems are fixed, that being said I haven't been able to, but its been a whiel since I last tried.

It looks as though you are on the correct path, so if you have the desire and know how you could possibly get this to work, but it would take some time.

BaReinhard, 2017-03-29:
I believe this issue has been resolved if not please reopen

closed 2017-03-29 by BaReinhard