BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #15: pHAT DAC on pi zero w reverb on bt

Phill23, 2017-03-14:
Hello,
I am trying to set up a pHAT DAC on a pi zero w.
so far I tried HifiBerry DAC Standard/Pro in the setup with no output.
With HifiBerry DAC Light i get output but with a horrible reverb on it.
pulse audio running on about 80% cpu.
dmesg gives me this:

[    6.158960] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)

[    6.548668] snd-hifiberry-dac soc:sound: pcm5102a-hifi <-> 20203000.i2s mapping ok

[    6.592596] cdc_ether 1-1:2.0 eth0: register 'cdc_ether' at usb-20980000.usb-1, CDC Ethernet Device, c0:56:27:90:7f:19

[    6.595722] usbcore: registered new interface driver cdc_ether

[    7.215424] usbcore: registered new interface driver brcmfmac

**_[    7.399692] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7_**

**_[    7.463521] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code_**

[    8.429048] systemd-journald[110]: Received request to flush runtime journal from PID 1

**_[    9.889526] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code_**

[    9.889579] cfg80211: World regulatory domain updated:

[    9.889591] cfg80211:  DFS Master region: unset

[    9.889602] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)

[    9.889617] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)

[    9.889631] cfg80211:   (2457000 KHz - 2482000 KHz @ 20000 KHz, 92000 KHz AUTO), (N/A, 2000 mBm), (N/A)

[    9.889641] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)

[    9.889653] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)

[    9.889667] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)

[    9.889691] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)

[    9.889702] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)

[    9.889713] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)

**_[   11.035275] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists_**

**_[   11.035304] brcmfmac: brcmf_add_if: ignore IF event_**

[   11.043037] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

[   11.043091] brcmfmac: power management disabled

[   11.910362] uart-pl011 20201000.uart: no DMA platform data

[   13.646433] Adding 102396k swap on /var/swap.  Priority:-1 extents:7 across:163840k SSFS

[   16.401197] Bluetooth: Core ver 2.21

[   16.405192] NET: Registered protocol family 31

[   16.405219] Bluetooth: HCI device and connection manager initialized

[   16.405250] Bluetooth: HCI socket layer initialized

[   16.405272] Bluetooth: L2CAP socket layer initialized

[   16.405327] Bluetooth: SCO socket layer initialized

[   16.416919] Bluetooth: HCI UART driver ver 2.3

[   16.416950] Bluetooth: HCI UART protocol H4 registered

[   16.416962] Bluetooth: HCI UART protocol Three-wire (H5) registered

[   16.417141] Bluetooth: HCI UART protocol BCM registered

[   16.738820] Bluetooth: BNEP (Ethernet Emulation) ver 1.3

[   16.738850] Bluetooth: BNEP filters: protocol multicast

[   16.738883] Bluetooth: BNEP socket layer initialized

**_[   24.471660] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists_**

**_[   24.471689] brcmfmac: brcmf_add_if: ignore IF event_**

[   24.636841] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

[   25.601081] systemd[1]: isc-dhcp-server.service: control process exited, code=exited status=1

[   25.620992] systemd[1]: Unit isc-dhcp-server.service entered failed state.

pimoroni says to set config.txt to `dtoverlay=hifiberry-dac` thats what I did.
[https://learn.pimoroni.com/tutorial/phat/raspberry-pi-phat-dac-install](url)

any idea?
thanks!

BaReinhard, 2017-03-14:
To me it seems as if the dtoverlay may be causing the issue, have you tried audio through alsa and not pulseaudio?

Phill23, 2017-03-14:
Ok, I turned off pulseaudio in /etc/pulse/client.conf and it got worse.

BaReinhard, 2017-03-14:
sorry I should have been a bit more succinct.

```
 sudo aplay -D plughw:1 /usr/share/sounds/alsa/Front_Center.wav
```
I would suggest not messing with pulseaudio just yet, until we can definitively say that pulseaudio is the cause.

If I understand the issue you are saying it could be several different things.

* First, what installation did you use? home version, custom (if so what did you install)

* Check your sink-inputs and make sure that only one sink is being played to the sink output: you can check via:`pactl list sink-inputs` If the above command comes back with two that may be your issue.

* What power supply are you using? 1A, 2A+?

* Lastly, if you are streaming data via WiFi with the Pi while bluetooth is being used there is a known issue here: [raspberrypi/linux#1402](https://github.com/raspberrypi/linux/issues/1402)

Phill23, 2017-03-15:
Well.. I did a clean install an a different sd and everything seems to work!
FWI psu is 2.4A and stable.
Did Car Install. 
BT and WIFI don't interfere.
Selected Hifiberry DAC Light for the pHAT DAC.

On the old install I setup a wifi connection to my home before I installed the repo.

Thanks for the project and your time!

Phill

BaReinhard, 2017-03-15:
I am glad it worked out, it's interesting that it worked. How long ago did you do the previous install?

We're the sd cards of a different class or were they identical?

My guess is that the wifi connection and Bluetooth together were an issue but if you can get your hands on a wlan adapter or use Ethernet it should circumvent the issue

closed 2017-03-15 by BaReinhard

Phill23, 2017-03-15:
The previous install was 1 day older and on an identical sd card.
I did a test on the old install and removed the wifi-settings from wpa_supplicant.conf but that didn't help.

BaReinhard, 2017-03-15:
Did you happen to do a custom install or a predefined one?

Phill23, 2017-03-15:
It was the Car install as well.

BaReinhard, 2017-03-15:
That's what I was thinking, although even the car install shouldn't have an issue as it should be copying the proper files and not patching them. I'll definitely look into that but I've done the same thing without any issues. However, I am glad its working.

Phill23, 2017-03-15:
Tanks a lot!

BaReinhard, 2017-03-15:
No problem, if the same issue happens again please feel free to reopen this isssue