BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #16: controlling system volume

Nerdodactylus, 2017-03-14:
Hey BaReinhard,

I managed to install your SSRPAR and it works great streaming via BT from OSX and also via Airplay! And setup was really like the name was implying :)

My Plan was to use it as a solution for my multiroom audio with airplay, and in that one specific room as a receiver for my Amazon Echo as well. I managed to couple it via Alexa app and she succeeds with coupling process, but there is no audio output from my speakers at all.
Can I even use BT and airplay at the same time, maybe that was the reason?

And one more question...
... how can I control system volume via terminal?

Sorry, maybe a beginners question, but I don't have any clue.
Your project is great, thank you very much!!

best wishes - Nerdodactylus

BaReinhard, 2017-03-14:
Hey no worries at all, to be honest I wouldn't even know how to start with coupling with the Alexa app, so kudos for that! I am glad you enjoyed the install 😄 .

There are a couple issues that may arise when trying to use bluetooth and airplay simultaneously.

* There is a known issue [raspberrypi/linux#1402](https://github.com/raspberrypi/linux/issues/1402) which using built in wifi and bluetooth will cause choppy playback and possibly even prevent audio from playing ( which is what I have found) 
* Second, the way that the project is setup is that shairport-sync and a2dp streaming will be using the same sound card, when one is playing audio the other cannot access the device. You may try looking at the alsa wiki or pulseaudio wiki to see what can be done about it. I remember reading something along the lines of using the `alsa dmix` plugin.

As far as audio controller is concerned, it will depend on what your card says it is but the following should work for bluetooth:

`amixer cset numid=3 {whatever percent you want}%`

You can also set the volume through `alsamixer` you may need to prepend a `sudo` in front of the command.

`amixer set Master {PercentValue}%` or amixer -c {card number} set Master {Percent Value}%`

of course you will need to remove percent value and put an integer value for these.

BaReinhard, 2017-03-14:
I have had great luck with multiroom audio and chromecasts. If you have an android, you likely don' need the Pi, but for iOS the following repo really worked well, but unfortunately I have only gotten it to work with Ubuntu MATE. It only works when you stream via AirPlay as the latency via A2DP is just absurd, its possible when I get some more time I can get that latency under control.

[Multi-Room Audio](https://github.com/bareinhard/Multi-Room-Audio-iOS)

Nerdodactylus, 2017-03-15:
Oh wow, thanks for the quick reply!
The coupling of alexa is really no big deal as she comes with an app to setup her. There you can connect to BT devices, you just have to select the new pi. 
I had used my pi3 before, with volumio as an airplay streamer and also had choppy playback, so I got myself an edimax wlan usb stick with antenna. It solved the problem for me after deactivation of my onboard wlan device.

Btw I forgot to mention - I am using a Hifiberry DAC+ PRO - and could not wait until you tested it, so I just went for a try...

I was able to change the volume with your suggested amixer command, and also tried to change it with alsamixer. I changed the values for every card listed there - but no change in volume was noticeable.

As for the alsa plugin:
I fully understand the theory, but I guess its far above my skill level to get this to work... :)
Maybe if I invest days and dig through pages and get past all of those little problems that will arise on my path.
I tried again and figured that its not a playback problem of simultaneous signals but of alexa and her connection to the pi.
At this time she will connect to the pi, but does not recognise it as a speaker. 
I will do further trial and error as I could find no experience on that even on google.

As an explanation why even to do this:
I am planning to get an echo dot and want it to control my music with her (e.g spotify). The dot has an 3.5mm jack output, but then the internal speaker is disabled - thats bad when you have connected an big amplifier and it is powered off when not in use, because alexa then cant answer you.
If coupled with BT, she still would try to anwer via the speakers/amplifier but I can just say alexa disconnect and then she can use her own speaker again, until I tell her to reconnect.

Again, thanks BaReinhard.
Greetings

Nerdodactylus, 2017-03-15:
Oh and I almost forgot.
As for you chromecast multiroom solution - I already have 3 airport express in use (also for connecting devices to my network over wifi) but will certainly think of chromcasts or even pizeros forn airplay only solution in the future.

BaReinhard, 2017-03-15:
Yeah unfortunately the pi doesn't allow for aux input in stereo only mono and that requires a second sound card with a mic input, give me an hour to get to work and I can link a walk through for that( it's only a $7 soundcard)

I actually tested the DAC+ pro just haven't updated the readme, it worked well for me, even tested the DAC zero from hifiberry that worked for my car setup too, just have to track down a ground loop.

From what I recall for the line input you just have to run the command 

`arecord -D plughw:(mic card number) -f dat | aplay -D plughw:(DAC card number) -f dat& ` given no other Bluetooth/AirPlay connection and it should pipe to the DAC output

Nerdodactylus, 2017-03-16:
Sorry, I think you misunderstood me.
Alexa with BT is the solution, because it can be connected and disconnected via voice.
I cant do that with aux - thats the problem and the reason I like your airplay and BT combined setup.

My problem with the low volume is that my amplifier has auto power of on no input and the input volume is below the threshold, means it powers off with music playing.

BaReinhard, 2017-03-29:
I believe this has been resolved if it has not please reopen

closed 2017-03-29 by BaReinhard