BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #17: No BT or Airplay audio, everything else works

FerventGeek, 2017-03-18:
First- This is the smoothest install of Kodi ever. Super simple indeed, well done.

I suspect there is some sort of routing issue for device audio.  I can pair BT and WiFi connect to AirPlay just fine, and when connected playback audio stops on my phone so it thinks it's going to a BT or AirPlay playback target.  I've looked at all the settings I can find in Kodi for an input source from the Pi but don't see one.  Should it be there?

Thanks again, great project.

iOS 10.2.1, Pi 3, default jack
Linux 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

BaReinhard, 2017-03-18:
First off a couple questions, what Install did you choose?

I would suggest looking at `/etc/asound.conf` and `/etc/shairport-sync.conf` if you don't have a sound card they should be outputting to card 0.

Shairport has a setting further down under alsa as output device or something along those lines. 

You can also check what audio is available via `aplay -l`

FerventGeek, 2017-03-18:
Thanks for the quick reply, details below.

I chose car install, no sound card. I will get a card, but only after I find software. ;-)

Again, Kodi works fine- it plays off the network, USB and menu sounds- and the phone seems to be streaming to the Pi, but the BT stream isn't routing out of the jack. 

```
cat /etc/asound.conf 
pcm.pulse {
    type pulse
    card 0
}
ctl.pulse {
    type pulse
    card 0
}

pcm.!default {
    type hw
    card 0
}
ctl.!default {
    type hw
    card 0
}
```
```
// These are parameters for the "alsa" audio back end, the only back end that supports synchronised audio.
alsa =
{
output_device = "hw:0,0"; // the name of the alsa output device. Use "alsamixer" or "aplay" to find out the names of devices, mixers, etc.
//  mixer_control_name = "PCM"; // the name of the mixer to use to adjust output volume. If not specified, volume in adjusted in software.
//  mixer_device = "default"; // the mixer_device default is whatever the output_device is. Normally you wouldn't have to use this.
//  audio_backend_latency_offset = 0; // Set this offset to compensate for a fixed delay in the audio back end. E.g. if the output device delays by 100 ms, set this to -4410.
//  audio_backend_buffer_desired_length = 6615; // If set too small, buffer underflow occurs on low-powered machines. Too long and the response times with software mixer become annoying.
//  disable_synchronization = "no"; // Set to "yes" to disable synchronization. Default is "no".
//  period_size = <number>; // Use this optional advanced setting to set the alsa period size near to this value
//  use_mmap_if_available = "yes"; // Use this optional advanced setting to control whether MMAP-based output is used to communicate  with the DAC. Default is "yes"
};
```
```
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

BaReinhard, 2017-03-18:
Try running `amixer cset numid=3 1` then try again. You may need to run that as `sudo` but I don't think it's necessary

BaReinhard, 2017-03-18:
You can always try using `sudo raspi-config` in the configurations under audio you should be able to set auto, jack, and hdmi

FerventGeek, 2017-03-18:
Sorry, I wasn't clear.  Audio is working. Kodi can play from plugins and it's menu system is fine.  So again, sound is working on the Pi.  What's not working is Bluetooth or AirPlay audio.  Those are "playing" from the device, but no sound comes from the Pi, while again all other sounds are coming from Kodi.  The raspy-config menu doesn't have a Bluetooth->jack routing config option.

FerventGeek, 2017-03-18:
Update what is odd is sound from PLEX doesn't work either and it did before the setup script ran. Interestingly in the desktop, you also can't manually set the audo Analog/HDMI value.

BaReinhard, 2017-03-18:
My guess is that kodi is using the sound card and AirPlay and Bluetooth can't access the sound card. You can try exiting kodi and try the playback from AirPlay or Bluetooth to see if that allows the playback. Unfortunately there is no mixing as I have found a way to get it to work.

FerventGeek, 2017-03-19:
I got another project working, so I know it's not my Pi. Details here: https://github.com/lukasjapan/bt-speaker. That project doesn't offer authentication which would be a problem in a car, but I may fork it and add basic password authentication in bt_manager/agent.py. It's also helping me understand how BT connection events are handled and what the agent does- the sink has to be opened and mapped to a stream.

BaReinhard, 2017-03-20:
Hey that's how it starts haha. Before this project, Bluetooth and pulseaudio was all magic to me. My guess is that something is interfering. with pixel installed there are several other factors at play, I believe pixel already had bluez and pulseaudio installed. It's entirely possible that there are two instances of pulseaudio running or even bluez. Initially, the project was created before Pixel was even out by another source, I then added multiple multiple other features with Jessie lite in mind as it was meant to be solely used for car audio. It then evolved to a home receiver and kept evolving into the project you see today. I would still say that the project doesn't really support pixel but I have made a branch for it to at least work.

Given that I don't actually use a GUI most the time I haven't ever tried to see if it works. 

That being said I would be more interested to see what your install.log looks like to see if anything failed during the installation that could possibly describe this.

BaReinhard, 2017-03-20:
Also with the new project that is working, is kodi still open while the sound plays?

FerventGeek, 2017-03-21:
Nope  https://github.com/lukasjapan/bt-speaker doesn't use Kodi.  It direct resamples and streams BT to the audio bus.

BaReinhard, 2017-03-21:
Just to be clear, this was installed on a completely new sd card correct? Not with the existing ssrpar install?

FerventGeek, 2017-03-21:
Correct.  Tried Jessie and Jesse Light.  Image, boot, change locale and keyboard, install.

BaReinhard, 2017-05-19:
Sorry for the late response, did you ever get to the bottom of this?

jbonlinea, 2017-09-01:
Hi there,

I tried @BaReinhard's Super-Simple-Raspberry-Pi-Audio-Receiver some month ago on Raspbian Jessy with pixel, and it worked, but let me develop a bit.

First I faced a kernel panic on install, [see](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/14).
Then, as mentionned above, @BaReinhard updated the installer and everything worked fine - awesome.

Yesterday I started from scratch, and had two tries, one with Raspbian Stretch, and then with the last release of Raspbian Jessy, but face the same issue than you. 

On the fresh install of Raspbian,Audio is working and I can switch the output (jack - hdmi - hifiberry dac+).

After running @BaReinhard Super-Simple-Raspberry-Pi-Audio-Receiver, I do not get any audio from pixel anymore... :-( ; but I do from kodi, either run as pi or as root

Would it be possible that one the update between Raspbian Jessy release of march and the latest has brocken something  ? 
It seems surprising, but that's my feedback. 

ps I let apart Stretch as I've seem the changed you mentionned.

BaReinhard, 2017-12-01:
closed due to inactivity.

closed 2017-12-01 by BaReinhard