BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #18: BT issues after one week of constant usage

nilayp, 2017-03-18:
I've installed three RPI3 units with Airplay and BT enabled. The Airplay is working flawlessly on all three units. The BT is having some trouble on all of them.

I'm connecting three different Amazon Echo Dot devices to each RPI3 via BT. Pairing the Dot to the RPI3 is easy and 100% reliable. Once paired, I can play music for some time without any skipping or problems. However, after some interval, music will stop coming from the speakers connected to the RPI3. Sometimes that is 10 minutes, other times 2 hours+. However, the Dot believes it is connected to the RPI3 and continues to play music. While I can't hear it, the Amazon Alexa iOS App shows the Dot is still playing tracks. Also, when I disconnect the Dot from BT, the music continues playing on the Dot's built-in speaker.

When the music stops playing on the speakers, if I disconnect the Dot from BT and then immediately reconnect it again, the music will resume on the speakers connected to the RPI3 until it stops again.

This is happening consistently on all three Dots. Each Dot is connected to it's own RPI3 via BT. Each RPI3 is connected to it's own receiver and speakers.

Does anyone else have this problem with an Amazon Dot or any other BT connected device? Does anyone have any suggestions on how I can debug this?

BaReinhard, 2017-03-18:
Do you have kodi installed along with bt and AirPlay?

nilayp, 2017-03-18:
No. I did a custom install with just AirPlay and Bluetooth.

BaReinhard, 2017-03-18:
I'm assuming that you have rebooted as well. I would look at the syslogs to see if anything shows up there. If I were to guess, I would say it has something to do with the sample rate. It's not defined inside the alsa config. I would look to wiki for proper setting of sample rate for the pulse plugin.

nilayp, 2017-03-18:
Brett - I'll keep my eyes on the syslog. 
To help me understand whats going on under the scenes, can you describe the relationship between bluez, pulseaudio and alsa re: BT connections? I'm new to BT on Linux and haven't groked how these work together.

Also - if you have any interest in an Echo Dot, I'm happy to send you one in exchange for helping me figure out what's going on here. You're welcome to keep it after as a thank you for helping me and championing this awesome project. My email address is on my profile page; drop me a line. 👍

FerventGeek, 2017-03-19:
I got another project working, so I know it's not my Pi.  Details here: lukasjapan/bt-speaker.  That project doesn't offer authentication which would be a problem in a car, but I may fork it and add basic password authentication in bt_manager/agent.py.  It's also helping me understand how BT connection events are handled and what the agent does- the sink has to be opened and mapped to a stream.

Lordeltor, 2017-03-23:
Hi, I have the same issue. Use a PI 3 with internal BT device and a Samsung S6 for playing music over BT. Pairing is no problem and the connection is fine all the time but after 5 to 10 minutes the output from pi stops. After reconnect the phone the sound output is working again for some minutes. I use last raspbian and the actual script.

BaReinhard, 2017-03-24:
Nilayp hey thanks again for that dot, I received it today and immediately began the testing. The good news, it seems to be a sample rate related issue, as disconnecting and reconnecting fixes the issue 100%

```
Mar 24 22:19:34 raspberrypi pulseaudio[615]: [pulseaudio] module-loopback.c: Sample rates too different, not adjusting (44100 vs. 462646).
```

Now that I know the issue I can work at addressing it.

BaReinhard, 2017-03-25:
I also found another issue that seems to cause the dropout as well:

```
Mar 25 02:11:45 raspberrypi pulseaudio[595]: [bluetooth] module-bluez5-device.c: SBC decoding error (-2)
```

It looks as though the SBC issue should be easier to fix, as I can reproduce this issue consistently, the other issue seems to happen as well but less often

nilayp, 2017-03-26:
Brett,

You are welcome! Thanks for troubleshooting this with me. Sorry for the
delayed response, this week and weekend has been nuts. I have two little
ones, 3 and 1 year old which must be kept occupied. :)

Two questions:

1) Can I re-run the installer? When I do, I get errors from the mkdir
commands (easy to fix, of course)... but I worry that some of the
configuration files will be appended to and cause problems. Is there an
easy to way to re-run the installation? I've been literally wiping the SD
card clean, but there must be a better way.

2) Where are you seeing these pulseaudio log messages, in /var/log/syslog?
I see bluez messages in the syslog, but not pulseaudio.

-- Nilay

nilayp, 2017-03-26:
Brett,

I saw the exact same SBC decoding error you did in /var/log/syslog. I did a little research and came across this blog post: https://delx.net.au/blog/2014/01/bluetooth-audio-a2dp-receiver-raspberry-pi where the author patched pulseaudio. (Look for the section labeled, "PulseAudio too picky when decoding SBC") 

Is that worth trying?

-- Nilay

BaReinhard, 2017-03-26:
I know how that goes with kids, I have a daughter just over a year old and she is just so busy all the time.

Unfortunately, rerunning the install isn't possible, it's meant for single use. I'd have to revisit all the code to see if I can get that implemented. For the time being I don't see it being done anytime soon, but who knows when I'll get some time for it.

Searching for errors I run:

cat /var/log/syslog | grep pulse && date

You could also search for bluez errors, but I don't think this is an issue with bluez, as it stays connected. To me it seems more of an issue with pulseaudio.

For the first error I've posted on stackoverflow to see if anyone has seen a similar issue:

Brett Reinhard, BS Kinesiology, Exercise Science, NFPT CPT
Technical Manager
NBC Universal Fitness Center
818.777.4496

BaReinhard, 2017-03-27:
As far as the SBC error, I would think it is definitely worth trying the SBC fix outlined in the above blog post. For the most part, I believe the biggest error that we are running into is the SBC error, as the SBC error can happen at any time interval from 1 minute to 3 hours into streaming.

If that fixes the issue we can make sure its implemented into the project.

nilayp, 2017-03-27:
I did a little bit of poking around on the patch above.
It turns out that it was merged into pulseaudio back in 2014: https://github.com/pulseaudio/pulseaudio/commit/4a5f48e7a42f997793db76e2001b7c252f8d93fe

And was first introduced in pulseaudio version 5.99.1. My RPI is one version back, version 5.0. 
Should I try to upgrade to 5.99.1? Do you think there is any harm in going all the way to the most modern version, 10.0?

BaReinhard, 2017-03-27:
What I would do before updating to the most recent pulseaudio version is backup all the files in `/etc/pulse`

```
mkdir ~/pulsebackup
cp /etc/pulse/* ~/pulsebackup/
```
Then continue with the upgrade, then simply move the backed up files into the new `/etc/pulse` directory.

I am assuming no extremely larger changes have occured to pulseaudio in the sense that they still use `/etc/pulse` and the respective `daemon.conf`, `system.pa` files

nilayp, 2017-03-28:
Okay, thanks for the tips. I probably won't be able to get to this until the weekend ... I'll report back shortly.

nilayp, 2017-04-16:
I finally had time to try to install a later version of pulseaudio. After installed Jessie Lite and SSRPARI, my RPI has pulseaudio 5.0 installed. Since the logs only show the SBC problem, I thought I'd install the pulseaudio release version that fixed this issue, which is [version 5.99.1.](https://github.com/pulseaudio/pulseaudio/releases/tag/v5.99.1)

I'll try newer versions if this one doesn't solve the problem. 

The steps to install are:
`
$ mkdir ~/pulsebackup
$ cp /etc/pulse/* ~/pulsebackup/
$ git clone --branch v5.99.1 https://github.com/pulseaudio/pulseaudio
$ sudo apt-get install intltool
$ sudo apt-get install libsndfile-dev
$ sudo apt-get install libcap-dev (may note be needed)
$ sudo apt-get install libjson0-dev
$ cd pulseaudio
$ ./bootstrap.sh
$ make
$ sudo make install
$ sudo ldconfig
$ sudo cp ~/pulsebackup/* /etc/pulse
`

I've run it for about 2 hours and haven't had any SBC errors yet. I'm going to allow it to run all night and see what happens in the morning. Hopefully everyone else having this issue can try to install v5.99.1 (or later) and report back.

nilayp, 2017-04-16:
Sigh. It didn't work for long. I'm still seeing SBC errors that create the same problem as before. T

```
Apr 16 09:28:57 familyroom-rpi pulseaudio[555]: [bluetooth] module-bluez5-device
.c: SBC decoding error (-3)
Apr 16 09:28:57 familyroom-rpi Getting dbus interface for device: /org/bluez/hci
0/dev_AC_63_BE_F2_78_E6/fd3 interface: org.freedesktop.DBus.Properties property_
name: org.bluez.MediaTransport1
```

and 
```
Apr 16 16:13:01 familyroom-rpi pulseaudio[555]: [bluetooth] module-bluez5-device
.c: SBC decoding error (-2)
Apr 16 16:13:01 familyroom-rpi Getting dbus interface for device: /org/bluez/hci
0/dev_AC_63_BE_F2_78_E6/fd4 interface: org.freedesktop.DBus.Properties property_
name: org.bluez.MediaTransport1
```

I'm going to upgrade pulseaudio to 10.0 and see if that resolves it.

BaReinhard, 2017-04-19:
I have been looking into similar issues, unfortunately I haven't been able to get access to my Pi to start any debugging as of late. However, I am suspecting that the configuration of pulseaudio may be causing some issues and I am hoping by simply running following that it will fix the issue:

```
mv ~/.config/pulse ~/.config/pulse.bad
```

nilayp, 2017-04-22:
Okay, after a lot of trial and experimentation... I found pulseaudio 6.0 works best. It's easy to install, just replace the argument to --branch in the git clone command with "v6.0"

I'm able to play audio for hours... and even keep the Echo Dot connected to BT overnight. I have gotten the SBC error once or twice during the course of the week and have to disconnect/reconnect bluetooth to fix it, but it's now very infrequent.

BaReinhard, 2017-04-22:
This is great! Thank you for the work in debugging this. I tried a few different configs but it kept appearing with the SBC error. I'll implement this in the Install script once I get a chance.

BaReinhard, 2017-04-22:
If you have the desire and/or time to do so, you can create a pull request in the bluetooth section of the installation and add lines you used to get it working properly. I will gladly merge that with the master, if not no problem at all and ill update it in the coming weeks.

BaReinhard, 2017-04-22:
Ok, I have added the fix in the `bt_pa_config.sh`, I will test it tomorrow/tonight. 

This is the fix i have added:

```
mkdir /etc/pulsebackup
cp /etc/pulse/* /etc/pulsebackup/
git clone --branch v6.0 https://github.com/pulseaudio/pulseaudio
apt-get install intltool
apt-get install libsndfile-dev
apt-get install libcap-dev
apt-get install libjson0-dev
cd pulseaudio
./bootstrap.sh
make
make install
ldconfig
cp /etc/pulsebackup/* /etc/pulse
```

closed 2017-05-19 by BaReinhard

armerpunkt, 2017-06-25:
After running the script, I was having the same problem as mentioned in this bug with the "SBC decoding error" and needing to disconnect and reconnect to get audio to work again.  After some investigation, I found that my system was still running pulseaudio 5.0 from /usr/bin, and pulseaudio 6.0 had been installed in /usr/local/bin.  And after running pulseaudio audio from /usr/local/bin with no success, I found that it was not compiled with Bluez enabled.  [I found this page about compiling Pulseaudio on the Pi](http://www.crazy-audio.com/2014/09/pulseaudio-on-the-raspbery-pi/) and installed all the prerequistites it mentioned except bluez-utils and bluez-alsa which no longer exist. And this time when I ran bootstrap.sh, it enabled Bluez.  So, I'm not sure what was needed from there, but it worked.  I then also edited /etc/init.d/pulseaudio to start /usr/local/bin/pulseaudio.

nilayp, 2017-11-27:
I've been really busy this year... sorry for totally dropping off. 

I couldn't ever get my RPIs to work with Bluetooth reliably, so I stopped using them. Thanks to @armerpunkt for uncovering that my previous instructions didn't actually use pulseaudio 6.0. Doh! Reading that comment was truly embarrassing.

Based on that comment and @maximus5684's PR, https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/pull/58, I took the opportunity to make my RPIs useful again. And, I can report 100% success. There have been no dropouts, no SBC errors or any other issues. Both Airplay and BT sound great and have been 100% reliable too. That said - the installation had multiple failures that I document below.

I have several RPI3 units with Hifiberry Digi+ sound cards. I'm not using the onboard WiFi (see issue #4). Also, there seems to be many issues with Stretch and I didn't want to get sucked into those... so I chose the last Jessie release, 2017-07-05. 

Here's my step by step from the beginning:

* Install Jessie with SSH enabled.

* SSH into RPi

* Recreate the SSH keys for security.
```
$ sudo rm /etc/ssh/ssh_host_*
$ sudo dpkg-reconfigure openssh-server
$ sudo service ssh restart
```

* Change the password. Important as SSRPARI installs with the default user, pi. You don't want to be using both the default user and default password!
```
$ passwd
```

* Update/upgrade OS
```
$ sudo apt-get update
$ sudo apt-get upgrade
```
* Install git
```
$ sudo apt-get install git
```

* Install SSRPARI
```
$ git clone https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install.git
$ cd Super-Simple-Raspberry-Pi-Audio-Receiver-Install
```

* Merge in PR #58 to get pulseaudio with bluez to compile correctly. There are clearly some missing parts, as I document below... but starting here is better than what's in the master today.
```
$ git fetch origin pull/58/head:btfix
$ git checkout btfix
```

* Install SSRPARI. I only installed Airplay, BT and a specific sound card. I have no experience with everything else offered in the package.
```
 $ ./install.sh
```

When this completes, Airplay works great with no problems. However, pulseaudio's bootstrap.sh fails due to three missing dependencies. I was able to install these dependencies after the failure.

Error 1 during pulseaudio bootstrap is a weird one. intltoolize is installed with the intltool package, which is specified as a pre-req in bt_pa_config.sh, but it doesn't seem to install. There are no errors in the log, so I don't quite understand what happened there.
```
Activated pre-commit hook.
./bootstrap.sh: line 46: intltoolize: command not found
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.
```

Error 2 during pulseaudio bootstrap is json-c is missing.
```
configure: error: Package requirements ( json-c >= 0.11 ) were not met:

No package 'json-c' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables LIBJSON_CFLAGS
and LIBJSON_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
make: *** No rule to make target 'clean'.  Stop.
```

Error 3 during pulseaudio bootstrap is that sndfile is missing.
```
configure: error: Package requirements ( sndfile >= *0.20 ) were not met:

No package 'sndfile' found
```

So... here are the exact steps I used to fix all these issues and make the system work.

* Installing intltool (again)
```
$ sudo apt-get install intltool
```

* Install json-c, Details are here: https://github.com/json-c/json-c
```
$ cd ~
$ git clone https://github.com/json-c/json-c.git
$ cd json-c
$ sh autogen.sh
$ ./configure 
$ make
$ sudo make install
```

* Installed libsndfile. Details are here: https://github.com/erikd/libsndfile/
```
$ cd ~
$ sudo apt install autoconf autogen automake build-essential libasound2-dev libflac-dev libogg-dev libtool libvorbis-dev pkg-config python
$ git clone git://github.com/erikd/libsndfile.git
$ cd libsndfile
$ ./autogen.sh
$ ./configure --enable-werror
$ make
$ sudo make install
```

* Rerun pulseaudio bootstrap and resume installation. (These commands must run as root since the original install.sh script was run as root.)
```
$ cd ~
$ cd Super-Simple-Raspberry-Pi-Audio-Receiver-Install\pulseaudio
$ sudo ./bootstrap
$ sudo make
$ sudo make install
$ sudo ldconfig
$ sudo cp /etc/pulsebackup/* /etc/pulse
```

* Change pulseaudio to run from /usr/local/bin/pulseaudio, instead of the default installed one in /usr/bin. Pulseaudio daemon is launched by init.d and the configuration file is `/etc/init.d/pulseaudio`. Change the line that says, `DAEMON=/usr/bin/pulseaudio` to `DAEMON=/usr/local/bin/pulseaudio`

* Reboot. Now, both Bluetooth and Airplay work great.

BaReinhard, 2017-11-28:
Hey @nilayp I’ve been away as well. Unfortunately/fortunately I won’t have a job by years end so there’s not been a lot of work to focus on so I’m a bit more active here. I just finished updating my iOS to chromecast project so the next one to update is this!

BaReinhard, 2017-11-28:
Also just wanted to say thank you for the effort and hard work! This is great news!

nilayp, 2017-11-28:
Welcome back to you too... And sorry to hear about your job situation. I hope you find something new quickly. 

Thanks to you too... I feel like I'm finally enjoying listening to music around the house without worrying it's gonna crap out... :)

BaReinhard, 2017-11-28:
Thanks! I have an internship lined up just waiting to hear back officially. It’s taken a while due to holidays.

I’m hoping to this working on the master branch. I also want to simplify the install and remove a few of these other parts. I think I was a bit overly ambitious with everything I originally included. I’ll probably keep some of it in other branches but I think the largest draw of this repo is the Bluetooth and airplay.

So the Bluetooth works solid even with the echo dots? I know that’s where I saw most of the issues.

BaReinhard, 2017-11-29:
Ok I believe I have a fix with the install process.

Anyone happy enough to try it out simply

```
git clone https://github.com/bareinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install
cd Super-Simple-Raspberry-Pi-Audio-Receiver-Install
sudo ./install
```
Follow the prompts and all should work. I am unable to check if airplay works correctly as where I'm at has some restrictions with network related processes. However, the bluetooth worked flawlessly. I only have one issue when trying to patch the /boot/config.txt , but thats fixed easily enough.

BaReinhard, 2017-12-02:
@nilayp I need to give you another thank you for this bt fix. I was playing around with my other repo https://github.com/BaReinhard/iOS-Audio-to-ChromeCast
And can now connect an echo dot to a house full of chromecast audios for synchronized multiroom audio. Still working on implementing on getting it working 100% of the time but it does work! Might be something you’re interested in as well since chromecasts are now cheaper than pi’s.

nilayp, 2017-12-04:
You're welcome! It was definitely a team effort ... and one of my thanksgiving guests shaming me for the music cutting out every so often. ;) It works so well now... the whole family is using it.