BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #20: Gmediarenderer don't start at boot on raspbian lite

jbonlinea, 2017-03-28:
Hi BaReinahrd

I've run your script on a fresh install of raspbian lite with minimal curation (expand file system, update/upgrade, set locales).

Everything went and work as expect-albe/ed, but Gmediarenderer do not start at boot :-(

if I type in a terminal `gmediarendeder` or `gmediarenderer -f VivaUPnP`, Gmediarenderer boots and is visible/usable, so the install itself went well.

After analysing your script, I typed `sudo nano /usr/bin/gmediarenderer`, the file opens whith thousands of crasy signs and lines... absolutly unreadable
But I tried the same for Shairport-sync `sudo nano /usr/bin/shairport-sync` and the file contains similar crazyness while Airplay is working perfectly and starting at boot... so I guess tha't not where the issue comes from. 

I tried to gather complementary info from others sources, [here](http://blog.scphillips.com/posts/2014/05/playing-music-on-a-raspberry-pi-using-upnp-and-dlna-v3/), [here](http://blog.scphillips.com/posts/2013/01/making-gmediarender-start-on-boot/,) and [there](https://www.raspberrypi.org/forums/viewtopic.php?f=65&t=41031)  but haven't solved my problem.

Any direction would be welcomed

Thank's in advance.

PS : I've seen your reply in my other post, thank's, I'll get back to you soon as I'm doing some test with raspbian lite now, but as the issue I'm rising here is pretty precise I belive it might be bette to split it :-)

BaReinhard, 2017-03-28:
I actually found this to be an issue this past weekend as well.

I believe the correct command should be 

```
gmediarender -f "Friendly Name Here"&
```

In the `/etc/rc.local` file you will see that the `gmediarender` line is possibly misspelled as `gmediarenderer` and missing a friendly name.

I would even suggest moving it to the pi's home directory in the `.bashrc` file

```
nano ~/.bashrc

gmediarender -f "Choose a Friendly Name" &

```
This way its run as a user level process and not as root. It may need to be called as root, but I have found its not necessary.

Also, I believe the gmediarender will need some more configuration in actually using it as a UPnP device, it will show up on apps like Kinsky, but theres nothing I can do with it at that point.

If you have any luck please feel free to let me know.

BaReinhard, 2017-03-29:
I believe this issue has been resolved if not please reopen

closed 2017-03-29 by BaReinhard

jbonlinea, 2017-03-29:
Hi there, 

Just to mention/confirm that replacing `gmediarenderer` by `gmediarender` solved the issue and that it can be run as root or as user as explained above by @BaReinhard.
Excellent ! 

---
> Also, I believe the gmediarender will need some more configuration in actually using it as a UPnP device, it will show up on apps like Kinsky, but theres nothing I can do with it at that point.

Use feedback :
- First configuration : UpnP server = _NAS (Synology)_ ; UPnPcontroler = _BubleUPnP (Android)_ ; UPnP renderer = _Gmediarender_. 
It worked like a charm OOB, whatever else may I have done (see below).

- Second configuration : UpnP server = _BubleUPnP (Android)_ ; UPnPcontroler = _BubleUPnP (Android)_ ; UPnP renderer = _Gmediarender_.
Like above

- Third configuration : UpnP server = _NAS (Synology)_ ; UPnPcontroler = _Kinsky (Android)_ ; UPnP renderer = _Gmediarender_.  
I wasn't aware of Kinsky so gave it a try, yes I'm into open source software. Everything worked OOB, only at first. Indeed, once you start messing up a bit, switching from Kinsky to BubbleUPnP was ok, but not the way back, or playing with two controlers on two android devices, Kinsky just lost its mind and the onyl thing it was able to do is play/pause the audio track curently playing. At the end of the test Kinsky wasn't able to detect the _UPnP room_ as it calls it, or _UPnP renderer_ in more usual language of the Rpi. 
Moreover Kinsky is pushing you to try its complementary app Kazoo, and i don't want that...

- Fourth configuration : UpnP server = _Air audio (Android)_ ; UPnPcontroler = _Air audio (Android)_ ; UPnP renderer = _Gmediarender_.
Air audio is an interesting app that enable you, to stream the whole audio of your android (rooted only) device, through various protocol including UPnP.
It works, but have to be handle with care (while it still seems more stable than Kinsky).
I can switch between Air audio et BubbleUPnP, or in other words between configuration 2 and 4 whithout trouble, same for 4 to 1 but switching from 1 to 4 requires to stop the rendering and disconect from the Rpi in Bublue UPnP before heading to Air audio, otherwise Air Audio may not manage to connect or will connect but you won't hear anything and Gmerdiarender may return various erros message such as 
> (gmediarender:2297): GStreamer-CRITICAL **: gst_segment_to_stream_time: assertion 'segment->format == format' failed
or 
> ERROR [2017-03-28 18:53:46.501716 | gstreamer] Failed to get track pos

This to say that UPnP testing/stability/usability has to be considered within the chain _Server->Controller->Renderer_ 

I'm by no may advertisng for BublleUPnP but the fact is that I had great results, and was able to stream FullHD video (including Highly compressed .MTS FullHD from GH1 and EM5 camera, but 100Mbps 4K  video from LX100 failed to play but didn't crashed the system).

Cheers, and thank's for your support !