BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #22: volume-watcher.py not receiving correct values

dustije, 2017-03-29:
I did a custom install (everything except kodi and the infrared remote). I can't control the volume with my smartphone. In /var/log/syslog i can see a message "Detected volume change" (being logged by volume-watcher.py) but no matter if i press volume up or down on my smartphone, the value is always: "127 (100.00)". So the service is started but just receives the same value over and over.
Is this feature working in general (and i did something completely wrong) or can someone confirm this?

Thanks in advance!

BaReinhard, 2017-03-29:
I will need to check this, what device are you using? And can you paste or make a gist of the output from `aplay -l` as well as `sudo pactl list sinks`

Last I checked this was working with my iOS device.

dustije, 2017-03-30:
Sorry i forgott to include my device. I use a Raspberry Pi 3 with onboard bluetooth and an android (version 6.0.1) but changing volume works with "normal" bluetooth speakers.
The output is:
[aplay -l](https://gist.github.com/anonymous/65eec7db3e7ab0f4b3a6740eeb2a2259)
[sudo pactl list sinks](https://gist.github.com/anonymous/947602c0ee335f2b65dc9988c643456d)
Both look ok to me as far as i can tell.
I'm not sure if it's relevant, but i had to use `sudo amixer -c 0 cset numid=3 1` to force audio through the jack output (neither the raspi-config nor `amixer cset numid=3 1` worked for me).
If you need any further information or logs i'd be happy to provide them. Thank you for looking into this!

BaReinhard, 2017-03-30:
Awesome thank you, just need one more thing, can you show me the contents of your `/etc/asound.conf` . Depending what it says I should have a fix for you by today.

dustije, 2017-04-04:
Sorry for my late reply, i had a lot of work to do.
Strange thing is, i don't have a /etc/asound.conf or any file with a similar name. Could that be the problem? Should i create one (i think there is one in an issue here, can i just copy that?) or is the file optional?
Here is my [install.log](https://gist.github.com/anonymous/bef62fbe1f33e36a047d6f5494a5a188) if it helps.

BaReinhard, 2017-04-04:
I would suggest trying the following to see if this actually fixes the issue,

change this line : https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/usr/local/bin/volume-watcher.py#L35

to
```
cmd = "amixer -c 0 cset numid=3 {}%".format(volume_percentage)
```
Then try again with the volume change.

Just to make sure, are you hearing audio output?

dustije, 2017-04-04:
Changing the line did not work.
I can hear sound.
I checked my syslog and it still shows the same value for each entry "Detected volume change: 127 (100.00)" so i don't think it's the script, rather in [line 32](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/usr/local/bin/volume-watcher.py#L32) it reads always the same value.

dustije, 2017-04-04:
I changed log info but i can't find anything except (i have absolutely no clue if this is relevant):
from pulseaudio: "ALSA woke us up to write new data to the device, but there was actually nothing to write!"
and some messages later again from pulseaudio: "Sample rates too different, not adjusting (4410 vs [changing numbers, between ~65'000 and ~462'000])"
and from rsyslogd-2007: action 'action 17' suspended, next retry is ..."

BaReinhard, 2017-04-04:
the log for when debugging is set is `/dev/log` so I would try looking there.

The alsa log is something I have seen multiple times, but haven't ever had it cause any issues.

The rsyslogd-2007 is a known log comment that is abnoxious, and from what I understand has seemingly no issues aside from cluttering your syslog.

Somewhat unrelated, when you see the Sample Rates too Different, not adjusting...
Does the music playback stop? Can you still hear output?

You may have better luck with using this line in place of L35

```
cmd = "amixer -c 0 cset numid=3 1 {}%".format(volume_percentage)
```

budda85, 2017-04-23:
Hi,
I have the same Problem. My volume is always on 100%.
I have a Pi2 with an Hifiberry AMP+, a fresh jessie lite and installed the home Version
I try the two changes but it don't work.

aplay -l
`**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sndrpihifiberry [snd_rpi_hifiberry_amp], device 0: HifiBerry AMP HiFi tas5713-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

pactl list sinks
`Sink 0
	State: SUSPENDED
	Name: alsa_output.platform-soc_sound.analog-stereo
	Description: snd_rpi_hifiberry_amp Analog Stereo
	Driver: module-alsa-card.c
	Sample Specification: s16le 2ch 44100Hz
	Channel Map: front-left,front-right
	Owner Module: 1
	Mute: no
	Volume: front-left: 52429 /  80%,   front-right: 52429 /  80%
	        balance 0.00
	Base Volume: 65536 / 100%
	Monitor Source: alsa_output.platform-soc_sound.analog-stereo.monitor
	Latency: 0 usec, configured 0 usec
	Flags: HARDWARE HW_VOLUME_CTRL LATENCY 
	Properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = ""
		alsa.id = "HifiBerry AMP HiFi tas5713-hifi-0"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "1"
		alsa.card_name = "snd_rpi_hifiberry_amp"
		alsa.long_card_name = "snd_rpi_hifiberry_amp"
		alsa.driver_name = "snd_soc_hifiberry_amp"
		device.bus_path = "platform-soc:sound"
		sysfs.path = "/devices/platform/soc/soc:sound/sound/card1"
		device.string = "hw:1"
		device.buffering.buffer_size = "17640"
		device.buffering.fragment_size = "1764"
		device.access_mode = "serial"
		device.profile.name = "analog-stereo"
		device.profile.description = "Analog Stereo"
		device.description = "snd_rpi_hifiberry_amp Analog Stereo"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card"
	Ports:
		analog-output: Analog Output (priority: 9900)
	Active Port: analog-output
	Formats:
		pcm`

`Sink 1
	State: SUSPENDED
	Name: alsa_output.0.analog-stereo
	Description: bcm2835 ALSA Analog Stereo
	Driver: module-alsa-card.c
	Sample Specification: s16le 2ch 44100Hz
	Channel Map: front-left,front-right
	Owner Module: 2
	Mute: no
	Volume: front-left: 26090 /  40% / -24.00 dB,   front-right: 26090 /  40% / -24.00 dB
	        balance 0.00
	Base Volume: 56210 /  86% / -4.00 dB
	Monitor Source: alsa_output.0.analog-stereo.monitor
	Latency: 0 usec, configured 0 usec
	Flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY 
	Properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "bcm2835 ALSA"
		alsa.id = "bcm2835 ALSA"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "0"
		alsa.card_name = "bcm2835 ALSA"
		alsa.long_card_name = "bcm2835 ALSA"
		device.bus_path = "/devices/virtual/sound/card0"
		sysfs.path = "/devices/virtual/sound/card0"
		device.string = "hw:0"
		device.buffering.buffer_size = "17640"
		device.buffering.fragment_size = "1760"
		device.access_mode = "mmap"
		device.profile.name = "analog-stereo"
		device.profile.description = "Analog Stereo"
		device.description = "bcm2835 ALSA Analog Stereo"
		alsa.mixer_name = "Broadcom Mixer"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card"
	Ports:
		analog-output: Analog Output (priority: 9900)
	Active Port: analog-output
	Formats:
		pcm`

Thanks
Jan

BaReinhard, 2017-05-19:
Have you tried changing the line 35 to the following :

`cmd = "amixer -c 0 cset numid=3 1 {}%".format(volume_percentage)`

BaReinhard, 2017-12-01:
This is likely more due to the device streaming to the PI's bluetooth stack. However, I am closing this due to inactivity.

closed 2017-12-01 by BaReinhard