BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #23: Rebooting while Installing!

DanielKng, 2017-04-03:
Hey, 
while running `sudo ./install.sh` my Pi3 reboots at the following part:

```
===> Executing: sudo cp etc/pulse/daemon.conf /etc/pulse
patching file /boot/config.txt
Hunk #1 FAILED at 54.
1 out of 1 hunk FAILED -- saving rejects to file /boot/config.txt.rej
patching file /etc/udev/rules.d/99-com.rules
Hunk #1 succeeded at 1 with fuzz 1.
===> Executing: sudo chmod 644 /etc/udev/rules.d/99-com.rules
patching file /etc/bluetooth/main.conf
Reversed (or previously applied) patch detected!  Assume -R? [n]
Apply anyway? [n]
Skipping patch.
3 out of 3 hunks ignored -- saving rejects to file /etc/bluetooth/main.conf.rej
patching file /etc/pulse/system.pa
Reversed (or previously applied) patch detected!  Assume -R? [n]
Apply anyway? [n]
Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file /etc/pulse/system.pa.rej
Done! You should reboot now
--------------------------------------------
===> Executing: ./sound_card_install.sh
--------------------------------------------
Ending at @ Mon  3 Apr 20:15:54 UTC 2017

```
Than it just closes the connection and is done. This shouldnt be it, should it?
I used the Custom Installation.

BaReinhard, 2017-04-04:
Hello Daniel,

This is proper functionality of the script as when the script reaches the end it is supposed to reboot for changes to take effect. However, when things fail, this can be quite frustrating so I will be implementing a prompt to reboot instead.

That being said, this should not fail in the manner that the above shows.

A couple of things about this project is that it requires a fresh install of jessie lite, some have also been fine with jessie with pixel. However, that is the extent of support for operating systems. Additionally, the install script is only meant to be run once and may cause issues if you run it more than once.

I need just a few pieces of information to get to the bottom of this:

1. What Operating System are you using? (E.G. Jessie Lite 2017-03-02)
2. Did you make any modifications to the OS before running the install script (E.G. installing something else or running the install script once before)

DanielKng, 2017-04-08:
1. Jessie Lite 2017-03-02
2. No

I still cant see the Bluetooth Device on every Smartphone I own.

BaReinhard, 2017-04-08:
The reason for that is that, the bluetooth patch failed, and it was not setup correctly.

You will have to manually fix them, in the log shown above you will see that each time a patch fails, it saves the file as `*.rej`. You can look at the respective `.rej` file and copy the code into the existing one. Just becareful while doing so, as some files are simply appending something to an existing line, while others are completely removing lines and replacing them.

I am unsure as to why it failed, it is likely that you may have already run the install script, or that you had made changes prior to installing, as I have tried this with the latest Jessie lite and didn't have any errors on the install.

I will check again once I get home, but if you haven't made any extensive changes to your os, I would suggest starting with a fresh install of Jessie lite.

DanielKng, 2017-04-08:
Hey,
i just did a fresh install, no Update, no Upgrade. Pure Jessie Lite.
Still doesnt show up. Where can i find the .rej file?

DanielKng, 2017-04-08:
Complete Log (Warning: LONG!)
```

pi@raspberrypi:~ $ sudo apt-get install git
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  git-man liberror-perl rsync
Suggested packages:
  gettext-base git-daemon-run git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-arch git-cvs git-mediawiki git-svn
The following NEW packages will be installed:
  git git-man liberror-perl rsync
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 4,388 kB of archives.
After this operation, 21.0 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://mirrordirector.raspbian.org/raspbian/ jessie/main liberror-perl all 0.17-1.1 [22.4 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian/ jessie/main git-man all 1:2.1.4-2.1+deb8u2 [1,267 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian/ jessie/main git armhf 1:2.1.4-2.1+deb8u2 [2,725 kB]
Get:4 http://mirrordirector.raspbian.org/raspbian/ jessie/main rsync armhf 3.1.1-3 [373 kB]
Fetched 4,388 kB in 2s (1,888 kB/s)
Selecting previously unselected package liberror-perl.
(Reading database ... 31366 files and directories currently installed.)
Preparing to unpack .../liberror-perl_0.17-1.1_all.deb ...
Unpacking liberror-perl (0.17-1.1) ...
Selecting previously unselected package git-man.
Preparing to unpack .../git-man_1%3a2.1.4-2.1+deb8u2_all.deb ...
Unpacking git-man (1:2.1.4-2.1+deb8u2) ...
Selecting previously unselected package git.
Preparing to unpack .../git_1%3a2.1.4-2.1+deb8u2_armhf.deb ...
Unpacking git (1:2.1.4-2.1+deb8u2) ...
Selecting previously unselected package rsync.
Preparing to unpack .../rsync_3.1.1-3_armhf.deb ...
Unpacking rsync (3.1.1-3) ...
Processing triggers for man-db (2.7.0.2-5) ...
Processing triggers for systemd (215-17+deb8u6) ...
Setting up liberror-perl (0.17-1.1) ...
Setting up git-man (1:2.1.4-2.1+deb8u2) ...
Setting up git (1:2.1.4-2.1+deb8u2) ...
Setting up rsync (3.1.1-3) ...
Processing triggers for systemd (215-17+deb8u6) ...
pi@raspberrypi:~ $ git clone https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install.git
Cloning into 'Super-Simple-Raspberry-Pi-Audio-Receiver-Install'...
remote: Counting objects: 577, done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 577 (delta 1), reused 0 (delta 0), pack-reused 571
Receiving objects: 100% (577/577), 92.62 KiB | 0 bytes/s, done.
Resolving deltas: 100% (297/297), done.
Checking connectivity... done.
pi@raspberrypi:~ $ cd Super-Simple-Raspberry-Pi-Audio-Receiver-Install
pi@raspberrypi:~/Super-Simple-Raspberry-Pi-Audio-Receiver-Install $ sudo ./install.sh
1. Install the Raspberry Pi Audio Receiver Car Installation
2. Install the Raspberry Pi Audio Receiver Home Installation
3. Install the Raspberry Pi Network Without Internet Installation (For teaching!)
4. Install the Volumio (Bluetooth Only) Installation
5. Install a Custom Raspberry Pi Audio Receiver
Which installation would you like to choose? (1/2/3/4/5) : 5
Do you want AirPlay Enabled? (y/n) : n
Do you want Bluetooth A2DP Enabled? (y/n) : y
Do you want to setup as an Access Point? (Necessary for AirPlay, in location without a Wireless Network) (y/n) : n
Do you want Kodi installed? (y/n) : n
Do you want to use infrared remotes? (y/n) : n
Do you want to use a Sound Card? (y/n) : y
Do you want to setup device as a UPnP Renderer? (y/n) : n
Do you want all the Devices to use the same name? (y/n) : y
Device name: AH-DK-191
0. No Sound Card
1. HifiBerry DAC Light
2. HifiBerry DAC Standard/Pro
[... 566 lines cut ...]
===> Executing: sudo cp usr/local/bin/bluez-udev /usr/local/bin
===> Executing: sudo chmod 755 /usr/local/bin/bluez-udev
===> Executing: sudo cp usr/local/bin/simple-agent.autotrust /usr/local/bin
===> Executing: sudo chmod 755 /usr/local/bin/simple-agent.autotrust
===> Executing: sudo cp usr/local/bin/say.sh /usr/local/bin
===> Executing: sudo chmod 755 /usr/local/bin/say.sh
===> Executing: sudo cp usr/local/bin/bluezutils.py /usr/local/bin
===> Executing: sudo cp etc/pulse/daemon.conf /etc/pulse
patching file /boot/config.txt
patching file /etc/udev/rules.d/99-com.rules
Hunk #1 succeeded at 1 with fuzz 1.
===> Executing: sudo chmod 644 /etc/udev/rules.d/99-com.rules
patching file /etc/bluetooth/main.conf
patching file /etc/pulse/system.pa
Done! You should reboot now
--------------------------------------------
===> Executing: ./sound_card_install.sh
--------------------------------------------
Ending at @ Sat  8 Apr 18:26:20 UTC 2017

Broadcast message from pi@raspberrypi on pts/0 (Sat 2017-04-08 18:26:25 UTC):

The system is going down for reboot NOW!

```

DanielKng, 2017-04-08:
Edit:

Okay, now it works. Why - I dont know, i just have let it sit for about
half an hour.  I just need to get my Pimorini PhatDac working. Any idea? :)

BaReinhard, 2017-04-08:
You just need 
`dtoverlay=hifiberry-dac` in your `/boot/config.txt`

BaReinhard, 2017-04-08:
You may also need to edit your `/etc/asound.conf` to use card 1 instead of card 0
Should look like this 

```
pcm.pulse {
    type pulse
    card 1
}
ctl.pulse {
    type pulse
    card 1
}
pcm.!default {
    type hw
    card 1
}
ctl.!default {
    type hw
    card 1
}
```

DanielKng, 2017-04-08:
Ah, nice! It's working very well, thanks.

One last thing: Can I set up my PI so it automatically connects to my Phone
once its booted? Because than I can get in my car, fire it up and don't
have to click on the Pi, since its connecting to my phone on its own. ;)

BaReinhard, 2017-04-08:
Absolutely, you have a couple options outlined here:

http://raspberrypi.stackexchange.com/questions/44520/bluetooth-autoconnect-on-reboot-car-setup-with-scripts

of note I found this one worked for last year:

```
echo "connect AA:BB:CC:DD:EE:FF \nquit" | bluetoothctl
```

closed 2017-05-19 by BaReinhard