BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #24: Renaming the device

sjoenne, 2017-04-13:
Hi!
Is there any way to change the visible name for BT and Airplay (maybe individually?)
Changing it via hciconfig / config files seem to be reverted after reboot.

thanks for a great installer!

sjoenne, 2017-04-22:
(This is mostly for upping the topic, sorry if it is not allowed)

Any way to rename the Name of the Bluetooth?

BaReinhard, 2017-04-22:
Hey sorry I meant to respond to this, not sure how it slipped through the cracks. The install.sh will allow for renaming each specific device during installation. As far as doing it after the fact there's not a simple way to do so, you would just have to go to each specific file and make the changes.

You can go through the bt-pa-config.sh and see where it places the bluetooth name. With airplay/shairport-sync its located in the `/etc/shairport-sync.conf` file which is near the top of the file next to name.

BaReinhard, 2017-04-22:
I believe for the bluetooth name you need to change the file /etc/machine-info and change `PRETTY_HOSTNAME=` to `PRETTY_HOSTNAME=TheNameYouWantHERE`

sjoenne, 2017-04-22:
Thank you! Could not find(not skilled enough yet) that anywhere. But there it was 👍

closed 2017-04-22 by sjoenne

BaReinhard, 2017-04-22:
Cool I'm glad it worked out!