BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #25: No upnp and no sound with kodi

Tagor12, 2017-04-18:
hi i have the problem, that after the installation is finished i didn't get a upnp device.
and with kodi i get upnp working but i can't get anny sound.
I have the newest raspbian and i am using an hifiberry amp.
thanks for your great work and help.

BaReinhard, 2017-04-19:
Hey tagor12, thanks for your interest in the project. Unfortunately, I have no way of checking whether or not UPNP gets any sound output. To be honest, I didn't do any configuration for gmediarender, so its possible it needs something done before it can be used.

That being said you, you may be better off looking for a solution with the respective repo: https://github.com/hzeller/gmrender-resurrect

Or you can use the Kodi built-in one and look at the Kodi forums for fixes. unfortunately, I simply don't have time for the coming weeks to debug this project at the moment.

If you find any info please feel free to post it here and I can fix it on the repo or you can create a pull-request.

wzooff, 2017-04-23:
Hi. Same issue. Looks like binary path is wrong [here](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/gmrender_install.sh#L25) 

Should be [gmrender](https://github.com/hzeller/gmrender-resurrect/blob/master/scripts/init.d/gmediarenderer#L34). May be mistake

BaReinhard, 2017-05-19:
Merged the pull request, this issue should be fixed now

closed 2017-05-19 by BaReinhard