BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #28: Bluetooth audio hanging/lagging

sk22, 2017-04-26:
Hi!

I am able to connect, but the sound transmitted hangs every second or so. I was able to connect without having this issue once, I don't why that was.
I don't know which part of the setup causes this problem, so I hope you/someone will be able to help me with this issue.

Thanks!

sk22, 2017-04-26:
Seems to be related to WiFi. https://github.com/raspberrypi/linux/issues/1402
I completely disabled it using the Raspi's config.txt, now it works just fine!
Would be neat to know though if there's any workaround. :)

BaReinhard, 2017-04-27:
Unfortunately, since it is a bug with the drivers that the raspberry pi foundation uses for the BCM wifi and bluetooth chip. There is no current work around software-wise aside from disabling the wlan adapter. You can however, use an external wifi dongle and the problem ceases to occur.
There is a bug fix underway from the manufacturers of the chip but using a different driver than the current one used by the RPI Foundation. They have stated that they will not have a fix until the other driver fix is out and stable. This is unfortunately not been given a time frame from what I understand.

sk22, 2017-04-27:
Okay, thank yoi a lot!

closed 2017-05-19 by BaReinhard