BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #29: A2DP Volumio

Benny187, 2017-04-28:
Hallo,

i have a Raspberry B+ with Bluetooth Dongle with no Soundcard ( AudioJack).
With Jessie Lite + Bluetooth Install (4) / works fine with no Problems.
With Volumio 2,141 + Bluetooth Install (4) / i can not find Bluetooth.
With my Raspberry 3 its the same Problem.

Can you help me ?

Sry for my bad English.

Benny

jdforest, 2017-05-02:
I have the same problem on my Pi3b with Volumio 2.141. I just tried again with a fresh install and still had no luck.

Benny187, 2017-05-03:
I have it.

1. you need a user "Pi"
2. then apt-get install patch
3. then Script with Volumio (4)

Greets

EnerGeh, 2017-05-15:
I've got the same problem with Volumio 2.163. I've got several errors during the installation, I think the main issue is that I'm actually unable to upgrade systemd:
```
Unpacking systemd (215-17+deb8u7) over (215-17+deb8u6) ...
dpkg: error processing archive /var/cache/apt/archives/systemd_215-17+deb8u7_armhf.deb (--unpack):
 trying to overwrite '/usr/share/dbus-1/system-services/org.freedesktop.systemd1.service', which is also in package systemd-shim 9-1
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
addgroup: The group `systemd-journal' already exists as a system group. Exiting.
Processing triggers for dbus (1.8.22-0+deb8u1) ...
Errors were encountered while processing:
 /var/cache/apt/archives/systemd_215-17+deb8u7_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Had no chance to look further into that, perhaps it will help s.o. to locate the main issue?!

BaReinhard, 2017-05-19:
@Benny187 So have you got a working setup using volumio and this project now?

I don't have any experience with volumio, so I don't know that I can be much help on that end of things, but if you need help on the ssrpari side of things I should be able to lend a hand

EnerGeh, 2017-05-19:
As I'm very new to Volumio and according to my previous post,

> I think the main issue is that I'm actually unable to upgrade systemd

I've done some research. I found out, that `apt-get upgrade` is not supported on Volumio 1 as it is supposed to be an appliance-like software package ([https://volumio.org/forum/sudo-apt-get-upgrade-t2240.html]). I couldn't find any information about apt on Volumio 2, but seems to be designed with a similar appliance approach. So I think there are two ways to adress the issue:
1.  Run the scripts without upgrading installed packages (could be difficult with dependencies). I actually don't have the time to play with that, but could be useful for some of you with more free time. :)
2.  Try to implement that stuff as an Volumio plugin as this is the recommend way to extend Volumio's feature set. I actually have neither the time nor the skills to test that, but could be helpful for some of you. :)
I'll keep you posted whet I've had the time play a bit more.

fmiratsch, 2017-06-12:
@Benny187 
I have the same issue with volumio2. what are exatly the steps you did to get it working?

BaReinhard, 2018-01-11:
I had a branch for volumio, it has since been merged with master. You can give it a shot by redownloading it, but depending when you initially downloaded the repo it may not work as it doesn’t have the stock files. This has since been fixed with recent merges but isn’t backwards compatible. You may also need to start from a stock  volumio image.

closed 2018-01-11 by BaReinhard