BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #30: Uninstallation?

hotwheels927, 2017-04-28:
Hello All,

I seem to be having some issues with the uninstallation of the software.
I keep running into errors during boot.

Is there a script that I can run to uninstall all of the changes made during the installation process?
(I suppose the libraries do not need to be uninstalled, but I would like for the software to not run automatically on boot.)

Thanks.

BaReinhard, 2017-05-19:
Unfortunately, there is no script for that.

What specific errors are you getting on boot?

It should be simple enough to disable some of the processes using systemctl, you just have to choose which ones you want to stop.

A sample of how to manage this is:

```
sudo systemctl disable shairport-sync
```

Instead of shairport-sync simply choose whatever process you want to stop starting on boot (given that they are started by a service file)

BaReinhard, 2017-05-31:
I want to revisit this, I am considering restructuring the project so that it is installed all into a local folder so that it makes installation a bit easier. This would take some time to implement but ultimately would make life easier for those who don't just use their pi as an audio player and use it for other things that require saving of more data on their cards

BaReinhard, 2018-01-11:
Uninstallstion is included in recent merges. Some of it still needs testing but a full uninstall has been shown to work for Jessie Lite versions.

closed 2018-01-11 by BaReinhard

RottenRonnie, 2018-03-23:
I cannot find the uninstallation script nor documentation.

BaReinhard, 2018-03-24:
documentation is lacking, but you should be able to uninstall via:

```
source functions.sh
source dependencies.sh
full_uninstall
```

BaReinhard, 2018-03-24:
See https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/functions.sh#L164

for all uninstall options

RottenRonnie, 2018-03-26:
Thank you. I downloaded an earlier version and had to uninstall some of the items manually as I was desperate for resources and this was on the expendable list. 

I worked well once I purchased a DAC for better sound quality. 

I'd suggest taking a minute and editing README.md with the uninstall instructions.