BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #32: multiple airplay instances for multi-room zones?

cre8dojo, 2017-04-29:
can you have multiple airplay instances running each outputting to different audio out devices (usb sound cards) and allow multiple users to stream to different 'zoom' at the same time?

BaReinhard, 2017-04-29:
This would be a better question for shairport-sync. I am unaware of this possibility. I guessing no based on the port utilization, but I don't have enough background information to give a good answer.

cre8dojo, 2017-04-29:
Ah, your install uses shairport-sync then for the airplay part, correct? I'm currently using shairport-sync in the way that I asked about. I thought your software was an alternative to shairport-sync. 

So, you've got bluetooth and shairport-sync connection options on the same rpi. Can the bluetooth connection stream to the same output as shairport-sync either one at a time or at same time?

BaReinhard, 2017-05-19:
Sorry for the late response, 
Yes, my installation uses shairport-sync as the program to control AirPlay connections.
In theory I would imagine that if you had multiple shairport-sync installations that you could indeed to choose to output to multiple sound cards, that being said I don't know that I would do this, at the very least I would be mindful of what type of interpolation was used if you did use this.

This project in particular is using pulseaudio and shairport-sync, pulseaudio to handle the playing of bluetooth connected devices, and shairport-sync to handle the playing of AirPlay connected devices. By default they both use the same output source. 

Early on I tried to get some sort of ALSA mixer setup so that there would be no lag in between changing of sources (i.e. stop playing from AirPlay device, and begin playing from a bluetooth connected device), unfortunately, I wasn't able to find a suitable fix for this and chose that I could live with this delay as during normal use I don't typically switch input sources often if ever at all.

So in short, you must choose one input source bluetooth or airplay to play over any given output.

closed 2018-01-11 by BaReinhard