BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #33: No analog audio

taskmasterjosh, 2017-05-01:
Hi,
no matter how i install this, when i try to use analog audio output it wont work. it will only output audio through HDMI, when i try to use **amixer cset numid3 1** so it will output to analog i get an error, 
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied  

amixer: Control default open error: Connection refused

as i am trying to use this with speakers with analog input any help fixing this would be great. 
Thanks.

BaReinhard, 2017-05-19:
Hey taskmasterjosh, sorry I haven't been very active in this repo for some time. A lot has been on my plate lately, sorry no one has jumped in to help get this solved.

Based on the error message to me is looks like you may not have the permissions to run this command, have you tried prepending sudo to this command?

```
sudo amixer cset numid3 1
```

ways, 2017-08-01:
Might not help you, but for me it worked to run.$ sudo raspi-config and select Advanced -> Audio -> force 3,5 mm jack.

closed 2018-01-11 by BaReinhard