BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #34: Snapcast support

HVPereira, 2017-05-05:
I've been thinking of using this project with pulse audio and as a snapcast server did any of you tried this? How can I achieve it?

BaReinhard, 2017-05-19:
I haven't ever tried using this with snapcast, but if you do get it work please feel free to update us with the information.

I chose a different route for synchronous multiroom audio, you can see it here : 
https://github.com/BaReinhard/Multiroom-Audio-iOS

I preferred this setup as it works well with iOS devices in particular, unfortunately, the more components you add to an audio setup the longer the delay becomes.

I solely use iOS devices for my music playback so this worked very well for me. At the moment, I am still in the process of adding a similar autoconnect feature for bluetooth for the Multiroom Audio iOS project, once this is done, it will essentially function as this current repo, but with the sole purpose of performing multiroom audio.

BaReinhard, 2017-12-01:
If it is possible I would not use the install script itself, but go line by line and execute the source for the bt_* files. I would also omit any commands that alter `/etc/asound.conf` and omit any that alter `/boot/config.txt`

Even with that it's not guaranteed as I've never used snapcast. 

Alternatively,
If you could explain to me how snapcast receives its audio to play over the server, I'm sure I could help make a simple snapcast branch.

cmfcmf, 2017-12-03:
Jumping on here: Snapcast receives its audio through the `/tmp/snapfifo` pipe. Anything piped into there is picked up by the snapserver.
> Snapcast can be used with a number of different audio players and servers, and so it can be integrated into your favorite audio-player solution and make it synced-multiroom capable. The only requirement is that the player's audio can be redirected into the Snapserver's fifo /tmp/snapfifo.

https://github.com/badaix/snapcast#setup-of-audio-playersserver

BaReinhard, 2017-12-03:
I did a brief look and for AirPlay it should be simple, change this line in the `/etc/shairport-sync.conf` file to the pipe https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/airplay_config.sh#L94

I will need to investigate for Bluetooth a bit more but I highly suspect it is possible.

BaReinhard, 2017-12-03:
According to the snapcast docs it should be easy enough to add pulseaudio too

```
pactl load-module module-pipe-sink file=/tmp/snapfifo sink_name=Snapcast
pactl update-sink-proplist Snapcast device.description=Snapcast
```
This could be added to the `/usr/local/bin/pa_fix.sh` file and it should show up as a sink on the web ui.

I don’t have snapcast so I wouldn’t be able to test effectively but that is how I would start.

BaReinhard, 2017-12-03:
Got this confused with my other repo, there is no web ui for this project and no `pa_fix.sh` lol sorry. The commands above can be apart of any root level start script, given it’s loaded after pulseaudio, this could possibly be don’t in the `/etc/pulse/daemon.conf` but some testing would be needed.  you may need to add `pactl set-default-sink Snapcast` or whatever the sinks name is, this should also be added to the same start script.

BaReinhard, 2017-12-04:
I started working on getting this implemented because I really like the simplicity of it and it opens a lot of doors for multiroom support. Currently I am creating a separate script that will automatically compile and install snapcast as a server, client, or both on top of the sspari installation. This way you can have multiple rpis in different rooms, stream to them individually via Bluetooth or airplay, or have them apart of a snapcast server and steam to the hub which will output to all devices on the server. I still need to implement some logic to decide what is what but I already have airplay and Bluetooth able to output to the `/tmp/snapfifo` pipe. If all goes well this will be implemented before years end. If all goes according to my plans it will be done before weeks end.

nilayp, 2017-12-04:
Brett - this all sounds awesome!! I'm totally in for testing... so let me know how I can help.

HVPereira, 2017-12-04:
Me 2 I will soon start to pick this up again.

BaReinhard, 2017-12-04:
I have a `snap-addition` branch that I am currently in the middle of testing. It asks if you’d like to install a client, server or both, but is only incorporated in a custom install option. I will be implementing a Bluetooth, airplay and snap install later today. I haven’t included a config for snap, or airplay, but pulseaudio has a sink setup for /tmp/snapfifo. 

Aside from that I don’t actually know how to use snapcast and get things streaming. So I need to research that a bit more.

I’m thinking a web ui will be necessary to make things simpler. Ideally it would be nice to use snap clients as apart of the multiroom setup and individually as audio receivers so that logic is in the works as well. 

If anyone has ideas I’m happy to listen.

Also anyone with insight on configuring snapcast please reach out.

totti2, 2017-12-17:
i could get this running with the prebuilt packages from [badaix's](https://github.com/badaix/snapcast/releases) site. but I have to reassign bluetooth-sink-input to sinks after every reboot.
this is what I did
```
#   2017-07-05-raspbian-jessie-lite.img, RPI3, no bluetooth-dongle

1  sudo raspi-config                                   #   timezone, memorysplit, expand filesystem
2  sudo apt-get install git 
3  sudo apt-get update
4  sudo apt-get install git 
5  git clone https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install.git
6  cd Super-Simple-Raspberry-Pi-Audio-Receiver-Install/
7  sudo ./install.sh

    8  nano sound_card_install.sh                           #   missing "
    9  sudo ./sound_card_install.sh                         #   installation fails
   43  sudo mv /etc/asound.conf /etc/asound.conf_bkp        #   reset alsasound from 3.5mm jack
   44  sudo reboot                                          #   sound from 3.5mm jack

#   install Snapcast (downloaded before)
   57  sudo dpkg -i snapserver_0.12.0_armhf.deb

#   correct dependencies
   58  sudo apt-get -f install

#   create Snapcast-Stream = new pulseaudio-sink
#   SNAPSERVER_OPTS="-d -b 250 --sampleformat 44100:16:2 -s pipe:///tmp/snap_blue?name=Bluetooth&mode=read"
   61  sudo nano /etc/default/snapserver
   62  sudo reboot

#   create systemwide pulseaudio-pipe-sink
78  sudo nano system.pa

#   load-module module-pipe-sink file=/tmp/snap_blue sink_name=bluetooth
79  sudo reboot

80  sudo pactl list sinks                                #   list pulseaudio sinks
81  sudo pactl list sink-inputs                        #   list pulseaudio sink-input
82  sudo pactl move-sink-input 0 1                #   combine sink-input # with sink #

83  sudo dpkg -i snapclient_0.12.0_armhf.deb             
85  snapclient --list                                    #   list soundcards
86  sudo nano /etc/default/snapclient                    #   configure snapclient
#   SNAPCLIENT_OPTS="-h multipi.local -s 25 -d"          #   host soundcard daemonize
87  sudo systemctl restart snapclient.service    ```

totti2, 2017-12-17:
I just realized, that the index/number of the sink-input is increasing by 1 every time a new device connects

plus:
"uninstalling" the onboard sound by changing one line in 
```
/boot/config.txt
...
dtparam=audio=off
```
didn't help, since pairing stops working

totti2, 2017-12-18:
I put my progress [here](https://gist.github.com/totti2/41ed90feb1eb5838cc6a789d2b2dd5a7)
Bluetooth and spotify are working.

Regarding the reassignment of sink-inputs, I finally remembered [this](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/6#issuecomment-284573272)

Didn't try, should work:
When deactivating the onboard soundcard in `/boot/config.txt` one should choose "no soundcard"  in the SSRPARI install routine, since there's only one sound card available in `aplay -l` with sink index `0`

BaReinhard, 2017-12-18:
Looks good, I am assuming autopairing and connecting are working on bluetooth for you?

totti2, 2017-12-18:
Yes.
I will prepare AirPlay and try it this week.

I even omitted the 250msec buffer in `snapserver`-config-file. 

I'll install a Pi0 as Client. Will report back, probably after christmas

BaReinhard, 2017-12-21:
@totti2 I’ve been toying with snapcast and seem to have been able to get Bluetooth to output over /tmp/snap_blue and have it play to the snapclient all on the same board at once.

I’m wondering if you could give me some insight in installing another board simply as a client and getting it to play on that as well.

What is needed on the other board? Simply a snapcast client and it should play automatically? I tried installing just a client and used your config but couldn’t get any audio output. Am I missing something?

totti2, 2017-12-22:
Yes, `snapclient` is all you need.
Check the config-file for the `-h` (host) argument and just replace `localhost` with whatever you call your snapserver or its IP-address.

totti2, 2017-12-22:
You also need an Android device and the Snapcadt-App to assign snapserver-streams to the clients.

BaReinhard, 2017-12-22:
I finally got it working. Had to read through the docs of snapcast. Thanks for the follow up!

BaReinhard, 2017-12-22:
I also went a slightly different route for airplay and changed the systemd service file to output to a the pipe. This made it so I didn’t need an android device to set the streams, when I connect through Bluetooth it streams to the /tmp/snap_blue and when connect through airplay it does the same. It makes sense why I couldn’t get airplay working without th android device.

totti2, 2017-12-22:
If I got you right, any of your clients are playing the same stream?

With the Android app you can assign different streams to different clients. This made sense on my last birthday-party with bluetooth and rtp/pulseaudio sources streaming to different rooms/clients.

Maybe, with the help of the web-ui iPhone-users can use full functionality.

Am 22. Dezember 2017 16:30:31 MEZ schrieb Brett Reinhard <notifications@github.com>:
> I also went a slightly different route for airplay and changed the
> systemd service file to output to a the pipe. This made it so I didn’t
> need an android device to set the streams, when I connect [...]

BaReinhard, 2017-12-22:
For my setup all clients play the same stream. If I’m understanding you correctly. I could stream Bluetooth and airplay to the same device and choose which clients get each stream? Or would I need multiple snapservers for that?

totti2, 2017-12-23:
You need only one server for that. Have a closer look at the screenshot in the [snapcast docs](https://github.com/badaix/snapcast#control).

BaReinhard, 2017-12-23:
Wow I just realized that. Awesome! I’ll have to look into that more.

closed 2018-01-11 by BaReinhard