BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #35: Play Audio with Chromium after I

Basti215, 2017-05-18:


BaReinhard, 2017-05-18:
The functionality should be available. Did you happen to install a desktop as well? If so I would look at what it says in the sound options. Also how are you playing sound? Sound Card, builtin, or soundcard hat?

Basti215, 2017-05-19:
Yes on my Pi is Raspbian Jessie with Pixel installed.
I tried with (Hifiberry DAC+) and without a soundcard.
Which Sound options do you mean?

BaReinhard, 2017-05-19:
In the GUI of the desktop, there should be some sound configuration in the dropdown at the top left corner. I don't remember exactly where its located as I don't usually use the GUI with the Raspberry Pi.

What you can try doing in the terminal is list the sound cards available,

```
aplay -l
```
http://raspi.tv/wp-content/uploads/2016/10/raspi-config-300x295.png

But instead click on audio device settings and see what the default sound output is, from there you should be able to choose input and output.

What also may be happening is that a device is still connected and hasn't released the sound device for other inputs to output sound over.

Basti215, 2017-05-19:
In the audio device settings everything is on and in the Raspberry Pi configuration i don't find any Option for Sound/Audio preferences.

aplay -l says:
`**** Liste der Hardware-Geräte (Playback) ****
Karte 0: sndrpihifiberry [snd_rpi_hifiberry_dacplus], Gerät 0: HiFiBerry DAC+ HiFi pcm512x-hifi-0 []
 Sub-Geräte: 1/1
 Sub-Gerät #0: subdevice #0`

I have disconnected my Bluetooth-Device and has removed it from the Raspberry Pi.
Still no sound when I play something with Chromium.

armerpunkt, 2017-06-25:
I was having this same problem (mpd wouldn't work either).  Switching from Pulseaudio 5.0 to Pulseaudio 6.0 fixed it.

Basti215, 2017-06-25:
Okay cool thanks
How can I switch the version of Pulseaudio? 
Is there an update function?

armerpunkt, 2017-06-25:
If you still have the Super-Simple-Raspberry-Pi-Audio-Receiver-Install folder, there should be a pulseaudio directory in there which contains version 6.0.  It was downloaded and compiled automatically by the script.  But on my system, it was installed in /usr/local/bin, and the system kept using the one from /usr/bin which was version 5.  So if you run `ps ax | grep pulse` it will show you where pulseaudio is running from.  Then run that command using the full path and add "--version" to check if it's 5.0 or 6.0.  If it's 5.0, check if you have pulseaudio in /usr/local/bin and if so check what version it is.

(One other problem I had was that 6.0 was not originally compiled with BlueZ support, so I had to install some more things and recompile it, but hopefully that was just me)

jbonlinea, 2017-09-01:
Hi there,

On a fresh install of Raspbian Jessy (latest release updated), I can't get any sound from pixel once I ran @BaReinhard script, while I do get audio in Kodi (run as pi since launcher from pixel).

It appears that : 
- pulseaudio is indeed run from `/usr/bin` and is in version 5 and not 6
- I do have a pulseaudio folder in the Super-Simple-Raspberry-Pi-Audio-Receiver-Install folder
- there is no pulseaudio folder/version or whatsoever in `/usr/local/bin/`

Am I suppose to install manually pulseaudio 6 from the Super-Simple-Raspberry-Pi-Audio-Receiver-Install folder ? 
how do I do that ?
And how do I define that this is the version I want to be ran on boot ?

Thank's

BaReinhard, 2017-09-07:
from my testing and understanding I haven't been able to get this to work on Raspbian with Pixel, and unfortunately do not have the time to make changes to the project to incorporate this at this time.

BaReinhard, 2018-01-11:
Recent merges should allow for this functionality on pixel now. There is also another issue
#102 which should also work for stretch but I haven’t tested on pixel just yet.

closed 2018-01-11 by BaReinhard