BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #36: I have a RPI Z W + PHATDAC - Not Working

spin1974, 2017-05-29:
Hey guys,

I have a Raspberry Pi Zero W with the PhatDAC from Pimoroni.  I've been trying for a few weeks to get this working with no luck.  I've tried a bunch of different combinations and re-imaged with Jessie Lite many times to start from Scratch.

Anyone want to collaborate on this with me?  I'm more then happy to share results/success with this GIT to get it working as I imagine this would be a really great platform (RPIZw+PHATDAC) for this install.  

As a background, I can get sound coming out of the speakers using the Pimoroni install method and OMXPLAYER, so the soldering worked fine. 

I have a few issues:

1.  Can't hear anything out of the speakers when doing AIRPLAY.
2.  Can't get Bluetooth to stay connected, I'll connect to it, and then it will disconnect quickly.  
3.  If I can get Bluetooth to stay connected, I can't get sound to come out.

And in terms of my skill set, pretty low, don't expect too too much from me!  I've been on this planet for a number of years so my problem solving is pretty good, but I've spent 20 years in the IT industry and for most of it have been a "Windows guy"!

Thanks,

Chris

spin1974, 2017-05-30:
Hey peoples,

I think I solved this.

If you have a Pimoroni Phatdac (https://shop.pimoroni.com/products/phat-dac) and an RPI Zero W, then I have found that this is the sequence you have to follow to get Airplay and Bluetooth working properly.  Didn't test the other features such as Kodi or the uPNP.

1.  Installed Jessie, I used PiBakery to build the image for me, so it was Jessie Pixel that was installed and not the Lite version, don't know if that matters.
2.  I installed git.
3.  Used 
[image] twice, the first to install and then the second to test to make sure that you could hear the front right...front left...front right...front left, which was successfull.
4.  Did the git copy as per the instructions.
5.  Did a manual install (option 5) and selected airplay, bluetooth and NO sound card.
6.  Let it run for a while and it worked perfectly.

This gives you the ability to play music over bluetooth and airplay, but no access point or other features.  Hope this helps someone, took a ton of tries to get it to work.  I have another RPI 0 W and phatdac that I'll test it out with, probably with Jessie Lite to see if that makes a difference.

Thanks

BaReinhard, 2017-05-31:
Thanks for posting! There are some compatibility issues when running other scripts. I believe that the phat-dac script shouldn't be needed if running the install script with the proper settings. It has been a while since I was able to test a phat-dac, but from what I recall I have implemented everything that their bash file does. That being said it is entirely possible that I was looking at a different version or entirely a different product. 

I will take a look at the code and see if there is a better way to implement this to make it easier.

spin1974, 2017-06-02:
Let me know if you need me to test anything.  I'm happy to help, this Git is exactly what I was looking for.

BaReinhard, 2017-06-03:
Will do, I still need to get back into gear here. Been working on getting a job in web development so I have been focusing more on portfolio pages and getting more time with web apps. That being said I have just finished an internship so I should have a bit more free time until August so I should have a better just of what needs to be done here soon, if theres anything I need some assistance on I'll for sure reach out.

Thanks for the offer!

seehadizzle, 2017-06-27:
Thank you spin1974 for posting the steps you used!  I previously installed a Jessie image, ran this script and could not pair any devices with the Pi Zero. The one time I was able to pair was by manually authorizing the device from the Pi and even then, was unable to control the volume. This works perfectly now!

mgeorgia, 2017-09-18:
Wowzers.  I wish I had found this post earlier!  I had been running into the same problems you were having.  I tried all sorts of various bluetooth libraries and just couldn't get them working.  I followed this to a t with Jessie Full (the non-lite version) and it works great.  Also with a Pi Zero W/pHAT DAC Combo.

BaReinhard, 2017-12-23:
This somehow popped back up on my radar. Has anyone recently tried an install with the phatdac? I made a few new changes to how sound cards are handled with the latest branch, which I assume fixes a lot of issues with sound cards, but I don’t have access to a phatdac. I am thinking about purchasing one but won’t have my hands on it until at least mid January.

spin1974, 2017-12-23:
I do, and I'm meaning to spend time on this because it doesn't consistently connect for airplay.  Give me a couple days and I'll test it with my kit and let you know.

BaReinhard, 2017-12-23:
https://learn.pimoroni.com/tutorial/phat/raspberry-pi-phat-dac-install

It looks like this should be a good place to start debugging. Check the manual install portion and each of the files.

BaReinhard, 2017-12-23:
Now that I think about it, I should have asked whether the device works for any other audio output.

BaReinhard, 2018-01-11:
I’m unable to test with a phatdac, if anyone can provide testing and let me know. However, based on the provided changes to soundcard install it should work as expected. If it doesn’t please reopen this issue.

closed 2018-01-11 by BaReinhard