BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #38: Fragmented audio via Bluetooth

dipietro-salvatore, 2017-06-14:
Hi, 
I just installed the BT receiver on my RPi 2 with Raspbian and a Bluetooth USB dongle without any external USB sound card.
I am transmitting the audio content from my smartphone. At the beginning the audio is ok but, after 30 seconds, it starts to have empty gaps between the music. These gaps become always more frequent in time during the execution and after few minutes I am not able to listen anything.

Do you have any idea what should I do to fix it?

Thanks

Mackie1, 2017-07-19:
I'm running the PI Zero with Phat Dac, i have the same issue, the audio cuts out on when using my android phone and also jumps half a second when using my laptop. this results in video and sound being out of sync and somehow it then catches up for a bit 

I installed the home setup, and selected 0 (no soundcard) and then i ran the Phat Dac script as per their site.

seehadizzle, 2017-08-09:
This can be resolved by disabling the onboard wireless. I had the same issue and it's working flawlessly after getting a usb wireless adapter and disabling the onboard.

old-square-eyes, 2017-11-11:
@seehadizzle what method did you use to disable the onboard WiFi?

old-square-eyes, 2017-11-11:
Awesome got it to work from here  https://raspberrypi.stackexchange.com/a/62522/55926

old-square-eyes, 2017-11-12:
Success was short lived. I'm using my USB WiFi module and had several hours of crisp bluetooth audio on my mbox micro USB sound card. Today I connect and get no audio. Kernel log just says bluetooth has crashed. No amount of rebooting, hot plugging or re-pairing seems to bring it back. Meanwhile I can listen via the Pi/USB audio from MPlayer with no issue.

Tuchos, 2017-11-17:
I had the same problem. I was using the on-board bluetooth, so i bought a usb dongle and disable the on board bluetooth

`CD boot
sudo nano config.txt`

And add this at the bottom  "dtoverlay=pi3-disable-bt"

reboot (will need to pair the phone to the pi again) and you are ready to go.

BaReinhard, 2017-11-18:
Sorry guys, been busy with other things. Recently was informed I wouldn't have a job by Jan 1. The good news I may have more time for this. Currently, I don't have a lot of time to work this. I am looking to find a group of devs that may be able to take over the development of this repo. I believe I may have a possible group, but only time will tell. I hope to be able to return to this at some point, Ideally I will be going through the code, making it simpler removing a lot of code that isn't being used and creating a new repo of the unused code to focus on the important parts of it. The idea is to make the repo a compilation of the auto connection of BT and easy access to AirPlay for iOS devices. Still a ways away but if anyone knows of developers who have experience with BT or Linux will be ideal for this process. The actual coding for this repo isn't that advanced as far a software development is concern but more intensive on Linux architecture (at least from what I see, I could be wrong. This was one of my first FOSS projects and I would love to see it prosper)

BaReinhard, 2018-01-11:
I believe these issues were common before we made the switch from using pulseaudio 5 to 6. Bt playback has been solid for well over 12 hours. If you are still having issues I would suggest using the latest master branch and ensure your pi has an adequate power supply. I suggest 2A just to be safe.

closed 2018-01-11 by BaReinhard

bermudaparty, 2018-03-24:
> using the latest master branch

If I install it using the normal instructions (from readme), this should be the case. Correct?

bermudaparty, 2018-04-02:
> I believe these issues were common before we made the switch from using pulseaudio 5 to 6. Bt playback has been solid for well over 12 hours. If you are still having issues I would suggest using the latest master branch and ensure your pi has an adequate power supply. I suggest 2A just to be safe.

I just upgraded my power supply (official Raspberry Pi supply 5.1V, 2.5A) and the problem persists when I don't disable wifi. Any ideas? I installed fresh last week, so well after you indicated that it shouldn't be a problem anymore.