BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #39: No Audio Output with HifiBerry DAC+ Pro

DefinitelyNotSam, 2017-06-16:
Hello,

I recently picked up a HifiBerry DAC+ Pro to pair with a Raspberry Pi B+ board to use as a bluetooth audio receiver in my vehicle. My needs are pretty simple, I would just like to use the bluetooth connection to bring audio into my vehicle AUX over the HifiBerry DAC+ Pro.

Unfortunately, after getting everything setup I'm not getting any audio out of my HifiBerry DAC. The bluetooth connection appears to be working correctly (my phone successfully pairs to it, and displays it as a media capable device, so I am assuming A2DP is working correctly). I simply get nothing for audio out of the two RCA connectors on my DAC.

After doing a bit of googling, I did see from another thread that you were able to get your setup working with the DAC+ Pro, and I was wondering if maybe I made a configuration error while setting it up.

On initial install, I selected the following options:

1. Install the Raspberry Pi Audio Receiver Car Installation

2. HifiBerry DAC Standard/Pro

Were these the correct options for my configuration, or would there be more appropriate options for what I'm trying to accomplish?

Thank you for all of the work you've done on this, any any assistance you can give me, I really appreciate it!

Dmac9244, 2017-08-09:
I don't think you made an error in setting up your pi, I used the custom option when setting up mine, and I had the same problem. You're right, A2DP is working, the sound is coming from the aux port!
I found a fix for it, though! As soon as I can, I'll make a proper fix, but for now, you can go into your /boot/config.txt file, and comment out the line that says "dtparam=audio=on." This line is what makes sound go through the aux port, disabling it disables the aux port, and allows the audio to go to the DAC+. After a reboot, you should be fine.

BaReinhard, 2017-12-01:
@Dmac9244  hit it right on the head. I would also ensure that the proper dtparam is uncommented for the DAC+ PRO, i don't remember exactly what its called but it should be in a list of commented lines in the `/boot/config.txt` file

closed 2017-12-01 by BaReinhard