BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #40: Bluetooth audio going through HDMI, HIFI Berry AMP+

Wvspek, 2017-06-23:
Hello!

I'm experiencing an issue where airplay audio is streaming fine through the Hifi Berry AMP+, but any audio d]sent through Bluetooth is being routed through HDMI instead, if the system is started up headless, Bluetooth audio still does not work at all.

I'm using raspian Jessie lite, and of course, an Amp+

Wvspek, 2017-06-23:
If I blacklist the onboard audio chipset from alsa, Bluetooth now works as intended, but airplay plays no audio at all.

Dmac9244, 2017-08-10:
Hey, I figured out your problem! I was having the same issue trying to fix the bluetooth audio issue with the HiFiBerry DAC+. The problem is that in /etc/shairport-sync.conf, line 62, which says, 
"output_device = "hw:1,0";" should say "hw:0,0" after getting rid of the onboard audio chipset. Otherwise, it's asking for a device that doesn't exist.

closed 2018-01-11 by BaReinhard