BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #41: Possible Solution for say.sh

ibrodowski, 2017-07-05:
@BaReinhard 

I know that this isn't necessarily an issue, so I apologize in advance.

I noticed that **say.sh** was all commented out, so I decided to figure out what's going on with it.

**My RPi setup**:
- Raspberry Pi 3 Model B with built-in bluetooth and wi-fi
- Raspbian Jessie with Desktop (PIXEL) (RD = 2017-06-21; KV = 4.9); ran _**sudo apt-get update**_ and _**sudo apt-get dist-upgrade**_
- Adafruit Speaker Bonnet for Raspberry Pi (hifiberry-dac) aka **HifiBerry DAC Light** in **_install.sh_**

Ran **install.sh** using the following options:
- (1) Install the Raspberry Pi Audio Receiver Car Installation
- (1) HifiBerry DAC Light
- Used same name for both services

After reboot, the following **manual changes** were made:
- **/etc/asound.conf** (change all **card** references from 1 to 0)
- **/etc/shairport-sync.conf** (change **output_device** = "hw:1,0" to "hw:0,0")

List audio-devices using **aplay -L**, but the known **working** output devices are as follows:
1. **plughw**
2. **sysdefault**

Modify **say.sh** in **/usr/local/bin** (change **_sysdefault_** to **_plughw_** as needed):
` #!/bin/bash`
`say() {`
`   local IFS=+;`
        `espeak $* --stdout | aplay -D 'sysdefault'`
`} 2>/dev/null`
`say $*`

**Results**:
- Currently, whenever I connect or disconnect my **iPhone 7 Plus** (10.3.2, 14F89) to this Raspberry Pi 3, **espeak** says **_Device_**; it's a bit distorted and it does crackle/pop a little.  However, that might be because the kernel doesn't support mono audio output via the I2C interface (unless this changed recently?); whereas espeak is outputting "16-bit Little Endian, Rate 22050 Hz, Mono".

BaReinhard, 2017-07-05:
This is great, I never got around to getting into it with say.sh. I was only able to have it say something when connected via hdmi and somewhat lost interest. If you are able to implement this into a pull request I will gladly merge it.

ibrodowski, 2017-07-05:
I'll work on implementing this potential fix into a PR after I've completed additional testing.

I've got a Raspberry Pi Zero W that I use in the car, which was set-up using the custom option, bluetooth-only, no shairport, so I'd like to make sure it works as expected.

I'll need to figure out a simple way to read the output from **aplay -L** and search for the keywords of **_plughw_** and/or **_sysdefault_** and then update **say.sh** to match _either-or_ accordingly (without having to resort to some crazy kung-fu scripting skills).

I've got another question regarding **_simple-agent.autotrust_**, so I'll just submit another issue.

BaReinhard, 2017-12-01:
Been MIA lately, but I think ill be more active for at least a couple weeks until a new job starts. The easiest way to get data from aplay would be something like

```
sysdef=`aplay -l | grep sysdefault`
plughw=`aplay -l | grep plughw`

if [ -z "$sysdef" ]; then
  update say.sh for sysdef
  exit 0
fi
if [ -z "$plughw" ]; then
  update say.sh for plug hw
  exit 0
fi
```

Or something along those lines, there are likely some syntax errors , but the grep commands should hold water.

BaReinhard, 2017-12-21:
closed for inactivity. If a solution is provided, I would gladly add to the install.

closed 2017-12-21 by BaReinhard

Lauszus, 2018-04-08:
I got espeak working using the following command in say.sh:

```bash
espeak $* --stdout | aplay -Dsysdefault
```

Note that setting the device to ```-Dplughw:0,0``` instead work as well.