BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #42: Sap driver initialization failed

jayheavner, 2017-07-06:
After installation of bluetooth I get the following errors when running a sudo service bluetooth status.

pi@raspberrypi:~ $ sudo service bluetooth status
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled)
   Active: active (running) since Thu 2017-07-06 21:32:22 UTC; 4min 16s ago
     Docs: man:bluetoothd(8)
 Main PID: 647 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─647 /usr/lib/bluetooth/bluetoothd

Jul 06 21:32:22 raspberrypi bluetoothd[647]: Bluetooth daemon 5.23
Jul 06 21:32:22 raspberrypi systemd[1]: Started Bluetooth service.
Jul 06 21:32:22 raspberrypi bluetoothd[647]: Starting SDP server
Jul 06 21:32:22 raspberrypi bluetoothd[647]: Bluetooth management interface 1.14 initialized
Jul 06 21:32:22 raspberrypi bluetoothd[647]: Sap driver initialization failed.
Jul 06 21:32:22 raspberrypi bluetoothd[647]: sap-server: Operation not permitted (1)
pi@raspberrypi:~ $

BaReinhard, 2017-07-06:
I am not 100% on this but I believe with Raspbian Jessie you should be getting status' of services with systemctl. For example, do you get the same error with 

```
sudo systemctl status bluetooth 
```
or

```
sudo systemctl status bluetoothd 
```

BaReinhard, 2017-07-06:
You may want to try this:

(Possible Fix)[https://raspberrypi.stackexchange.com/questions/40839/sap-error-on-bluetooth-service-status]

jayheavner, 2017-07-07:
I was actually using an instance of Volumio (that I've since abandoned). It's an interesting project but I've gone back to Jessie with Pixel.

I do have a couple questions and I can move them to separate issues if that helps.

After installing custom with everything but Kodi and IR I've lost my wifi interface (no wireless interface found) which makes me think it's broadcasting as a hotspot. Is that the case? If so, what's the SSID? If it's not the case then what do I need to do to get it back?

I don't see an airplay connection option from IOS devices. Is this related to the wifi?

Bluetooth pairing works but I'm getting a prompt on the pi to accept. Any way to auto-accept the connection?

Also, I selected a hifiberry soundcard during setup. I've confirmed it's selected but I'm still getting audio through HDMI. I can go through the hifiberry config but I thought I'd ask if there's a simple fix before I do so.

BaReinhard, 2017-12-01:
Sorry been gone for a while.

If it is acting as a 'Hot Spot' it would have asked you to name the SSID.

Also to see the AirPlay device you will need to be on the same network as the pi for it to be accessible on your iDevice.

The auto connection should already be implemented. Unfortunately, you will be need Jessie Lite, Pixel doesn't work. you can download it [here](http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/)

For the hifiberry card, I would ensure that dtparam=audio=on is commented out, and the correct dtparam for the hifiberry card is uncommented. In the /boot/config.txt there should be a number of lines commented out near the bottom of the file that tells you what each line is for, simply uncomment the line under the name of the hifiberry card you have.

closed 2017-12-01 by BaReinhard

Landris18, 2019-10-16:
It doesn't resolve my problem

azamet90, 2019-12-20:
i have this same problem on my fresh installed kali... i see this in google and posted here