BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #43: No Audio with Bluetooth A2DP on HifiBerry DAC+

tulpenwase, 2017-07-13:
Hello BaReinhard,

i have a Rassoberry Pi2, with HifiBerry DAC+ and Raspbian Jessie with desktop. I Install Raspberry Pi Audio Receiver Install Home Version. 
When i play Musik via Bluetooth A2DP i have only sound on the Audio Jack. The HifiBerry is still not working with Bluetooth A2DP. But when i play Musik via Airplay i have sound with my Hifiberry. 
I switch the default Sound in Audio Device Setting to snd_rpi_hifiberry_dacpuls (Alsa mixer) (Default) but without results.

I am confused. Is there a option to switch Audio output for Bluetooth A2DP to Hifiberry?

Dmac9244, 2017-08-09:
I had the same problem, but I found a quick fix for it! On your pi, once you ssh into it, there's a file /boot/config.txt. In there, there's a line, "dtparam=audio=on". If you comment it out, and reboot, it'll work!

I'll make a fix for it if I can, but for now, I'm telling you so you can get it fixed quickly.

Dmac9244, 2017-08-10:
Oh, and because you use Airplay, you have to change a line in /etc/shairport-sync.conf to say "hw:0,0" instead of "hw:1,0". It's line 67, I think, under the alsa configuration section. If you don't do this, Airplay will stop working because the hifiberry will be device 0 instead of device 1 after doing the other thing. After a reboot, everything should work properly.

tulpenwase, 2017-08-31:
Yep thats is. The Hifiberry works now perfekt. Thanks

closed 2017-08-31 by tulpenwase

Dmac9244, 2017-08-31:
You're welcome!

BaReinhard, 2017-09-07:
thank you @Dmac9244  for the help here!

Dmac9244, 2017-09-07:
You're welcome, @BaReinhard! Thank you for approving my work!