BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #44: RPi3 Install for Home Audio Failed

burgessr, 2017-08-04:
Hi BaReinhard,
I must say I began installing your solution with hope in my heart after wasting a lot of time on outdated solutions which did not work. However, sad to say, that after the lengthy install process, during which I noted a few errors saying device was full, the machine rebooted itself but would not allow me to log in - seems that the password got lost corrupted or forgotten. However I was able to bluetooth connect my iphone at which point the speakers spoke a single word - i'm pretty sure it was "device". But that was all - music still plays on the iphone even when it believes it is connected to the RPi.

So now I guess I'll have to reinstall raspbian and forget about this so simple but so difficult business.
Regardless of my experience, thanks for your efforts.

BaReinhard, 2017-08-04:
Hey burgessr I'm sorry for the not so simple experience, it has been a while since I've been able to test out the Install process, as new projects, getting myself into a new line of work, and raising my daughter have gotten "in the way". That being said, some information would be useful: what operating system are you using?, what raspberry pi, and was it a fresh install?

The install is meant for a fresh raspbian lite solution, no use of pixel or other add one, it has been a while since I was able to test it so it's possible that the new raspbian images has essentially broke the install. In which case will require some work to get a fix.

burgessr, 2017-08-04:
Hi Brett, thanks for your quick response. It was an out of the box RPi 3B running raspbian and pixel, firstly updated and upgraded, and I followed your github install description. I have an RPi0w here which I could perhaps try. Again, many thanks and don’t feel pressured, I have lots of other stuff to play with.

Regards, Richard Burgess

burgessr, 2017-08-09:
Hi Brett,

 

Works fine with pixel loaded on a 16Gb card!

 

Cheers, Richard

BaReinhard, 2017-12-01:
Been MIA lately, so I apologize for that. So are you saying it worked on Raspbian Jessie w/ Pixel? If so that is a first! lol

BaReinhard, 2018-01-11:
This has been fixed with the latest merges and now should work for others as well!

closed 2018-01-11 by BaReinhard