BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #49: Can't get sound through USB Soundcard M Box Micro

old-square-eyes, 2017-08-18:
I'm streaming to my Pi over A2DP and hearing the audio through the terrible 3.5mm jack. I've spent hours trying to get audio through the USB sound card.

- I have set the card as the index 0 device according to http://plugable.com/2014/11/06/how-to-switch-to-usb-audio-on-raspberry-pi/

Any ideas?

Dmac9244, 2017-08-18:
According to that article, that fix works only for the 2014 version of Raspbian. It seems like things are different if you're using the recent 2016/17 version of Raspbian, Raspbian Jessie. The change you want to make is in /etc/asound.conf. change "card x" in each instance to say the number of the USB soundcard. Make yourself sure of it's number by using the command aplay -l. You are using Raspbian Jessie, right? (Lite version or the Full version?)

After you commented out dtparam=audio=on, did you reboot? And if you did, did audio still come out of the onboard sound card? If it didn't, your problem is half-solved, I think. If it did, then I'm as confused as you are, but you can still try doing the above. After commenting it out and rebooting, the only available device in aplay -l should be your USB sound card, numbered as card 0. /etc/asound.conf should also say 0 if that's the case. If you don't see your card, then ALSA isn't initializing it, and you might need /boot/config.txt to load the right kernel module to initialize your soundcard. I can't help you there because I found very little on Google about using your soundcard with the RPi.  

I don't have access to my RPi right now, so I can't see exactly how it's configured. I also don't have your USB sound card to test with, so I can't try things out myself to see what works. Hopefully ALSA sees your card, and hopefully configuring /etc/asound.conf will work for you. Otherwise, I'm not sure how to help.

Dmac9244, 2017-08-18:
Also, as a side note, the HiFiBerry DACs are incredible little devices. They attach directly to the GPIO pins and sit above the Raspberry Pi. The DAC+ Pro model is cheaper than the MBox 2 Micro, but should sound really good. The standard version comes in RCA and AUX output models, each about $40. The Pro version is ~$60. It might not be an upgrade from your USB DAC, but it should be of relatively equal quality, and will be a lot easier to configure and use!

old-square-eyes, 2017-08-19:
Thanks for the help:) I think I'd already got through most of that. I'm determined to get this card working. It's for the spirit of the project (which is to make a media centre out of junk and hifi I have lying around for the amp and speakers I just purchased second hand - Topping AMP and Polk Audio speakers). The card just happens to also sound great, and is supposed to be supported according to [here](http://www.zamaudio.com/?p=97). Mine is the micro but uses the same chip.

Now I've been on the job for a few hours now:) I have finally got somewhere.

You are right. Commenting out `dtparam` had my card loading (apparently) correctly. BUt nothing in the world I could do could play audio. I tried bluetooth (previously working on local 3.5mm) and no luck. Then I tried...

 - `omxplayer example.mp3 -o alsa:hw:0,0` - Including card name (M2) variations. Seememplayerd to play but no sound??
 - Forcing usb 1.3 in /boot/cmdline.txt
 - `aplay --device=hw:0,0 exampl.mp3` - Just got a sample/bitrate problem

All things pointed to the card being loaded correctly but not being utilised properly.

Finally, `speaker-test -c2 -D plughw:0,0` gave me a nice crisp pink noise on each speaker.

After that I narrowed it down to problems playing my example.mp3. Google tells me to try mplayer, which I install and then `sudo mplayer example.mp3`. Bingo.

So the card works on the default kernel and I'm thrilled. I'm currently listening to internet radio via mplayer. I have to be root but it's working,

So now... how get SSRPAR to follow the same audio route/device?

old-square-eyes, 2017-08-19:
On another note I saw [this post](aliases.conf). Which talks about `snd-usb-audio index=-2` also being set in an alias file. It didn't help me but it might help someone else.

Dmac9244, 2017-08-19:
Hm, is the user you're using added to all the necessary user groups? Sometimes it needs to be to get things like audio working properly. 

Also, you might have to configure Pulseaudio. Pulseaudio is like a front-end to ALSA, it takes care of the sound sources and sends the data to ALSA, which plays it through the speakers. I still can't look around on my pi to see how it's configured (I'm on vacation, no way to access it!) but looking at the Arch Wiki page for Pulseaudio configuration, it says that /etc/asound.conf is the file that configures ALSA and Pulseaudio to work together well. (That's the best I can explain it.) I'm thinking now that Pulseaudio isn't working with ALSA properly.

Might this help? [Change Default Audio Device from CLI (Question on ask Ubuntu)](https://askubuntu.com/questions/14077/how-can-i-change-the-default-audio-device-from-command-line)

Or this? [USB card as my default device (question on Rpi Stack Exchange)](https://raspberrypi.stackexchange.com/questions/19705/usb-card-as-my-default-audio-device)

The answers there (link 2) say basically the same thing as I was trying to say, but there is another answer that worked for Raspbian Wheezy that might also work.

What if you try setting snd-usb-audio index=-2 to 0, and getting rid of anything that says snd-dtparam-audio to 1? I say that because it seems like setting those things to 0 enables them, while setting them to 1 "disables" them. And perhaps try using alsamixer to select the sound card, and turn up the system volume for it? 

As I said, I wish I could try it out myself, and find a proper solution instead of throwing possible solutions at you. It would even be easier to find a solution if I could test things out, see what's happening. I think you might be able to fix things either with adding a user to a group, playing around in alsamixer, or playing around in asound.conf. Maybe try playing around more in aliases.conf too, and see how it goes. Good luck!

old-square-eyes, 2017-08-21:
- `$ fgrep -ie 'audio' /etc/group` gives me `audio:x:29:pi,pulse,shairport-sync` - Do I need to be in any other groups?
- `pulseaudio` gives `Unable to autolaunch a dbus-daemon without a $DISPLAY for X11

No joy yet. I'll keep trying. FYI I'm running this from SSH. Should that matter?

Dmac9244, 2017-08-21:
Does that mean that user "pi" is in group "audio," or is user "audio" in groups "pi, pulse, and shairport-sync?" Whatever user you're using, it should be in group "audio".

Everything should work from SSH, it doesn't matter that you're using it. SSH is the way the pi streams its terminal to other computers, using it is like attaching a monitor and keyboard to the pi, and using the terminal that way. 

So, running `pulseaudio` gives that error? Is it running at all? Please check with `pactl list`. That should tell you if the Pulseaudio server is running, what modules are loaded, and the available sinks (ALSA) and sound devices (your sound card). Hopefully, it should be running, ALSA should be an available sink, and your sound card should be seen. If that is the case, it could be possible that things aren't working because your user isn't in group "audio".

[This website](https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/PerfectSetup/) is part of the Pulseaudio documentation, and can tell you more than I can about Pulseaudio. Reading through it, and other pages on the site might tell you a bit more about how Pulseaudio is supposed to be configured.

old-square-eyes, 2017-08-22:
- I read it as `pi` (my user) is in the group `audio`
- `pactl list` only works as root, but spits out a wall of text about my USB audio device. I pasted that [here](https://pastebin.com/9zRiYzb2). My card seems to be listed.

I'll check out that page thanks :)

Dmac9244, 2017-08-22:
You're welcome!

Hey, I just noticed on your pastebin that the sink for your card is labelled as sink #1, instead of sink #0. When I did `pactl list` on my Rpi (finally!), my DAC was labelled as sink #0. Also, when I have audio running through bluetooth, that command says that the sink is RUNNING instead of SUSPENDED, can you do another favor for me, please? While you have audio running from your phone to the pi, can you do that command again? (it's fine that it needs to be run as sudo.) It should say there is a sink #1 that is RUNNING that is your USB sound card, a source that is named "bluez-source" that is also labeled with the name of your phone- "iPhone", "Pixel", something like that. This would mean that Pulseadio is successfully receiving audio from your phone, and successfully playing it into the right sink, and the problem is elsewhere. If the source is RUNNING but the sink is SUSPENDED, then the problem is there somewhere. In that case, [this](https://askubuntu.com/questions/71863/how-to-change-pulseaudio-sink-with-pacmd-set-default-sink-during-playback) has a way to change the default sink, which might still work. If the source isn't RUNNING, then the problem might be with bluez, but I don't see that happening because bluez was working properly with the onboard audio.

Dmac9244, 2017-09-01:
@old-square-eyes Did you get your setup working?
If not, I'm sorry I couldn't be of more help. If so, then great!

old-square-eyes, 2017-09-04:
Thank you for your help. Unfortunately not. I put it down for a bit after pulling all my hair out :) I feel like I might re-flash my sd card. Is there any point trying a different Rasperian?

Dmac9244, 2017-09-06:
You're welcome!
As long as you're using the latest Raspbian Jessie version, you should be good to go. Don't use Raspbian Stretch, it won't work.

A re-flash might work, just remember to disable the onboard sound card once you do! From there, we can figure out what's happening and hopefully find a solution.

old-square-eyes, 2017-09-09:
Bluetooth audio

In Jessie, we used PulseAudio to provide support for audio over Bluetooth, but integrating this with the ALSA architecture used for other audio sources was clumsy. For Stretch, we are using the bluez-alsa package to make Bluetooth audio work with ALSA itself. PulseAudio is therefore no longer installed by default, and the volume plugin on the taskbar will no longer start and stop PulseAudio. From a user point of view, everything should still work exactly as before – the only change is that if you still wish to use PulseAudio for some other reason, you will need to install it yourself.

old-square-eyes, 2017-09-09:
@Dmac9244 - Thank you for sticking with me. Stretch didn't work. I assume the scripts need updating. So I flashed Jesse, and ran the script with my Audio Kontrol USB Card attached. Without any tweaks I got audio from my card for 30 seconds. I came back here to say a clean wipe fixed it and I thought I'd better test the M Box card. I hot swapped, Rasperian mouse and keyboard crashed and nothing has worked since :/ I'm in the midst of a new flash.

On another note. I'm using full Rasperian Jesse this time with the desktop UI. I'm thinking maybe PulseAudio only works like this and not in headless. Nevertheless back to hair pulling.

old-square-eyes, 2017-09-09:
OK wasn't crazy. Reflashed, ran script and it finished this time with no errors. Connected phone to BT and got music through the Audio Kontrol 1 again. This time I just listened. And it cut out after about 10 seconds. It's not recovered. Are there any logs I can look at?

old-square-eyes, 2017-09-10:
[Here](https://pastebin.com/eMi7TfPa) is a new paste with my phone BT connected and streaming.

Dmac9244, 2017-09-11:
... I am thoroughly stumped now. I think it's a PulseAudio thing now, or a Bluetooth thing. Your Pastebin shows that both your sink and your source are suspended, meaning they aren't running. Maybe your Bluetooth audio isn't making it to PulseAudio? If you want, maybe you should check asound.conf again? Otherwise, it might be best to start asking around on some PulseAudio forum, where you might get answers from people who know it better than I do.

It makes sense to me (for some reason) that VLC works only when you choose "ALSA output" and then "device." It's good that it is working, that's further than we were before, right?

Quickly:

- It doesn't matter whether you use the full or lite version of Raspbian, but it seems to be helpful that you chose the full version this time!

- I'm pretty sure that the sound card should be attached when you run the script. That way, it can be configured properly.

- Yeah, running the script multiple times causes problems.

- The powered hub is a good idea, I had to use one to get my USB HDD working.

- I've personally never had any related problems with the script, but I've seen others have these sorts of problems with it. I don't think it should be causing the problems we're trying to fix, though.

- Your Audio Kontrol 1 cut out? I don't know why that happened. Did you flash over again after that happened? Sorry, I'm a bit out of the loop, I've had a lot of work to do (school, actually) in the past couple days.

- If you ever decide to reflash and try the script again, I suggest plugging in the MBox before running the script, along with your Bluetooth USB dongle, if you're using one. Is that what you did this time? 

- Oh, you're welcome! I'm happy to help.

I find it amazing you're still trying to get this, I would have given up weeks ago if I were in your place, even if I had a friendly (dashing, eloquent, handsome, intelligent... ;)) stranger helping me on the internet!

old-square-eyes, 2017-09-12:
I'm flashing again. Do I need to make any tweaks before running the script (setting default device, disabling internal etc)?

old-square-eyes, 2017-09-12:
OK. I'm back on the Audio Kontrol 1 . Clean flash. I got bluetooth audio again for 10 seconds. Then it simply stops. I haven't been to a Pulse forum yet (does one exist?), but I want to find some logs. I just had an idea that it might be a buffer problem.

old-square-eyes, 2017-11-11:
Finally got it to work. Droppouts stopped when I disabled onboard WiFi!

https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/38?_pjax=%23js-repo-pjax-container

Dmac9244, 2017-11-11:
That's great! I'm sorry I couldn't be more helpful. Not only did I run out of ideas, I ran out of time!

old-square-eyes, 2017-11-12:
There's something else at play. I happily listened for a few hours. But today I can't get it to pass audio. I'm getting the crashing bluetooth process in the kernel log. Seems to be the way you hold your tongue. As always, playing audio via mplayer works fine. So I'm sure I have the Pi set up correctly in terms of USB audio. There's just something very unstable about the bluetooth stack.

BaReinhard, 2017-12-01:
Sorry I have been MIA, this issue has been fixed as of 2? maybe 3 days ago. We ended up making a change to what version of pulseaudio we were using. We are compiling pulseaudio 6 locally and making a change in the  client.conf file. However, this fix is still only for Raspbian Jessie LITE, which can be downloaded [here](http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/). I also found a solution to the error of starting pulse without X11, we ended up exporting the Display and all errors are gone.

Sorry again I was gone for all your issues

closed 2017-12-01 by BaReinhard