BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #50: Install failed under Raspbian Stretch

muskatnuss123, 2017-08-18:
Hi,
tried to run the install scrip last night on Raspbian Stretch, but it failed. It can't install the Bluetooth option.
What is the Sense of this part in bt_pa_config.sh ? Make pulseaudio failed. I have also installed libjson-c-dev manually. 

```
# BT FIX
--
  | mkdir /etc/pulsebackup
  | cp /etc/pulse/* /etc/pulsebackup/
  | git clone --branch v6.0 https://github.com/pulseaudio/pulseaudio
  | apt-get install intltool
  | apt-get install libsndfile-dev
  | apt-get install libcap-dev
  | apt-get install libjson0-dev
  | cd pulseaudio
  | ./bootstrap.sh
  | make
  | make install
  | ldconfig
  | cp /etc/pulsebackup/* /etc/pulse
```

Furthermore, there seems to have been some major changes regarding Pulsaudio and Bluetooth. See more:

https://www.raspberrypi.org/blog/raspbian-stretch/

> BLUETOOTH AUDIO
> 
> In Jessie, we used PulseAudio to provide support for audio over Bluetooth, but integrating this with the ALSA architecture used for other audio sources was clumsy. For Stretch, we are using the bluez-alsa package to make Bluetooth audio work with ALSA itself. PulseAudio is therefore no longer installed by default, and the volume plugin on the taskbar will no longer start and stop PulseAudio. From a user point of view, everything should still work exactly as before – the only change is that if you still wish to use PulseAudio for some other reason, you will need to install it yourself.

BaReinhard, 2017-08-18:
i believe this should be an easy fix by adding sudo apt-get install pulseaudio, but i am unsure if that will work 100%. It will need some revisiting for sure

BaReinhard, 2017-12-01:
Still have yet to get started on fixing any errors with Stretch, also I believe once it is fixed for stretch it will first be implemented for stretch lite, then eventually stretch desktop. It looks like we will need to be either removing bluez-alsa or disable it, and compile bluez for use with pulseaudio, or completely re do things with alsa. Unfortunately, I don't think I have the time to try with alsa.

BaReinhard, 2018-01-08:
Alright, I have a stretch version that is in the testing phase at the moment, go [here](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/102) and follow the instructions. It will need to be on a new installation of Raspbian Stretch, (it has been tested on the LITE version, but i suspect it will work for Stretch w/ Pixel). Please leave feedback in issue #102 .
  

Any issues or feedback please redirect to issue #102

closed 2018-01-08 by BaReinhard