BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #51: Bluetooth audio ouput issues

gstvbrg, 2017-09-01:
Hey! 
First off thanks for all your work on this great project, it's an awesome solution to all the disparate rpi audio tools out there. I recently set it up on a rpi3 running Raspian Stretch, with PulseAudio installed and no sound card using the Home installation. My main goal was to setup a bluetooth receiver for an amazon echo, and for the most part it seems to be working in so far as the devices pair and interact as expected. The (critical) problem is that I'm not getting audio output! Through raspi-config I set the audio jack as the forced output, which is what I hoped to use. I'm wondering what other trouble-shooting steps I should try. Would just getting a sound card be the simplest solution? I'm already using an Adafruit 20W amp, so I don't think I necessarily need a sound card but I would get one just to make it work. Or maybe trying an older version of Raspbian? Any suggestions would be greatly appreciated. Thanks!

eschriek, 2017-09-01:
A slight change in the bluez-udev is needed under raspbian   Stretch since the audio source name is incorrect. This is likely due to a newer pulseaudio version, but I might be wrong.
Anyway, if you run 'pactl list sources short' you will see your Bluetooth source if connected. You will also notice that the source in the bluez-udev script lacks the '.a2dp-source' part. If you add this to the script it will work fine.

LordSchippers, 2017-09-02:
I face the same problem. I edited /usr/local/bin/bluez-udev and replaced audio_source=bluez_source.$(sed 's/:/_/g' <<< $name) with the following lines:
variable1=bluez_source.$(sed 's/:/_/g' <<< $name)
variable2=".a2dp-source"
audio_source="$variable1$variable2"
It still does not work. Can I get an instruction on what I have to do to fix it/can you fix it?
Thanks!

eschriek, 2017-09-03:
It is kind of hard to help you based on 'It does not work'.
Please provide any logs that give some insight, for instance the bluez-udev scripts logs to /var/log/syslog.
Connect your BT device and do a 'tail /var/log/syslog'. Likely you will see something related to some udev failure. If that is the case you should enable udev debugging using udevadm.

By the way, if you just want a quick fix I suggest you install Raspbian Jessie and be done with it.

old-square-eyes, 2017-09-09:
OK doing this now. I see the new Rasperian is out though. And this is very interesting...

> Bluetooth

 audio

In Jessie, we used PulseAudio to provide support for audio over Bluetooth, but integrating this with the ALSA architecture used for other audio sources was clumsy. For Stretch, we are using the bluez-alsa package to make Bluetooth audio work with ALSA itself. PulseAudio is therefore no longer installed by default, and the volume plugin on the taskbar will no longer start and stop PulseAudio. From a user point of view, everything should still work exactly as before – the only change is that if you still wish to use PulseAudio for some other reason, you will need to install it yourself.

old-square-eyes, 2017-09-09:
I ran the script and I can't tell if it finished because after about 20 minutes of verbose text I was kicked off the SSH session. Something must have worked because my named BT device was visible to my phone. Except I don't get as far as you guys. I get "Unable to connect to <my device>" from my phone when trying to pair. Now I can't tell if it's Stretch or the script didn't finish properly.

I tried Stretch because I couldn't get it to work on my USB sound card on Jesse.

BaReinhard, 2017-12-01:
Sorry I haven't been around for a while guys. For now I haven't tried installing it on stretch whatsoever. This past week there was a change in the installation in regards to bluez, we ended up scraping the normal bluez that comes with Raspbian Jessie Lite and compiled bluez 6. I would assume that this may solve the issues with bluez-udev, but I am not certain. 

If you want a for sure install, download the latest Raspbian Jessie Lite [here](http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/). This will work for sure in regards to bluetooth and airplay. I have done 3 installs with it in the last 2 days. 

For those interested in trying with the new installation on stretch feel free to give it a try. I would suggest using stretch LITE though download it [here](https://downloads.raspberrypi.org/raspbian_lite_latest) . I am unsure of the differences  between lite and desktop, but I am assuming it has to do with some of the patch commands ( I will be implementing some sed commands in place of them so the script is less fragile).

closed 2017-12-01 by BaReinhard