BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #52: BT Audio not playing

jefflivid55, 2017-09-09:
Hi, I am using a Pi3 with Raspbian Jesse

I run the install script, and after a while the Pi restarts.
I see the normal desktop upon reboot, and the BT symbol is flashing.
I can pair with the Pi from my phone, but when I play audio from my phone it does not play through the Pi.
I have tried HDMI and audio jack.

Other people mentioned Kodi, but I can't see any Kodi installed.

Any help would be greatly appreciated!

TheJulianJES, 2017-09-26:
I have the same problem.

mdEpbH0n86, 2017-10-24:
This worked to get audio to play through the audio jack: 
`sudo amixer -c 0 cset numid=3 1`

BaReinhard, 2017-11-30:
This is because this installation is only meant for Jessie LITE, not w/ Pixel. Kodi only shows when on LITE.

I haven't spent much time trying to get this working on Desktop versions, but  it will likely require some small changes to the /etc/asound.conf file

BaReinhard, 2018-01-11:
This should now be fixed with the latest master branch. There is also another issue #102 which should work for stretch for those interested.

closed 2018-01-11 by BaReinhard