BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #53: Auto Connect?

Jitblass, 2017-09-17:
Anyone have a plan for how we could get this to auto-connect to the most recently used phone upon pi startup?  Love this project, and would love to see it add this capability next so I don't have to autoconnect every time I power up... I'm not sure where to go from here though.

Thoughts?

BaReinhard, 2017-09-18:
If you use the same phone each time you should be able to follow the instructions here:

https://raspberrypi.stackexchange.com/questions/44520/bluetooth-autoconnect-on-reboot-car-setup-with-scripts

I haven't actually used this in practice, but came up with the idea a while ago. It may need some playing around with the existing files, but should be relatively straight foward.

Jitblass, 2017-09-19:
Thanks, Brett!

My end goal to support multiple phones/devices, but I'll start here.  I suppose to achieve such a longterm goal I would need to write the mac address for the most recent device in attempt to autoconnect (of course.... I wonder if that means read only is not an option anymore....)

JT

BaReinhard, 2017-09-19:
It should be easy enough to make a list of supported MAC addresses and loop
Through them on boot see if one is in range or not. But yeah I think it
would need at least some write capabilities to dynamically add to a file.

Jitblass, 2017-09-26:
Okay.... one more...

I added to the file at "/etc/rc.local" the following line (I added it both before and after "exit 0" with the same results:

echo "connect AA:BB:CC:DD:EE:FF \nquit" | bluetoothctl

where AA:BB:CC:DD:EE:FF is the address for my iphone. I rebooted... and nothing changes. I have to still manually connect each time as I did before inputting the code. Anything I'm missing here?

Thanks!

JT

Hacksore, 2017-10-11:
@Jitblass try putting the -e flag when you invoke echo.

`echo -e "connect <mac> \nquit | bluetoothctl`

This is because we need to pass an escape char "\n" so you need to pass the `-e ` flag. Otherwise, bluetoothctl is interpreting `connect <mac> quit `as one command, which we know is not valid.

BaReinhard, 2018-01-11:
I’ve been thinking about this more, you could place a line in /usr/local/bin/bluez-udev to check if the bluetooth mac is in a file already, if not add it. Then have a script to poll for nearby devices and check said file for MAC address, if it’s available then connect. However, I think this would be a better topic on the rpi forum or other related forums. I doubt many interested parties will find this here. I’ll keep updating back if I find anything suitable on the subject, but for now I will be closing this issue.

closed 2018-01-11 by BaReinhard