BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #54: bt_pa_config.sh references nonexisting libjson package

kswtch, 2017-09-23:
In line 155 of bt_pa_config.sh you try to install <code>libjson0-dev</code> which is not available in the latest raspbian release (9.1). The package is now named <code>libjson-c-dev</code>.

BaReinhard, 2017-11-30:
Are you trying to install on Raspbian Jessie or Stretch?

Currently this isn't working with Stretch, I have yet to even try updating for Stretch. This will likely happen in the next month

BaReinhard, 2018-01-08:
Good news! I have got this working for raspbian stretch. Working on a final version and should be up tonight. Look for a new issue in regards to stretch support.

BaReinhard, 2018-01-08:
Alright, I have a stretch version that is in the testing phase at the moment, go [here](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/102) and follow the instructions. It will need to be on a new installation of Raspbian Stretch, (it has been tested on the LITE version, but i suspect it will work for Stretch w/ Pixel). Please leave feedback in issue #102 .

Please redirect all new issues in regards to raspbian stretch to issue #102

closed 2018-01-08 by BaReinhard