BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #59: Bluetooth Meta-data?

mrwulff, 2017-10-29:
I am using this in my car with a 5 inch screen.  Is it possible for me to show the mp3 tag info on this using something like pygame?

BaReinhard, 2017-11-30:
This is a good question, I like the idea of getting the meta data at the moment I am at a loss at how to do this. I will place it as a potential feature. If you do find a way to get it done and outline it either here or in a pull request I will gladly add it

bermudaparty, 2018-04-02:
I am also very interested in this. As I posted on another issue as well, the following post provides some python code that goes into that direction: https://www.raspberrypi.org/forums/viewtopic.php?p=635509#p635509

mrwulff, 2018-04-05:
awesome. now do album art. jk

dzbiku, 2018-04-19:
I still try to remake code from forum, and now on my bitbucket i did make little app with arguments- it allows to display temporary notifications on Raspbian os.: