BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #62: bluetooth paired, no audio

jedwards4b, 2017-11-12:
I have volumio 2.310 on an rpi with hifiberry dac+ 
I was able to establish a bluetooth pairing to my android phone, but don't get any audio out 
from the player.  

All my normal volumio functionality is still there, so the problem is just in connecting the audio path from bluetooth input to the output device (I think).     Can you provide any troubleshooting suggestions?

A (perhaps) related question - is there a way to add the new bluetooth input (assuming I get it working) to the web based volumio controller?

Thanks

Tuchos, 2017-11-13:
Try the home installation.

Notilix, 2017-11-20:
Got it work by downgrading to Jessie, you'd give it a try.

BaReinhard, 2017-11-30:
Unfortunately for now, this is only working on Raspbian Jessie. I haven't found to time to test with Raspbian Stretch. I believe there are some major breaking changes, but this is all speculation.

As far as adding to the volumio controller would be better suited for a Volumio forum, but it will largely depend on how the web based controller grabs inputs.

In a similar project I have a web based controller for multiroom output, it simply uses `pactl list sinks` to grab relevant outputs. If that is the case, there is nothing else that is required it should just show up once it is connected. If that isn't case I would speculate that there is some sort of configuration file.

closed 2017-11-30 by BaReinhard