BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #63: How to select soundcard

pubso, 2017-11-12:
I didn't get a list of soundcards during install at the beginning. I want to use USB soundcard. I chose to install Volumio bluetooth only option 4. I can pair my phone with Raspberry, but I can't hear any sound.

jkobti, 2017-11-27:
same for me, hope someone answers

BaReinhard, 2017-11-27:
Setting up a sound card is as easy as creating an asound.conf file let me check the source code real quick

BaReinhard, 2017-11-27:
So the file is /etc/asound.conf and will depend on what the output of your `aplay-l` shows

BaReinhard, 2017-11-27:
As a default the file should just need

```
pcm.pulse {
    type pulse
    card 1
}
ctl.pulse {
    type pulse
    card 1
}
pcm.!default {
    type hw
    card 1
}
ctl.!default {
    type hw
    card 1
}
```

jkobti, 2017-11-28:
thanks @BaReinhard

BaReinhard, 2017-11-28:
No problem, let me know if that works for you.

BaReinhard, 2017-11-30:
closing as it should be fixed, if not feel free to re open and we can fix this

closed 2017-11-30 by BaReinhard