BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #64: no documentation for AUX line input

ageorgios, 2017-11-21:
Hi,

I cannot understand what do you mean by AUX line input since raspberry pi does not have a Line-in input.

Please explain, I think it would be useful for me.

Thanks.

BaReinhard, 2017-11-21:
It requires an external usb soundcard with a mic input. The issue is that it will only playback in mono and not stereo. I haven't found a suitable solution for it. What is needed is that the mic input stream must be piped to the speaker output. I can look to see if I still have any documentation for it, I ended up scraping it due to lack of hardware to support it in stereo.

BaReinhard, 2017-11-30:
Once you have a working sound card for input and output you can simply use the following command
arecord -D plughw:1 -f dat | aplay -D plughw:1 -f dat&

this will take the audio from the aux line input on card 1 and output it to the output on card 1,

I used the sabrent audio card for this: [here](https://www.google.com/shopping/product/2306321428961959764?lsf=seller:6136318,store:1358544555752304294&prds=oid:7203634167783694949&q=sabrent+usb+audio+card&hl=en&ei=L1AgWrj0FOnR0gKerabQDQ&lsft=gclid:CjwKCAiA9f7QBRBpEiwApLGUisXUzZJlOZlIL-lWAXYO-O43vXecCI-IyYn57cktVsoNaw6oK4AkCBoCxfgQAvD_BwE)

closed 2017-11-30 by BaReinhard