BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #67: Clean Jesse Light - Script Fails

old-square-eyes, 2017-12-01:
I came back after weeks of testing because my issue got closed with promises of improvements. But just reporting that after an hour of installing Jesse and updating, the script ran for 30 minutes, late into the night and eventually failed and the pi rebooted. When it came back my Bluetooth device was not visible to my phone, but it was available to my Mac. The Mac paired but couldn't connect as it said the device didn't support the necessary features. Airplay is available but streams silence to my USB sound card. Do I have to start all over again?

BaReinhard, 2017-12-01:
You shouldn't need to. When did you run the installation? The script has been updated as of November 29th. That being said there are still some steps that can be taken to fix the issues if you have a slightly older version. 

#What I do need to help walk you through this is.

* What installation options you chose, e.g. Home, Car, Custom, etc...
* What the contents of your `/boot/config.txt` is
* What the output of `ps -ef | grep pulse` is
* What the contents of `/etc/asound.conf`
* What the out of `aplay -l` is

old-square-eyes, 2017-12-01:
Post updates. i.e. right before this thread was started. never mind I'm starting again Will report back.

old-square-eyes, 2017-12-01:
Sorry didn't mean to sound grumpy. Just had my hopes up after weeks of trying when I saw the updates :) I live the idea of this project and have purchased external BT and WiFi dongles to go with my M Box Micro to get it to work. Will try them one by one if I have issues.

old-square-eyes, 2017-12-01:
Same thing happened. SSH link went down. Looks ungracefully. BT was available but I couldn't pair even with my phone. This is my terminal session before it went down...

`===> Executing: make install
Making install in src
make[1]: Entering directory '/home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/src'
(echo "#define GM_COMPILE_VERSION \"2017-09-09_fa3baff\"" > git-version.h-new; \
cmp -s git-version.h git-version.h-new || cp git-version.h-new git-version.h; \
rm git-version.h-new)
make[2]: Entering directory '/home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/src'
(echo "#define GM_COMPILE_VERSION \"2017-09-09_fa3baff\"" > git-version.h-new; \
cmp -s git-version.h git-version.h-new || cp git-version.h-new git-version.h; \
rm git-version.h-new)
 /bin/mkdir -p '/usr/local/bin'
  /usr/bin/install -c gmediarender '/usr/local/bin'
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/src'
make[1]: Leaving directory '/home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/src'
Making install in data
make[1]: Entering directory '/home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/data'
make[2]: Entering directory '/home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/data'
make[2]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/usr/local/share/gmediarender'
 /usr/bin/install -c -m 644 grender-128x128.png grender-64x64.png '/usr/local/share/gmediarender'
make[2]: Leaving directory '/home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/data'
make[1]: Leaving directory '/home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/data'
make[1]: Entering directory '/home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect'
make[2]: Entering directory '/home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect'
make[1]: Leaving directory '/home/pi/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect'
--------------------------------------------
Ending at @ Fri  1 Dec 20:21:03 UTC 2017
packet_write_wait: Connection to fe80::4be9:6bf:eb3:58c2%en0 port 22: Broken pipe
square-eyes:~ square_eyes$ 
`

Is that finished? MIght be good for some output saying completed successfully rebooting if that's the case.

BaReinhard, 2017-12-01:
It’s fine I understand the frustration been banging my head trying to update for stretch. What install did you use? It looks like it’s trying to install gmedia, I haven’t tested that yet and ultimately am looking to remove it. 

I did a custom install to get Bluetooth and airplay and had it working 100% with the latest raspbian Jessie lite. 

I would suggest trying that and then if you need gmedia go to their github and follow their directions, until I can get around to fixing that issue. Sorry if it’s not what you were hoping for.

BaReinhard, 2017-12-01:
Also when the script finishes it should echo completed successfully it might not log properly over ssh I’ll make an update so it does.

old-square-eyes, 2017-12-02:
I have no clue what gmedia is. I flashed a clean Jesse light and ran the commands from the [Readme.md](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install) selecting the options for home installation, and external USB.

old-square-eyes, 2017-12-02:
OK tried that (custom with BT and Airplay). Walked away seemed to reboot. Fail to load pulse audio and fail to set discoverable 0000 on boot. Device shows up as airplay but doesn't work. No BT found on my phone.

old-square-eyes, 2017-12-02:
**/boot/config.txt**

https://pastebin.com/1msPx8TJ

**ps -ef | grep pulse**

```
pi        2595     1  8 04:46 ?        00:01:14 /usr/bin/pulseaudio --start --log-target=syslog
pi        2631  1102  0 05:01 pts/0    00:00:00 grep --color=auto pulse
```

**/etc/asound.conf**

```
pcm.pulse {
    type pulse
    card 1
}
ctl.pulse {
    type pulse
    card 1
}

pcm.!default {
    type hw
    card 1
}
ctl.!default {
    type hw
    card 1
}
```

**aplay -l**

```
card 1: M2 [Mbox 2], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

BaReinhard, 2017-12-02:
Change the card numbers in the asound.conf to 0. 

What does 
`sudo systemctl status pulseaudio` show?

BaReinhard, 2018-01-08:
closed due to inactivity, also update on those using Stretch:

Alright, I have a stretch version that is in the testing phase at the moment, go [here](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/102) and follow the instructions. It will need to be on a new installation of Raspbian Stretch, (it has been tested on the LITE version, but i suspect it will work for Stretch w/ Pixel). Please leave feedback in issue #102 .

closed 2018-01-08 by BaReinhard