BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #70: I like to use a Raspberry Zero W with an All miniBOSS Soundcard

jolandaanne, 2017-12-05:
Hi,

I have the latest Raspbian installed on a Raspberry Zero W. The Allo miniBOSS Soundcard is setup properly an can be accessed with hw-id 0.

Is it possible to use this soundcard in this great project ?

Best Jolanda-Anne

BaReinhard, 2017-12-05:
If you can view it with `aplay-l` it should be very easy to use. After the install just make sure that `dtparam=audio=on` is commented out in `/boot/config.txt` if you’re having trouble open and issue and we can walk you through the process.

jolandaanne, 2017-12-05:
Thanks for the swift reply !! `/boot/config.txt` is configured properly and I do see the soundcard with `play -l` :

```
root@raspberrypi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: BossDAC [BossDAC], device 0: Boss DAC HiFi [Master] pcm512x-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0  
```

With installation shall I walk through ? Just select on of the sound cards or non ?

BaReinhard, 2017-12-05:
Since the sound card is already installed I would suggest using a custom install, and choose n for snapcast, y for airplay and Bluetooth, and n for sound card. Then after the install process just ensure `dtparam=audio=on` is commented out in `/boot/config.txt` aside from that it should afterwards.

jolandaanne, 2017-12-05:
Installation (custom, no snap cast, airplay and bluetooth (name = "miniboss") and no sound card was successful. I can play music from my MacBook Air via shairport-sync. Thanks for that. But I can't see any bluetooth devices named "miniboss". I tried to pair from my Mac, my iPhone and my Amazon Echo Dot.

BaReinhard, 2017-12-06:
What raspbian are you running? Jessie/stretch lite or with pixel ?

jolandaanne, 2017-12-08:
Issue is now solved. I installed now the correct Raspbian version....

BaReinhard, 2017-12-08:
Glad it’s working for you now!

closed 2017-12-08 by BaReinhard