BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #71: Installation failed on Volumio with Raspberry Pi Zero W

Feserich, 2017-12-06:
Hey,
i've tried to install the Volumio (Bluetooth Only) Version, but it failed.
The device is rebooting after the quoted concole output and I can't find a visible bluetooth device after reboot.
I'm using the newest Volumio Version (2.323). 

I think mkdir failed because on Volumio the default user is 'volumio' not 'pi'. 
I can give you the complete console output, if you need it. 

```
===> Executing: adduser pulse audio
The user `pulse' is already a member of `audio'.
===> Executing: adduser root pulse-access
The user `root' is already a member of `pulse-access'.
===> Executing: adduser pulse lp
The user `pulse' is already a member of `lp'.
Done! You should reboot now
--------------------------------------------
===> Executing: su volumio -c ./bt_pa_config.sh
mkdir: cannot create directory '/home/pi/pyScripts': No such file or directory
===> Executing: sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
sudo: no tty present and no askpass program specified
Exiting script due to error from: sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
Exiting script due to error from: su volumio -c ./bt_pa_config.sh
--------------------------------------------
./install.sh: line 276: [: !=: unary operator expected
Ending at @ Wed Dec  6 17:50:55 UTC 2017
```

BaReinhard, 2017-12-06:
You’re correct. I can work on a fix for this today. 

When using volumio, and you use sudo does it prompt for user password?

I only ask because raspbian by default which could cause issues as well. My suggestion would be to go through line by line and run the commands or change the line from `pi` to `volumio` then re run bt_pa_config.sh as non root user until I can fix the script.

Feserich, 2017-12-06:
There's no hurry! 

I changed pi to volumio and the mkdir error is gone. 

It prompt for user password when I executing install.sh at the beginning (after: `sudo ./install.sh`)
After that it never prompt again for the password. 

When I execute the bt_pa_config.sh separately, it took a very long time and I had to enter the password a few times. After it was finished I also executed the bt_pa_install.sh script and rebooted my device. 
After reboot I also can't find the raspberry over Bluetooth. 

Do I have to execute another script? 
Should the raspberry be automatically visible for other devices or do I have to activate Bluetooth manually?

BaReinhard, 2017-12-06:
It should automatically be visible to you. It’s possible that one of the patch commands did not execute correctly. It looks like volumio will need some minor tweaks.

I would check the contents of 

/etc/bluetooth/main.conf . If you could show me your contents of that file it could help to diagnose the issue

Feserich, 2017-12-07:
Here is the content of main.conf:

```
[General]

# Default adaper name
# %h - substituted for hostname
# %d - substituted for adapter id
# Defaults to 'BlueZ'
#Name = %h-%d

# Default device class. Only the major and minor device class bits are
# considered. Defaults to '0x000000'.
#Class = 0x000100

# How long to stay in discoverable mode before going back to non-discoverable
# The value is in seconds. Default is 180, i.e. 3 minutes.
# 0 = disable timer, i.e. stay discoverable forever
#DiscoverableTimeout = 0

# How long to stay in pairable mode before going back to non-discoverable
# The value is in seconds. Default is 0.
# 0 = disable timer, i.e. stay pairable forever
#PairableTimeout = 0

# Automatic connection for bonded devices driven by platform/user events.
# If a platform plugin uses this mechanism, automatic connections will be
# enabled during the interval defined below. Initially, this feature
# intends to be used to establish connections to ATT channels. Default is 60.
#AutoConnectTimeout = 60

# Use vendor id source (assigner), vendor, product and version information for
# DID profile support. The values are separated by ":" and assigner, VID, PID
# and version.
# Possible vendor id source values: bluetooth, usb (defaults to usb)
#DeviceID = bluetooth:1234:5678:abcd

# Do reverse service discovery for previously unknown devices that connect to
# us. This option is really only needed for qualification since the BITE tester
# doesn't like us doing reverse SDP for some test cases (though there could in
# theory be other useful purposes for this too). Defaults to 'true'.
#ReverseServiceDiscovery = true

# Enable name resolving after inquiry. Set it to 'false' if you don't need
# remote devices name and want shorter discovery cycle. Defaults to 'true'.
#NameResolving = true

# Enable runtime persistency of debug link keys. Default is false which
# makes debug link keys valid only for the duration of the connection
# that they were created for.
#DebugKeys = false

#[Policy]
#
# The ReconnectUUIDs defines the set of remote services that should try
# to be reconnected to in case of a link loss (link supervision
# timeout). The policy plugin should contain a sane set of values by
# default, but this list can be overridden here. By setting the list to
# empty the reconnection feature gets disabled.
#ReconnectUUIDs=

```

BaReinhard, 2017-12-07:
The file should look like this

```
[General]

# Default adaper name
# %h - substituted for hostname
# %d - substituted for adapter id
# Defaults to 'BlueZ'
Name = BT_name_here

# Default device class. Only the major and minor device class bits are
# considered. Defaults to '0x000000'.
Class = 0x200414

# How long to stay in discoverable mode before going back to non-discoverable
# The value is in seconds. Default is 180, i.e. 3 minutes.
# 0 = disable timer, i.e. stay discoverable forever
DiscoverableTimeout = 0

# How long to stay in pairable mode before going back to non-discoverable
# The value is in seconds. Default is 0.
# 0 = disable timer, i.e. stay pairable forever
#PairableTimeout = 0

# Automatic connection for bonded devices driven by platform/user events.
# If a platform plugin uses this mechanism, automatic connections will be
# enabled during the interval defined below. Initially, this feature
# intends to be used to establish connections to ATT channels. Default is 60.
#AutoConnectTimeout = 60

# Use vendor id source (assigner), vendor, product and version information for
# DID profile support. The values are separated by ":" and assigner, VID, PID
# and version.
# Possible vendor id source values: bluetooth, usb (defaults to usb)
#DeviceID = bluetooth:1234:5678:abcd

# Do reverse service discovery for previously unknown devices that connect to
# us. This option is really only needed for qualification since the BITE tester
# doesn't like us doing reverse SDP for some test cases (though there could in
# theory be other useful purposes for this too). Defaults to 'true'.
#ReverseServiceDiscovery = true

# Enable name resolving after inquiry. Set it to 'false' if you don't need
# remote devices name and want shorter discovery cycle. Defaults to 'true'.
#NameResolving = true

# Enable runtime persistency of debug link keys. Default is false which
# makes debug link keys valid only for the duration of the connection
# that they were created for.
#DebugKeys = false

#[Policy]
#
# The ReconnectUUIDs defines the set of remote services that should try
# to be reconnected to in case of a link loss (link supervision
# timeout). The policy plugin should contain a sane set of values by
# default, but this list can be overridden here. By setting the list to
# empty the reconnection feature gets disabled.
#ReconnectUUIDs=
```

After that and a reboot you should see it

Feserich, 2017-12-08:
I changed the main.conf to your version and I rebooted, but I can't find the device.

adkp, 2017-12-08:
hi,
sudo hciconfig hci0 piscan
Can't get device info: No such device
So bluetooth agent don't start.

BaReinhard, 2017-12-09:
is there a default bluetooth adapter?

adkp, 2017-12-09:
Default bluetooth adapter 

sudo apt-get install -y libusb-dev libdbus-1-dev libglib2.0-dev libudev-dev libical-dev libreadline-dev libltdl-dev libsamplerate0-dev libsndfile1-dev libasound2-dev libavahi-client-dev libspeexdsp-dev liborc-0.4-dev intltool libtdb-dev libssl-dev libjson0-dev libsbc-dev libcap-dev

wget www.kernel.org/pub/linux/bluetooth/bluez-5.23.tar.xz
tar xvf bluez-5.23.tar.xz

cd bluez-5.23/
./configure --prefix=/usr --sysconfdir=/etc  --localstatedir=/var --enable-library --disable-systemd
sudo make
sudo make install
sudo install -v -dm755 /etc/bluetooth
sudo install -v -m644 src/main.conf /etc/bluetooth/main.conf

45 minutes later....

on boot
sudo /usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -

now
hciconfig
hci0:   Type: BR/EDR  Bus: UART
        BD Address: AA:AA:AA:AA:AA:AA  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING PSCAN ISCAN
        RX bytes:759 acl:0 sco:0 events:47 errors:0
        TX bytes:1555 acl:0 sco:0 commands:47 errors:0

It works :)

BaReinhard, 2017-12-09:
awesome! What version of bluez comes with Volumio ?

adkp, 2017-12-09:
I don't know the wersion comes with Volumio,but it's not stable :(

BaReinhard, 2017-12-10:
Interesting, also apparently hciconfig hci0 piscan has been deprecated. So adding
```
[Policy]
AutoEnable=true
```
To your `/etc/bluetooth/main.conf` is supposed to work the same. I haven’t tested it just yet but going to try it out in the coming days.

biva, 2017-12-13:
I followed the instructions of @adkp but I can't see my volumio bluetooth :(
Result of  `hciconfig` gives:
`hci0:   Type: BR/EDR  Bus: UART
        BD Address: AA:AA:AA:AA:AA:AA  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING PSCAN ISCAN
        RX bytes:752 acl:0 sco:0 events:46 errors:0
        TX bytes:1548 acl:0 sco:0 commands:46 errors:0
`
Any idea?
Config: RPI3 with volumio 2.323

Feserich, 2017-12-13:
I also followed the instructions of @adkp, but the installation of the libraries failed with the following output:
```
The following extra packages will be installed:
  libical1a libreadline6-dev libtinfo-dev
Suggested packages:
  liborc-0.4-doc readline-doc
The following NEW packages will be installed:
  libical-dev libical1a liborc-0.4-dev libreadline-dev libreadline6-dev libtinfo-dev libudev-dev libusb-dev
0 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
Need to get 65.4 kB/732 kB of archives.
After this operation, 3580 kB of additional disk space will be used.
Err http://archive.raspbian.org/raspbian/ jessie/main libtinfo-dev armhf 5.9+20140913-1
  404  Not Found
Unable to correct missing packages.
E: Failed to fetch http://archive.raspbian.org/raspbian/pool/main/n/ncurses/libtinfo-dev_5.9+20140913-1_armhf.deb  404  Not Found

E: Aborting install.
```
I've tried diffrent apt-get sources, but the installation also failed. 

`hciconfig` doesn't generate any output on my device (only next line for a new command).

adkp, 2017-12-13:
@Feserich maybe install  bluez-5.9 ? I think there is a probleme bluetooth by UART  with volumio

Feserich, 2017-12-18:
@adkp I tried your instructions from the last [post](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/71#issuecomment-350416641) again today and suddenly all worked fine. There was no problem during the libraries installation. 
So I installed bluez-5.23.

An error occurred at last command: 
```
sudo /usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -
bcm43xx_init
Cannot open directory '/etc/firmware': No such file or directory
Patch not found, continue anyway
Set Controller UART speed to 921600 bit/s
Device setup complete
```
after that, I've get the same output as @biva for `hciconfig`: 
```
hci0:   Type: BR/EDR  Bus: UART
        BD Address: AA:AA:AA:AA:AA:AA  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING PSCAN 
        RX bytes:731 acl:0 sco:0 events:43 errors:0
        TX bytes:1505 acl:0 sco:0 commands:43 errors:0
```
after `sudo hciconfig hci0 piscan` I can find the raspberry and pair successfully (I paired the two devices with `sudo bluetoothctl` before)
But I can't connect to the device and play music. 
After a reboot everything is gone. `hciconfig` gave me no output and I can't find my device. 
```
sudo hciconfig hci0 piscan
Can't get device info: No such device
```

BaReinhard, 2018-01-04:
@Feserich you may need to add a line before piscan

```
hciconfig hci0 up&
```

I am making a pull request to the master tonight for volumio, I have it working on my device, but it needs testing.

BaReinhard, 2018-01-04:
I have created a new branch for volumio users

```
# enable ssh through volumio.local/dev
# ssh to volumio

git clone https://github.com/bareinhard/super-simple-raspberry-pi-audio-receiver-install
cd super-simple-raspberry-pi-audio-receiver-install
git checkout volumio-fix
sudo ./install.sh

# Choose Bluetooth Option, #4

# Wait for it to complete, and enjoy
```

kekoa428, 2018-01-07:
@BaReinhard Install worked perfectly 4 days ago. Had to do a fresh wipe and now the install process (option 4)  is hanging. Is there a branch I can pull from 4 days ago? I noticed a few changes to the install script. It was working so wonderfully before!

BaReinhard, 2018-01-07:
Where does the install hang at?

kekoa428, 2018-01-07:
It goes through the entire Updating via Apt-Get. Then hangs in Upgrading via Apt-Get. Came back an hour later and it was still on this step.

BaReinhard, 2018-01-07:
what you can do is, removed `&> /dev/null` from the install script, https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/install.sh#L267

kekoa428, 2018-01-08:
I removed those lines. Of course it finished with no problems this time.. Thanks again! works like a charm

BaReinhard, 2018-01-08:
Glad it worked!! I’ll have to rethink the output of those commands. Thanks for the feedback!

closed 2018-01-08 by BaReinhard