BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #72: Feature request: Airplay password

gith-account, 2017-12-08:
I suggest to add the option to set up an Airplay password during install.

BaReinhard, 2017-12-08:
Yet another good idea, for the time being you can simply edit `/etc/shairport-sync.conf` and create a password there.

gith-account, 2017-12-09:
> you can simply edit /etc/shairport-sync.conf

Yup, that is what I did. :-) As regards the scripts I was thinking of this:

install.sh, add between lines 183 and 184:

```
if [ $AirPlay = "y" ]
then
	# Asks user if AirPlay password should be set
	AirPlaySecured="AirPlaySecured"
	AirPlayPass=""
	while [ $AirPlaySecured != "y" ] && [ $AirPlaySecured != "n" ];
	do
		read -p "Do you want to use an AirPlay password? (y/n) : " AirPlaySecured
	done
	# Prompts user for AirPlay password
	if [ $AirPlaySecured = "y" ]
	then
		read -p "Device AirPlay password: " AirPlayPass
	fi
fi
```

install.sh, change line 243 to:

```
	{ echo "${AirPlayName}"; echo "${SoundCard}"; echo "${AirPlayPass}";} | tst ./airplay_config.sh | tee -a $log
```

airplay_config.sh, add between lines 4 and 5:

```
read -p "AirPlay password: " AirPlayPass
```

install.sh, change lines 35 and 133 to:

```
if [AirPlayPass != ""]
then
	password = "$AirPlayPass";
fi
```

BaReinhard, 2017-12-09:
would you mind throwing that in a pull request? I would happily merge it.

gith-account, 2017-12-10:
I have submitted a pull request.

closed 2017-12-11 by gith-account