BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #73: Feature request: enclosed bluetooth audio environment

gith-account, 2017-12-08:
It suggest an option to install an enclosed bluetooth audio environment with customisable bluetooth PIN, and discoverable mode and pairable mode turned off unless temporarily activated with a script to allow for bonding new devices. When you have strangers within range of your bluetooth receiver (e.g., in a city apartment) an "open for all devices" approach is not desirable.

BaReinhard, 2017-12-08:
I was just thinking about this earlier this week. Thanks for making an issue on it, especially with a multiroom setup someone could start really messing with you. Ill be looking into this in the following weeks. If you accomplish this before hand feel free to make a pull request.

BaReinhard, 2017-12-08:
for the time being use this fix:

```
sudo nano /etc/init.d/bluetooth-agent

#remove the line:
sudo hciconfig hci0 piscan
sudo update-rc.d bluetooth-agent remove
sudo update-rc.d bluetooth-agent defaults
```

I believe this should work, then when you want to make it discoverable

use: `sudo hciconfig hci0 piscan` or 
```
sudo bluetoothctl
discoverable on
```

I haven't tested this, but I believe it should work

BaReinhard, 2018-01-12:
Ok, so I have figured this out. I have gotten the ability to pair the device with a given Pin code, but one caveat is that you need to input the pin within 4 seconds of the Number Input popping up. 

Given this shortcoming, it will not be implemented in this repo, but to enable this ability you need to do the following:

you need to edit the `/usr/local/bin/simple-agent.autotrust` on line : https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/blob/master/usr/local/bin/simple-agent.autotrust#L70
Change "0000" to whatever pin you want.

Then add the following before `exit 0`

```
hciconfig hci0 sspmode 0
sdptool add SP
```

Then pairing will work as expected, but you will have to be quick.

BaReinhard, 2018-01-12:
Ok, this still needs some work. The only way for this to work as of right now is if you reboot the system after every Bluetooth connection. Otherwise Bluetooth is not discoverable after disconnection of the Bluetooth device, but after the initial pair you can reconnect no issue.

BaReinhard, 2018-01-21:
@silelmot read the two above comments. It’s not perfect but does work. It will need some playing around to make it fully functional. For now I don’t have to time to get this feature completed

vmRelaxed, 2018-03-18:
You can add an fast on/off switch to not mess with your setup.

Lets make an on/off switch for Bluetooth (i know only bluetooth sry.)

`onoffbl`
`
#!/bin/sh
if  [[ $1 = "start" ]]; then
 bluetoothctl
discoverable on
exit
clear

else
    echo "No 'on' setting has been added"
fi
#!/bin/sh
if  [[ $1 = "stop" ]]; then
bluetoothctl
discoverable off
exit
clear

else
    echo "No 'off' setting has been added"
fi
`
Add this to 
nano onoffbl
Then do sudo chmod 777 onoffbl