BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install #74: Cannot paring to bluetooth

korte84, 2017-12-10:
Hello BAReinhard,

It's been a while i'm using your simple audio receiver and it's great! 
Here the problem:
I made a new install of it and i cannot paring my android phone (or any bluetooth device) anymore. 
the Rpi can be discovered but it's impossible to pair it.
On the BT status I figured out that the SAP drivers failed. Is that normal?

My config:  OS is raspbian jessie on a  Rpi3b + "HifiBerry DAC+ standard" 

here the BT status:
```
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled)
   Active: active (running) since Sun 2017-12-10 12:49:12 UTC; 3min 51s ago
     Docs: man:bluetoothd(8)
 Main PID: 861 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─861 /usr/lib/bluetooth/bluetoothd

Dec 10 12:49:12 raspberrypi bluetoothd[861]: Bluetooth daemon 5.23
Dec 10 12:49:12 raspberrypi systemd[1]: Started Bluetooth service.
Dec 10 12:49:12 raspberrypi bluetoothd[861]: Starting SDP server
Dec 10 12:49:12 raspberrypi bluetoothd[861]: Bluetooth management interface 1.14 initialized
Dec 10 12:49:12 raspberrypi bluetoothd[861]: Sap driver initialization failed.
Dec 10 12:49:12 raspberrypi bluetoothd[861]: sap-server: Operation not permitted (1)
```

BaReinhard, 2017-12-10:
That’s odd,

What does your
`/etc/init.d/bluetooth` file look like?

korte84, 2017-12-10:
thx for answering!! 
here it is:

```
#!/bin/sh -e
### BEGIN INIT INFO
# Provides:            bluetooth
# Required-Start:      $local_fs $syslog dbus
# Required-Stop:       $local_fs $syslog
# Default-Start:       2 3 4 5
# Default-Stop:        0 1 6
# Short-Description:   Starts bluetooth daemons
### END INIT INFO

. /lib/lsb/init-functions

DESC=bluetoothd
DAEMON=/usr/libexec/bluetooth/bluetoothd
#SSD_OPTIONS="--oknodo --quiet --exec $DAEMON --plugin=a2dp"
SSD_OPTIONS="--oknodo --quiet --exec $DAEMON" #Change to this if you want media control using DBus at the expense of volume contro$
HCI=hci0

case "${1}" in
    start)
       log_daemon_msg "Starting Bluetooth daemon bluetoothd..."
       start-stop-daemon --start --background $SSD_OPTIONS
       log_progress_msg "${DAEMON}"

       hciconfig $HCI up > /dev/null 2>&1
       log_end_msg 0
       ;;

    stop)
        log_daemon_msg "Stopping Bluetooth daemon bluetoothd..."
        start-stop-daemon --stop $SSD_OPTIONS
        log_progress_msg "${DAEMON}"
        log_end_msg 0
       ;;

    restart)
       ${0} stop
       sleep 1
       ${0} start
       ;;

    status)
        status_of_proc "$DAEMON" "$DESC" && exit 0 || exit $?
       ;;

    *)
         echo "Usage: ${0} {start|stop|restart|status}"
         exit 1
       ;;
esac

exit 0
```

korte84, 2017-12-10:
in this moment i can even detect the bluetooth

BaReinhard, 2017-12-11:
What is the contents of /etc/init.d/pulseaudio

And if you could also post your contents of install.log inside the sspari folder

BaReinhard, 2017-12-11:
Also the contents of /etc/init.d/bluetooth-agent would be helpful

korte84, 2017-12-11:
/etc/init.d/bluetooth-agent

```
#!/bin/sh -e
### BEGIN INIT INFO
# Provides: bluetooth-agent
# Required-Start: $remote_fs $syslog bluetooth pulseaudio # Required-Stop: $remote_fs $syslog # Default-Start: 2 3 4 5
# Default-Stop: 0 1 6
# Short-Description: Makes Bluetooth discoverable and connectable to 00$# Description: Start Bluetooth-Agent at boot time.
### END INIT INFO
#! /bin/sh
# /etc/init.d/bluetooth-agent
USER=root
HOME=/root
export USER HOME
case "$1" in start)
echo "setting bluetooth discoverable"
sleep 1 sudo hciconfig hci0 piscan
start-stop-daemon -S -x /usr/local/bin/simple-agent.autotrust &
echo "bluetooth-agent started" ;;
stop)
echo "Stopping bluetooth-agent"
start-stop-daemon -K -x /usr/local/bin/simple-agent.autotrust ;; *) echo "Usage: /etc/init.d/bluetooth-agent {start|stop}"
exit 1 ;;
esac exit 0

```
/etc/init.d/pulseaudio

```
#!/bin/sh -e
### BEGIN INIT INFO
# Provides: pulseaudio esound # Required-Start: $remote_fs $syslog # Required-Stop: $remote_fs $syslog
# Should-Start: udev network-manager
# Should-Stop: udev network-manager # Default-Start: 2 3 4 5
# Default-Stop: 0 1 6
# Short-Description: Start the PulseAudio sound server
# Description: System mode startup script for
# the PulseAudio sound server.
### END INIT INFO
DAEMON=/usr/local/bin/pulseaudio PIDDIR=/var/run/pulse
PIDFILE=$PIDDIR/pid
DAEMONUSER=pulse PATH=/sbin:/bin:/usr/sbin:/usr/bin
test -x $DAEMON || exit 0 
. /lib/lsb/init-functions
pulseaudio_start () { log_daemon_msg "Starting system PulseAudio Daemon" if [ ! -d $PIDDIR ]; then mkdir -p $PIDDIR
chown $DAEMONUSER:$DAEMONUSER $PIDDIR fi
start-stop-daemon -x $DAEMON -p $PIDFILE --start -- --system --$ status=$?
if [ -e /var/run/pulse/.esd_auth ]; then
chown pulse:pulse-access /var/run/pulse/.esd_auth
chmod 640 /var/run/pulse/.esd_auth fi if [ -e /var/run/pulse/.pulse-cookie ]; then
chown pulse:pulse-access /var/run/pulse/.pulse-cookie
chmod 640 /var/run/pulse/.pulse-cookie
fi
log_end_msg ${status} } pulseaudio_stop () { log_daemon_msg "Stopping system PulseAudio Daemon" start-stop-daemon -p $PIDFILE --stop --retry 5 || echo -n "...w$ log_end_msg $?
}
case "$1" in
start|stop)
pulseaudio_${1}
;; restart|reload|force-reload)
if [ -s $PIDFILE ] && kill -0 $(cat $PIDFILE) >/dev/nul$ pulseaudio_stop
pulseaudio_start fi
;;
force-stop)
pulseaudio_stop
killall pulseaudio || true
sleep 2
killall -9 pulseaudio || true
;; status)
status_of_proc -p $PIDFILE "$DAEMON" "system-wide Pulse$ ;; *)
echo "Usage: /etc/init.d/pulseaudio {start|stop|force-s$ exit 1 ;;
esac
exit 0 

```

BaReinhard, 2017-12-11:
Ok those both look good.

 Can you post the contents of ~/super-simple-raspberry-Pi-audio-receiver-install/install.log.

And what is the user you installed this under?

korte84, 2017-12-11:
1) It's on 2 lines
2) user: root (sudo -i)
3) install.log
```
0-0:armhf (1.4.4-2.1+deb8u2) ...
Setting up libkate1 (0.4.1-4) ...
Setting up libmimic0 (1.0.4-2.2) ...
Setting up libmodplug1 (1:0.8.8.4-4.1) ...
Setting up libofa0 (0.9.3-7) ...
Setting up libspandsp2:armhf (0.0.6-2) ...
Setting up libsrtp0 (1.4.5~20130609~dfsg-1.1+deb8u1) ...
Setting up libzbar0 (0.10+doc-10+b2) ...
Setting up gstreamer1.0-plugins-base:armhf (1.4.4-2+deb8u1) ...
Setting up gstreamer1.0-plugins-bad:armhf (1.4.4-2.1+deb8u2) ...
Setting up gir1.2-gstreamer-1.0 (1.4.4-2+deb8u1) ...
Setting up gstreamer1.0-alsa:armhf (1.4.4-2+deb8u1) ...
Setting up gstreamer1.0-libav:armhf (1.4.4-2) ...
Setting up gstreamer1.0-plugins-good:armhf (1.4.4-2+deb8u3) ...
Setting up liba52-0.7.4 (0.7.4-17) ...
Setting up libsidplay1 (1.36.59-6) ...
Setting up libtwolame0 (0.3.13-1.1) ...
Setting up gstreamer1.0-plugins-ugly:armhf (1.4.4-2+deb8u1) ...
Setting up libgstreamer1.0-dev (1.4.4-2+deb8u1) ...
Setting up libupnp6-dev (1:1.6.19+git20141001-1+deb8u1) ...
Setting up libupnp-dev (1:1.6.19+git20141001-1+deb8u1) ...
Processing triggers for libc-bin (2.19-18+deb8u10) ...
===> Executing: git clone https://github.com/hzeller/gmrender-resurrect.git
===> Executing: cd gmrender-resurrect
===> Executing: ./autogen.sh
===> Executing: ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... 64
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking for gcc option to accept ISO C99... -std=gnu99
checking for gcc -std=gnu99 option to accept ISO Standard C... (cached) -std=gnu99
checking whether make sets $(MAKE)... (cached) yes
checking for asprintf... yes
checking for exp in -lm... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for GLIB... yes
checking for GST... yes
checking for LIBUPNP... yes
checking how to run the C preprocessor... gcc -std=gnu99 -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
[... 62 lines cut ...]
(echo "#define GM_COMPILE_VERSION \"2017-12-10_704b182\"" > git-version.h-new; \
cmp -s git-version.h git-version.h-new || cp git-version.h-new git-version.h; \
rm git-version.h-new)
 /bin/mkdir -p '/usr/local/bin'
  /usr/bin/install -c gmediarender '/usr/local/bin'
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/root/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/src'
make[1]: Leaving directory '/root/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/src'
Making install in data
make[1]: Entering directory '/root/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/data'
make[2]: Entering directory '/root/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/data'
make[2]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/usr/local/share/gmediarender'
 /usr/bin/install -c -m 644 grender-128x128.png grender-64x64.png '/usr/local/share/gmediarender'
make[2]: Leaving directory '/root/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/data'
make[1]: Leaving directory '/root/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect/data'
make[1]: Entering directory '/root/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect'
make[2]: Entering directory '/root/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/root/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect'
make[1]: Leaving directory '/root/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/gmrender-resurrect'
--------------------------------------------
Ending at @ Sun 10 Dec 22:44:39 UTC 2017
```

BaReinhard, 2017-12-12:
ok, I think I may know whats happening. What is the output ok `ls /home/pi/`

korte84, 2017-12-12:
pi@raspberrypi:~ $ ls /home/pi/
json-c  libsndfile  pulseaudio  pyScripts

BaReinhard, 2017-12-12:
What you can do is navigate to the super-simple-.... folder and run bt_config.sh as your login user. Make sure it doesn’t say any errors and it should fix the issue.

korte84, 2017-12-13:
Hi sorry for the late replay
I've tried to enter the folder but i didn't have permissions, I think because I've installed your package as root. I ran bt_pa_config as root and i didn't have any error but nothing has changed.
I decided to load a fresh copy of Jessie lite. I installed the home version as pi user, and now I can detect the BT. The problem now is that I can pair the raspberry but the connection is not kept. 

NEW BT status 
```
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled)
   Active: active (running) since Tue 2017-12-12 19:57:44 UTC; 21h ago
     Docs: man:bluetoothd(8)
 Main PID: 954 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─954 /usr/lib/bluetooth/bluetoothd

Dec 12 19:57:44 raspberrypi systemd[1]: Starting Bluetooth service...
Dec 12 19:57:44 raspberrypi bluetoothd[954]: Bluetooth daemon 5.23
Dec 12 19:57:44 raspberrypi systemd[1]: Started Bluetooth service.
Dec 12 19:57:44 raspberrypi bluetoothd[954]: Starting SDP server
Dec 12 19:57:44 raspberrypi bluetoothd[954]: Bluetooth management interface 1.14 initialized
Dec 12 19:57:44 raspberrypi bluetoothd[954]: Sap driver initialization failed.
Dec 12 19:57:44 raspberrypi bluetoothd[954]: sap-server: Operation not permitted (1)
Dec 12 21:44:07 raspberrypi bluetoothd[954]: GAP and GATT are mandatory
Dec 12 21:44:07 raspberrypi bluetoothd[954]: gap-gatt-profile profile probe failed for BC:E6:3F:B6:12:96
```

File: /etc/init.d/bluetooth

```#!/bin/sh -e
### BEGIN INIT INFO
# Provides:            bluetooth
# Required-Start:      $local_fs $syslog dbus
# Required-Stop:       $local_fs $syslog
# Default-Start:       2 3 4 5
# Default-Stop:        0 1 6
# Short-Description:   Starts bluetooth daemons
### END INIT INFO

. /lib/lsb/init-functions

DESC=bluetoothd
DAEMON=/usr/libexec/bluetooth/bluetoothd
#SSD_OPTIONS="--oknodo --quiet --exec $DAEMON --plugin=a2dp"
SSD_OPTIONS="--oknodo --quiet --exec $DAEMON" #Change to this if you want media control using DBus at the expense of volume control 
HCI=hci0

case "${1}" in
    start)
       log_daemon_msg "Starting Bluetooth daemon bluetoothd..."
       start-stop-daemon --start --background $SSD_OPTIONS
       log_progress_msg "${DAEMON}"

       hciconfig $HCI up > /dev/null 2>&1
       log_end_msg 0
       ;;

    stop)
        log_daemon_msg "Stopping Bluetooth daemon bluetoothd..."
        start-stop-daemon --stop $SSD_OPTIONS
        log_progress_msg "${DAEMON}"
        log_end_msg 0
       ;;

    restart)
       ${0} stop
       sleep 1
       ${0} start
       ;;

    status)
[... 102 lines cut ...]
                ;;
        restart|reload|force-reload)
                if [ -s $PIDFILE ] && kill -0 $(cat $PIDFILE) >/dev/null 2>&1; then
                        pulseaudio_stop
                        pulseaudio_start
                fi
                ;;
        force-stop)
                pulseaudio_stop
                killall pulseaudio || true
                sleep 2
                killall -9 pulseaudio || true
                ;;
        status)
                status_of_proc -p $PIDFILE "$DAEMON" "system-wide PulseAudio" && exit 0 || exit $?
                ;;
        *)
                echo "Usage: /etc/init.d/pulseaudio {start|stop|force-stop|restart|reload|force-reload|status}"
                exit 1
                ;;
esac

exit 0

```

BaReinhard, 2017-12-13:
with the new installation what is the output of

```
pulseaudio --version
```

korte84, 2017-12-13:
5.0

BaReinhard, 2017-12-14:
Ok, so that is the issue, your system should be seeing pulseaudio 6.0, this means it didn't get compiled for one reason or another. So for some reason it is not getting through bt_pa_config.sh .

I would try running ./bt_pa_config.sh as a non-root user, pi. No need for sudo, the command should look like:

```
./bt_pa_config.sh
```
of course youll need to be in the super-simple-raspberry-pi-audio-receiver-install directory first.

Let me know what the output is and if any errors come up.

korte84, 2017-12-14:
Ok I have one error at the end 
```
configure: error: Unable to find libltdl version 2. Makes sure you have libtool 2.4 or later installed.
```

and this one 

`configure.ac:106: error: required file 'build-aux/config.rpath' not found`

```

pi@raspberrypi:~/Super-Simple-Raspberry-Pi-Audio-Receiver-Install $ ./bt_pa_config.sh
Bluetooth device name: AMPLI_HK
mkdir: cannot create directory ‘/home/pi/pyScripts’: File exists
===> Executing: sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
===> Executing: sudo chmod +x /usr/local/bin/volume-watcher.py
===> Executing: sudo cp lib/systemd/system/volume-watcher.service /lib/systemd/system/volume-watcher.service
===> Executing: sudo systemctl enable volume-watcher
===> Executing: cd .
===> Executing: sudo cp /tmp/machine-info /etc
===> Executing: sudo cp init.d/pulseaudio /etc/init.d
===> Executing: sudo chmod +x /etc/init.d/pulseaudio
===> Executing: sudo update-rc.d pulseaudio defaults
===> Executing: sudo cp init.d/bluetooth /etc/init.d
===> Executing: sudo chmod +x /etc/init.d/bluetooth
===> Executing: sudo update-rc.d bluetooth defaults
===> Executing: sudo cp init.d/bluetooth-agent /etc/init.d
===> Executing: sudo chmod +x /etc/init.d/bluetooth-agent
===> Executing: sudo update-rc.d bluetooth-agent defaults
===> Executing: sudo cp usr/local/bin/bluez-udev /usr/local/bin
===> Executing: sudo chmod 755 /usr/local/bin/bluez-udev
===> Executing: sudo cp usr/local/bin/simple-agent.autotrust /usr/local/bin
===> Executing: sudo chmod 755 /usr/local/bin/simple-agent.autotrust
===> Executing: sudo cp usr/local/bin/say.sh /usr/local/bin
===> Executing: sudo chmod 755 /usr/local/bin/say.sh
===> Executing: sudo cp usr/local/bin/bluezutils.py /usr/local/bin
===> Executing: sudo cp etc/pulse/daemon.conf /etc/pulse
./bt_pa_config.sh: line 66: /boot/config.txt: Permission denied
patching file /etc/udev/rules.d/99-com.rules
Hunk #1 succeeded at 1 with fuzz 1.
===> Executing: sudo chmod 644 /etc/udev/rules.d/99-com.rules
patching file /etc/bluetooth/main.conf
Hunk #1 succeeded at 7 with fuzz 1.
Hunk #2 succeeded at 12 with fuzz 1.
Hunk #3 FAILED at 17.
1 out of 3 hunks FAILED -- saving rejects to file /etc/bluetooth/main.conf.rej
patching file /etc/pulse/system.pa
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file /etc/pulse/system.pa.rej
mkdir: cannot create directory ‘/etc/pulsebackup’: File exists
fatal: destination path 'pulseaudio' already exists and is not an empty directory.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libsndfile1-dev' instead of 'libsndfile-dev'
intltool is already the newest version.
libavahi-client-dev is already the newest version.
libcap-dev is already the newest version.
libglib2.0-dev is already the newest version.
libjson0-dev is already the newest version.
libsamplerate0-dev is already the newest version.
libsbc-dev is already the newest version.
libsndfile1-dev is already the newest version.
libspeexdsp-dev is already the newest version.
libssl-dev is already the newest version.
libtdb-dev is already the newest version.
libtool is already the newest version.
libasound2-dev is already the newest version.
[... 894 lines cut ...]
checking whether a statically linked program can dlopen itself... no
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking ltdl.h usability... no
checking ltdl.h presence... no
checking for ltdl.h... no
configure: error: Unable to find libltdl version 2. Makes sure you have libtool 2.4 or later installed.
make: *** No rule to make target 'clean'.  Stop.
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.
```

totti2, 2017-12-14:
1st: thx for the effort! I'd love to get this running again!

I didn't skip the patches
`./bt_pa_config.sh`
...
`patching file /etc/bluetooth/main.conf`
`Reversed (or previously applied) patch detected!  Assume -R? [n]`
`Apply anyway? [n] y`
`Hunk #1 succeeded at 7 with fuzz 1.`
`Hunk #2 succeeded at 12 with fuzz 1.`
`Hunk #3 FAILED at 17.`
`1 out of 3 hunks FAILED -- saving rejects to file /etc/bluetooth/main.conf.rej`
`patching file /etc/pulse/system.pa`
`Reversed (or previously applied) patch detected!  Assume -R? [n] `
`Apply anyway? [n] y`
`Hunk #1 FAILED at 23.`
`Hunk #2 succeeded at 57 with fuzz 1.`
`1 out of 2 hunks FAILED -- saving rejects to file /etc/pulse/system.pa.rej`

pulseaudio gets installed and is version 6.0 
but still no pairing

BaReinhard, 2017-12-15:
Just to be sure, you have rebooted afterwards?

Can you view your raspberry pi from the Bluetooth menu on your device?

BaReinhard, 2017-12-15:
Additionally, any missing dependencies will need to be added either manually or from apt-get.

What disk image is being used? Raspbian Jessie or stretch? And lite or with pixel?

totti2, 2017-12-15:
my setup:
rpi3, **2017-07-05-raspbian-jessie.img**, no snd-card, no additional bluetooth-dongle

pairing works for me with these install-options
```
Do you want to install SnapCast? (y/n): n
Do you want AirPlay Enabled? (y/n) : y
Do you want Bluetooth A2DP Enabled? (y/n) : y
Do you want to setup as an Access Point? (Necessary for AirPlay, in location without a Wireless Network) (y/n) : n
Do you want Kodi installed? (y/n) : n
Do you want to use infrared remotes? (y/n) : n
Do you want to use a Sound Card? (y/n) : n
Do you want to setup device as a UPnP Renderer? (y/n) : n
Do you want all the Devices to use the same name? (y/n) : y
Device name: MultiPi
Do you want to use an AirPlay password? (y/n) : n
```
before, I was insisting on SnapCast. the issue might be lying there.
I'll try **jessie lite** without SnapCast, maybe we've found the culprit

totti2, 2017-12-15:
I flashed a jessie **lite**, same setup, same install options:
pairing works
very low volume though, but that's due to the known issues to the PI's onboard sound-card, I assume.

I will reflash it with the snapcast-server option
I have an Android Phone

BaReinhard, 2017-12-15:
I'm assuming that may be correct. Ill give that a shot right now as well and see what the issue may be.

totti2, 2017-12-15:
no issue with same options as given above plus snapserver option.
last test would be snapserver+snapclient.

btw:
after my first comment in this thread i reflashed my sd-card...which I forgot to mention
adapted my comment accordingly

BaReinhard, 2017-12-15:
thanks for the heads up. I am currently testing the snapserver + snapclient.

BaReinhard, 2017-12-16:
found the issue with SAP drivers, it looks like python-dbus was missing from the dependencies,

fix this via:

sudo apt-get install python-dbus

korte84, 2017-12-16:
Yes I rebooted every time i did any modification.
The operating sys is Jessie lite (updated to the last version)
hifiberry dac standard

```
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

then i'm gonna install dependencies.
Is kodi mandatory for your package ?

korte84, 2017-12-16:
what's this?
````
===> Executing: sudo cp etc/pulse/daemon.conf /etc/pulse
./bt_pa_config.sh: line 66: /boot/config.txt: Permission denied
patching file /etc/udev/rules.d/99-com.rules
Hunk #1 succeeded at 1 with fuzz 1.
===> Executing: sudo chmod 644 /etc/udev/rules.d/99-com.rules
patching file /etc/bluetooth/main.conf
Hunk #1 succeeded at 7 with fuzz 1.
Hunk #2 succeeded at 12 with fuzz 1.
Hunk #3 FAILED at 17.
1 out of 3 hunks FAILED -- saving rejects to file /etc/bluetooth/main.conf.rej
patching file /etc/pulse/system.pa
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
```

korte84, 2017-12-16:
`configure: error: Unable to find libltdl version 2. Makes sure you have libtool 2.4 or later installed.`

version 7 is installed

BaReinhard, 2017-12-16:
@korte84 first error was that it was unable to patch system.pa, you can manually add the lines in /etc/pulse/system.pa.rej to /etc/pulse/system.pa

And no kodi isn’t necessary. I would do a custom install and say no to kodi if you don’t want it. Or simply remove it sudo apt-get purge kodi (not sure that works, or if you need to use remove in place of purge)

It looks like there are missing dependencies for libtool. I haven’t ever seen that so I would try installing sudo apt-get install libtool

totti2, 2017-12-16:
Pairing possible here with snapserver and snapclient enabled 
This is what I did after flashing the sd-card with 2017-07-05-raspbian-jessie-lite
```
    1  sudo raspi-config 
    2  sudo apt-get update 
    3  sudo apt-get upgrade
    4  sudo apt-get install git
    5  git clone https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install.git
    6  cd Super-Simple-Raspberry-Pi-Audio-Receiver-Install/
    7  sudo ./install.sh 
```
in `raspi-config` I just did the usual:
local time, memory split, expand filesystem 
@korte84 Did you change any boot option, like `auto-login in cli`

python-dbus was installed already
sap drivers still not initializing with latest version of python-dbus
```
pi@multipi:~ $ sudo apt-get install python-dbus
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-dbus is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

pi@multipi:~ $ apt-cache policy python-dbus
python-dbus:
  Installed: 1.2.0-2+b1
  Candidate: 1.2.0-2+b1
  Version table:
 *** 1.2.0-2+b1 0
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status

pi@multipi:~ $ sudo journalctl |grep bluet
Dec 16 18:26:11 multipi bluetoothd[645]: Bluetooth daemon 5.23
Dec 16 18:26:11 multipi bluetoothd[645]: Starting SDP server
Dec 16 18:26:11 multipi bluetooth-agent[648]: setting bluetooth discoverable
Dec 16 18:26:11 multipi bluetoothd[645]: Bluetooth management interface 1.14 initialized
Dec 16 18:26:11 multipi bluetoothd[645]: Sap driver initialization failed.
Dec 16 18:26:11 multipi bluetoothd[645]: sap-server: Operation not permitted (1)
Dec 16 18:26:11 multipi bluetoothd[645]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSource
Dec 16 18:26:11 multipi bluetoothd[645]: Endpoint registered: sender=:1.3 path=/MediaEndpoint/A2DPSink
Dec 16 18:26:12 multipi bluetooth-agent[648]: bluetooth-agent started
Dec 16 18:29:04 multipi bluetoothd[645]: GAP and GATT are mandatory
Dec 16 18:29:04 multipi bluetoothd[645]: gap-gatt-profile profile probe failed for 9C:02:98:AF:AE:86
```

BaReinhard, 2017-12-16:
@totti2 Maybe what fixed it for me was,

```
sudo update-rc.d bluetooth enable
```

I went through and removed all system daemons and enabled them again. After that SAP drivers had no errors

korte84, 2017-12-17:
Q: there's a way to reset the bt? Some time it's discovered avec some old names...

@totti2 not at my knowledge
@bareinhard I will reinstall libtool as I can access my pc. 

Ps. I think I'm gonna buy a bt dongle

totti2, 2017-12-17:
I reflashed without Snapcast but with usb-soundcard
No pairing

renamed `/etc/asound.conf`
pairing and playing music **through 3.5mm jack**
is the content of `asound.conf` in `sound_card_install.sh` correct? 

@korte84 I am succesfully using this with and without extra dongle. I use this one
```
pi@multipi:~ $ lsusb
Bus 001 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
```
@BaReinhard how exactly did you reenable the daemons? what do you mean by 
> removed all system daemons and enabled them again. 
doesn't that mean, that the order of initialization of daemons has to be reviewed?

btw: in `sound_card_install.sh` line 6
`if [ -z "$SoundCard ]` is there a `"` missing?

korte84, 2017-12-17:
totti2 i used to 😢 then it stopped. Sometime I discovered the pi with old names that I used or with "raspberrypi " (never used) but never with the actual name. With airplay + hi-fi berry I don't have any kind of problem but I have to use my Mac...

totti2, 2017-12-17:
delete the list of paired devices in your phone and pc/laptop, turn off/on bluetooth on your phone and pc/laptop/mac, then reboot it... sounds desperate, but bluetooth is rather black magic than true science. anyways: fixed it for me...sometimes

korte84, 2017-12-17:
Ahahah I'll try!

korte84, 2017-12-17:
`
pi@raspberrypi:~ $ sudo bluetoothctl 
[NEW] Controller B8:27:EB:A4:E0:25 AMPLI_HK [default]
[NEW] Device A8:BB:CF:20:FB:89 MacBook Pro
[bluetooth]# agent on
Agent registered
[bluetooth]# default-agent 
Default agent request successful
[bluetooth]# scan on
Failed to start discovery: org.bluez.Error.NotReady 
`
any ideas???

totti2, 2017-12-17:
How does your `/etc/asound.conf` look like?
After running `sound_card_install.sh` I had some troubles.
Plus, when I disabled onboard sound in `/Boot/config.txt` by setting `dtparam=audio=off` pairing was broken again.
Maybe dig deeper into alsa-configs?!

BaReinhard, 2017-12-17:
take a look at /usr/local/bin/bluez-udev you may need to change the default sink.

If you disable onboard sound, you may need to change your /etc/asound.conf to use card 0 instead of card 1.

korte84, 2017-12-17:
@totti2 and @BaReinhard I disabled onboard card and changed asound.conf using 0 instead of 1
it looks like:
````
cm.pulse {
    type pulse
    card 0
}
ctl.pulse {
    type pulse
    card 0
}

pcm.!default {
    type hw
    card 0
}
ctl.!default {
    type hw
    card 0
}

````

@BaReinhard what is the default sink?
[bluez-udev.txt](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/files/1566186/bluez-udev.txt)

BaReinhard, 2017-12-18:
@korte84 I would try tracking down the bug this way.

```
sudo systemctl stop bluetooth-agent
simple-agent.autotrust&
# Connect your device, see if any errors occur.
sudo pactl list sinks
sudo pactl list sink-inputs
```
Let us know what you find.

korte84, 2017-12-18:
ok i can pair my android phone but I cannot connect.
```
pi@raspberrypi:~ $ simple-agent.autotrust&
[1] 19645
pi@raspberrypi:~ $ Agent registered

pi@raspberrypi:~ $ sudo pactl list sinks
Connection failure: Connection refused
pa_context_connect() failed: Connection refused
pi@raspberrypi:~ $ sudo pactl list sink-inputs
Connection failure: Connection refused
pa_context_connect() failed: Connection refused
pi@raspberrypi:~ $ 

```
at the blank line it waits and i have to press enter to continue ...

totti2, 2017-12-18:
He's talking about [this](https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/6#issuecomment-284573272)
but this going to work after successful pairing only.

Have a look at [this](https://gist.github.com/totti2/41ed90feb1eb5838cc6a789d2b2dd5a7) .
It's a bit more than you need, but works for me. and it's extendable.

BaReinhard, 2017-12-18:
@korte84 The failure of pactl is more telling than anything. You will need to ensure that pulseaudio is running as a daemon. As of right now it is not started.

```
sudo update-rc.d pulseaudio remove
sudo systemctl disable pulseaudio
sudo update-rc.d pulseaudio defaults
sudo update-rc.d pulseaudio enable
sudo reboot
```
Before trying to connect log back in and see if pulseaudio is running
`sudo systemctl status pulseaudio`
Then try connecting connecting to the Pi, but before you do forget the PI on your device.

korte84, 2017-12-18:
```pi@raspberrypi:~ $ sudo update-rc.d pulseaudio remove
insserv: Service pulseaudio has to be enabled to start service bluetooth-agent
insserv: exiting now!
update-rc.d: error: insserv rejected the script header```

BaReinhard, 2017-12-18:
Can you list the full log

`sudo systemctl status pulseaudio -l`

korte84, 2017-12-19:
```
pi@raspberrypi:~ $ sudo systemctl status pulseaudio -l
● pulseaudio.service - LSB: Start the PulseAudio sound server
   Loaded: loaded (/etc/init.d/pulseaudio)
   Active: active (exited) since Mon 2017-12-18 23:44:18 UTC; 11h ago
  Process: 587 ExecStart=/etc/init.d/pulseaudio start (code=exited, status=0/SUCCESS)

Dec 18 23:44:18 raspberrypi systemd[1]: Started LSB: Start the PulseAudio sound server.

```

i found out, doing a `sudo systemctl` that the `● bluetooth-agent.service                                                                                 loaded failed failed    LSB: Makes Bluetooth discoverable and connectable to 0000` has failed

totti2, 2017-12-19:
Please allow me a shot in the dark:
Just to be sure: did you use a Raspbian *Jessie*?

Please don't get me wrong:
Have you ever since reflashed the OS and tried from the beginning?
Since it's working for me, it's probably the better/faster way.

korte84, 2017-12-19:
Don't worry ;) yes it's jessie
```
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
I think I'm gonna close this issue because I feel like  I'm wasting your time.
I really thank you for the super reactive support and for the time you spent to try to solve the problem 
If Ill be able to find the solution I will post here a message 

Best regards 

E

BaReinhard, 2017-12-19:
@korte84

The bluetooth-agent is the issue here. Simply try the command

`simple-agent.autotrust`

If an error comes up list it here. Also if possible try 
`sudo systemctl status bluetooth-agent -l`

I’m assuming there is some missing dependency for the simple agent which causes the Bluetooth-agent to failed.

korte84, 2017-12-19:
Ok I'll try. I couldn't access to my raspi for the next 3 days. I ll let u know! Thanks :)

BaReinhard, 2017-12-19:
@korte84 cool sounds good! Ill be waiting to hear from you.

korte84, 2017-12-24:
Finally I have BT that works!!! 
I flashed again jessie but this time without the hifiberry plugged in. I figured out it gave some problems like rebooting my router...

I still have the sap-server problem but it doesn't affect the bt or the sspari system at all.

For the record:
Rpi with jessie
NO sound card _(it was the problem)_
SSRPARI home installation (option 2)

Thank you both (@BaReinhard and @totti2) for your support ! 
I wish you merry Christmas 

bye 

E

BaReinhard, 2017-12-24:
What a gift huh? Haha nothing better than things working like they should. Merry Christmas enjoy your holiday and music 😁

BaReinhard, 2017-12-24:
It’s interesting that the Dac being plugged in during install causes problems. I rarely have anything plugged in besides an Ethernet cable for install, so I wouldn’t have encountered that. Recently I have changed the sound card install so when you install a sound card it will remove the onboard sound and remove any issues caused by previous installs.

closed 2017-12-24 by BaReinhard

korte84, 2017-12-24:
Wait wait... I think is my DAC to have a problem nothing with your package or installation BUT  I'm gonna test again